FormerLurker/Octolapse #136: Add text overlays for print/timelapse statistics

FormerLurker, 2018-04-12:
Several users have expressed interest in adding overlay text to the snapshot.  Perhaps it could work in the following way:

1.  Add a text box within either the snapshot or rendering settings (not sure of the best place) for video overlay that can handle the following tokens:
{CurrentTime}
{TimeElapsed}
{LayerNumber}
{LayerHeight}
... more?
2.  Choose an orientation (top, bottom, maybe something else?) and a font size.
3.  Add the overlay to each frame as it's captured.
4.  Party

Ideas, thoughts?

smeg1, 2018-04-12:
hi TimeElapsed would be great to keep track of long prints

Shadowen, 2018-07-23:
I did some investigating and it looks like ffmpeg [drawtext filter](https://ffmpeg.org/ffmpeg-filters.html#drawtext-1) isn't really up to the job. I'm going to try to bake the text/other information in using PIL, which - thankfully - should be much easier.

However, this means we will need to remove the flip/rotate/tranpose filters from the ffmpeg processing pipeline as the text will not stand up to being rotated very well. I will try to implement them in PIL instead (I make use of rotate, so this is important to me!).

We may or may not see a performance hit since PIL is probably slower than ffmpeg. Hopefully this is not an issue. I will prepare a fork and you can check it out.

FormerLurker, 2018-07-23:
I'm getting an error regarding this font:  Pillow/Tests/fonts/FreeMono.ttf

I'm thinking we need to include this file in the project, add it to the setup.py.  Thoughts?

Shadowen, 2018-07-23:
Oops. I assumed it came with Pillow. I'll have to look into what fonts are
safe/can be included.

Shadowen, 2018-07-23:
So #204 should fix the font issue. It also fixes a crash in the timelapse start (sorry!).

I started testing on my Raspberry Pi and it seems like the timestamps I'm using aren't very accurate. I'll take a look at `snapshot.py`, but for some reason the file modification times are quite wonky?

I might just have to move ahead with recording metadata into a seperate file during the snapshot.

In conclusion, no one use the overlay feature yet ;)

FormerLurker, 2018-07-23:
Lol!  I noticed the error on start and chalked it up to using the wrong
signature when you resolved the conflicts.  I forgot about it, to be
honest.  Happens to us all.

Regarding the metadata, should be simple enough.  Just keep in mind that
snapshots can fail, and make sure there are no crashes if there is no
metadata since images can now come from scripts (new feature).  Might not
be an issue depending on how you do things.

Oh, and regarding the odd timestamps, could the problem be that all of the
image files are renamed before processing?  I'm pretty sure they are, but
am not in front of my pc and can't double check.

Shadowen, 2018-07-24:
The images are renamed and it COULD be the cause of the timestamp issue. It could also be the nature of the asynchronous download. I'm not too sure.

It's probably not worth figuring out since it proves that filesystem timestamps are too unreliable anyways.

You'll have to fill me in on loading images from scripts though. Wouldn't pictures taken externally also need to have metadata included to indicate time taken, layer, etc?

For now I'll just put some exception handlers in the rendering code so if the metadata selected for the overlay doesn't exist, it fails to render.

---

As an aside, does it make sense to have a pure standalone rendering flow? Or is that not an intended usage?

Shadowen, 2018-07-24:
Hi, can you test my branch at https://github.com/Shadowen/Octolapse/tree/overlay with the overlay option set to something like this?

```
N={snapshot_number}
T={current_time}
E={time_elapsed}
```

I overhauled the snapshot and rendering process a lot to accommodate the metadata. It WILL fail rendering without a metadata file right now, but I should be able to generate one for the script-based capture as well.

FormerLurker, 2018-07-24:
I'll take a look as soon as possible!  I'll also try to run it with my DSLR and see what happens.

Shadowen, 2018-07-24:
I'll let you know when I integrate the external script with the metadata
method :) I haven't gotten around to doing that yet.

FormerLurker, 2018-07-24:
@Shadowen, check out [this beta wiki page](https://github.com/FormerLurker/Octolapse/wiki/Configuring-and-External-Camera) I created regarding the external camera functionality.  It would be pretty easy to create a script to 'fake' photos if you don't have or want to connect a DSLR to test things out.  I pass in all the necessary arguments (and a few extra ones) to take a picture.  I'll probably many more arguments as it becomes more obvious what will be useful.

Both the document and the feature are very much beta.  I might integrate libgphoto2-dev and write a non-script method too at some point.

FormerLurker, 2018-07-24:
Whoops, wrong url (I changed the title to correct a typo).  Here is the right one:  https://github.com/FormerLurker/Octolapse/wiki/Configuring-an-External-Camera

I will correct the above comment too.

Shadowen, 2018-07-24:
I pasted the metadata file generating code into the external script snapshot section. It **should** work, but please test :)

I'll set up a (maybe fake) DSLR at some point if I get a chance.

FormerLurker, 2018-07-24:
I will definitely do some thorough, and will fix any issues I find.  I'm THRILLED to test this most excellent work.  I may even push some additional data into your metadata file.  I'll be sure to update the text overlay templates to match.

If I create a 'Fake' DSLR script before you get around to it I'll send it to you.  I have only two batteries and no adapter, so I can't do a whole lot of testing with my real camera right now.  I'm planning to get an adapter, but they are $$.

FormerLurker, 2018-07-25:
I've done some preliminary testing with the webcame and external scripts.  There were some minor issues that I corrected, but I have no idea how to create a pull request into your forked project, lol!  Github doesn't seem to like that process.  Let me know if you have any idea how to do this.

All of the changes are in the snapshot.py file.  I also added a timeout to the camera capture script because I had a lockup on my PC.  

Here is a link to a gist, which might be easier than figuring this out:  https://gist.github.com/FormerLurker/f629b96ab07c1cdb995de0dfd7ce7df3

Shadowen, 2018-07-25:
I think you can do a pull request the same way, just with different forks/branches specified. I suppose merging manually is probably easier though :)

Branch is updated and PR incoming.

Shadowen, 2018-07-26:
Closing this since #206 seems to have fixed it. The settings page still needs cleanup, but the core functionality works (more or less).

FormerLurker, 2018-07-26:
@Shadowen, turns out fc-list isn't working on my pi :(  I think we are missing some component.  Of all the little things, lol!

I'll mess around and see if I can create a new routine in the utility.py file.  If you can figure out a way to make it work, let me know.  I'm re-opening this issue since most users are running Octolapse on an rpi.

FormerLurker, 2018-07-27:
Yeah, that be what I end up doing, at least until I can figure out the install issue I'm having with fontconfig.  Some people still can't get pillow installed either, and I haven't been able to replicate the issue even.

Im sure this all comes down to ignorance on my part.  I'm slowly learning though.  Better late than never I guess.

FormerLurker, 2019-10-12:
FYI, I've added some enhancements to the awesome font overlay, and wanted to mention that before closing this.  I added a default font, added a very primitive text outline feature, and a few other small things.  THANK YOU SO  MUCH for your contribution here.  I love me some overlays!

Closed :)

PR #201 (2018-07-23): Overlay text
Please beta test if you are interested!

---

Only have {current_time} and {time_elapsed} implemented so far. This should cover most features (should fix #136) requested anyways. Other features that require G-code knowledge will require metadata files to be generated at snapshot-time. I'll consider this in the future if there is any demand.

---

Also there is currently no way of moving, resizing, colouring, or formatting the text. I'll get around to that sometime.
closed 2019-10-12 by FormerLurker