FormerLurker/Octolapse #137: After position acquisition, the next gcode may not be sent to printer

FormerLurker, 2018-04-13:
Any commands configured for automatic position acquisition (G28, G29, etc) could lead to a situation where the very next gcode is not sent to the printer IF it is not mapped within the parser.

This has been fixed already in the devel branch and will be pushed to the master branch as soon as testing is complete.

AbuMaia, 2018-04-14:
I have been noticing this recently with my bed temp being set before any G28 or G29, but the hotend temp setting after both a G28 and a G29. The hotend cools off, and the Marlin firmware stops the extruder, but the print continues as "normal" otherwise. There is no problem if I disable Octolapse, or if I manually set the hotend temperature before starting a print.

As a workaround, should I have Slic3r insert a dummy gcode between the positioning codes and the temp codes?

FormerLurker, 2018-04-14:
I have a fix in the devel branch if you are willing to install a dev
version I'll send instructions

FormerLurker, 2018-04-14:
Oh, and yes, your workaround will likely solve the issue if you'd rather
try that.

AbuMaia, 2018-04-14:
About to finish a print started with the pre-heat method. Once it's done, I'll run another with the dummy gcode method to see if it works. If that fails, I'll try the dev version.

FormerLurker, 2018-04-14:
> No hotend temperature issues with the dummy gcode. Print is proceeding
> normally.

Excellent.  That's good confirmation.  I'll have a release candidate ready
this weekend, and will let you know when it is finished.

Thanks for your input, I really appreciate it.

AbuMaia, 2018-04-14:
Installed the "devel" branch, giving it a try now. I took out the dummy codes, and there are no hotend temp issues.

FormerLurker, 2018-08-11:
This has been solved.  Closing.  Thanks for participating!

closed 2018-08-11 by FormerLurker