FormerLurker/Octolapse #139: Printer profile lulzbot taz5

mbur371189, 2018-04-14:
#### If this is a feature request describe it here

FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: OCTOLAPSE_VERSION_GOES_HERE

#### Version of OctoPrint

OctoPrint Version: OCTOPRINT_VERSION_GOES_HERE

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES_OR_NO

#### What were you doing when the problem occurred

1. STEP_ONE_GOES_HERE
2. STEP_TWO_GOES_HERE
3. STEP_...

#### What should have happened?
PUT_YOUR_DESCRIPTION_HERE

#### What happened instead?
PUT_YOUR_DESCRIPTION_HERE

#### Operating System running OctoPrint and Octolapse

OS Name: OS_NAME_GOES_HERE
Os Version: OS_VERSION_GOES_HERE

#### Printer model & used firmware incl. version

Printer Model: PRINTER_MODEL_GOES_HERE
Printer Firmware Version: PRINTER_FIRMWARE_VERSION_GOES_HERE

#### Browser and version of browser, operating system running browser

Browser: BROWSER_VERSION_GOES_HERE
Browser OS: BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.  I'm currently saving for some additional test cameras in preparation for develiping multi-camera support.

FormerLurker, 2018-04-14:
Can you give me some info about the printer?  Perhaps the easiest way to start would be to tell me any differences between this printer and the Taz6, since there is already a profile for that printer.  It will likely come down to print volume adjustments, and perhaps retraction/detraction/xyz speeds and retraction/detraction length differences.

When I add a default profile I want it to match any bundled/default slicer profiles if possible, so if you have tweaked your slicer settings it might be best to switch to the default settings for this.  I want to make sure novice users can get octolapse working without understanding all of the slicer settings.

Thanks for submitting this feature request!

FormerLurker, 2019-10-12:
So I've added a new profile repository, and have already added the Taz6.  If you have created a Taz 5 profile, I'd love to see it so it can be added to the new repo.  Starting in Octolapse V0.4, you'll be able to configure a printer simply by selecting the make/model from a dropdown list, so I'm pretty keen on adding more printers to the repo!

Thanks, and if you want me to reopen this issue please let me know!

closed 2019-10-12 by FormerLurker