FormerLurker/Octolapse #140: [Request] GIF as Output File Format

blazze11, 2018-04-14:
#### If this is a feature request describe it here

Would it be possible to add GIF as Output File Format?

#### Version of Octolapse
 
Octolapse Version: Octolapse v0.3.1

#### Version of OctoPrint

OctoPrint Version: OctoPrint 1.3.7 / OctoPi 0.14.0

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES_OR_NO

#### What were you doing when the problem occurred

1. STEP_ONE_GOES_HERE
2. STEP_TWO_GOES_HERE
3. STEP_...

#### What should have happened?
PUT_YOUR_DESCRIPTION_HERE

#### What happened instead?
PUT_YOUR_DESCRIPTION_HERE

#### Operating System running OctoPrint and Octolapse

OS Name: OS_NAME_GOES_HERE
Os Version: OS_VERSION_GOES_HERE

#### Printer model & used firmware incl. version

Printer Model: PRINTER_MODEL_GOES_HERE
Printer Firmware Version: PRINTER_FIRMWARE_VERSION_GOES_HERE

#### Browser and version of browser, operating system running browser

Browser: BROWSER_VERSION_GOES_HERE
Browser OS: BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.  I'm currently saving for some additional test cameras in preparation for develiping multi-camera support.

FormerLurker, 2018-04-14:
I'll definitely look into it.  If it's not too complicated to add I'll push it out with the next release.  I must say I've been waiting for someone to request this format :)

Shadowen, 2018-05-11:
Hi, please see https://github.com/Shadowen/Octolapse/tree/gif_output for GIFs. This change depends on my watermarks branch, PR #158. I'll submit a PR for this one afterwards.

If you want a pre-release sneak peek, there's instructions for installing from source [here](https://github.com/FormerLurker/Octolapse/wiki/Installation#manually-install-a-specific-branch-via-the-plugin-manager). It's not hard and I would appreciate a tester :)

Shadowen, 2018-05-16:
@blazze11, issue is resolved :)

closed 2018-05-16 by Shadowen