FormerLurker/Octolapse #141: Possition error with G28 in octoprint  "Before print job starts" section

penkerton, 2018-04-14:
#### If this is a feature request describe it here

FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: 0.3.1

#### Version of OctoPrint

OctoPrint Version: 1.3.7

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: NO

#### What were you doing when the problem occurred

1. delete G28 from slicer "before print script".
2. embend G28 G-code to octoprint GCODE Scripts tab "Before print job starts".
3. start printng

#### What should have happened?
normal plugin work
(with G28 in file - it's ok, in GCODE Scripts - Possition error)

#### What happened instead?
many many Position errors all around window
"X None Y None Z None E value"

#### Operating System running OctoPrint and Octolapse

OS Name: OctoPi
Os Version:  0.14.0

#### Printer model & used firmware incl. version

Printer Model: custome AM8 (ex Anet A8)
Printer Firmware Version: Marlin 1.1.8

#### Browser and version of browser, operating system running browser

Browser: Chrome 65.0.3325.181
Browser OS: Windows 7

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  I have issue with uploading... content of the file is the same like:
2018-04-14 18:14:14,336 Position - Coordinates (X:None,Y:None,Z:None,E:2020.98200) are out of the printer area!  Cannot resume position tracking until the axis is homed, or until absolute coordinates are received.

#### Link to octoprint.log

Link to octoprint.log:  https://gist.github.com/penkerton/c1d848472cb054a5a2351572ffd67985

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.  I'm currently saving for some additional test cameras in preparation for develiping multi-camera support.

FormerLurker, 2018-04-14:
That's interesting.  Thanks for reporting this.  I have a good idea why this might be the case, and I need to look at the gcode tagging system to see what I can do about it.

FormerLurker, 2018-04-15:
I've created a more specific issue Ocotprint the start gcode problem here #144.  A hotfix will be available soon, and I'll let you know when it's ready and how to install if you are willing to try it out.

Thanks again!

FormerLurker, 2018-04-15:
I spoke too soon.  I cannot fix this bug at the moment because Octoprint will not allow me to stop gcode processing during while it's sending its start gcode sequence.  I've sent a feature request into the OctoPrint repo, so fingers crossed!

I'm going to close this issue and will be using issue #144 to report updates.  I'm sorry about the delay.

closed 2018-04-15 by FormerLurker