FormerLurker/Octolapse #146: Control issues leading to print failures with Marlin firmware.

CptanPanic, 2018-04-17:
As shown below I was having various strange failures that i never had until I realized that I had installed octolapse, once uninstalling it everything works as expected again.  I do not have any logs at the moment, but can run another test to collect data if that is helpful.

#### Version of Octolapse
 
Octolapse Version: v0.3.1

#### Version of OctoPrint

OctoPrint Version: 1.3.8

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: NO

#### What were you doing when the problem occurred

Ran various prints.

#### What should have happened?
Print completed succesfully.

#### What happened instead?
I got various errors randomnly that would cause the print to fail of "expected line x instead of y".  Also viewing print in person I would see the head making strange overshoots.  Note I never had these issues before, and everything is working as expected after unistalling plugin.  I had 4 failures in 5 prints.  

#### Operating System running OctoPrint and Octolapse

OS Name: Octopi stretch
Os Version: stretch 
Raspi 3 b+

#### Printer model & used firmware incl. version

Printer Model: Generic Prusa i3.
Printer Firmware Version: Marlin 1.1.8

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: TBD

#### Link to settings.json

Link to settings.json with all passwords removed: TBD

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  TBD

#### Link to octoprint.log

Link to octoprint.log:  LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: LINKs_GO_HERE

FormerLurker, 2018-04-17:
Are you using start/end gcode within Octoprint by chance?

Yes, your logs would be most helpful, especially if you can run a print with the 'Full Diagnostic - Live Print' mode and download and post your plugin_octolapse.log file.

Thank you for reporting the issue.  I hope I can get it fixed soon!

FormerLurker, 2018-09-10:
Since this issue has been open for a long time with no activity I'm going to close it out.  It's possible that this issue was solved within one of the many bugfixes now in v0.3.4.  If you want me to re-open this issue just reply and I will do so.

Thanks for posting!

closed 2018-09-10 by FormerLurker