FormerLurker/Octolapse #151: [Request] Profile for Printrbot Simple Metal (Extended)

OctopusProteins, 2018-04-27:
Hi, I would like to use this plugin with my Printrbot. A profile doesn't exist yet. Would you be willing to make one with my help, and if not, what would I have to do to make and submit a profile? Thanks.

FormerLurker, 2018-04-27:
Sure!  First copy all settings from your slicer into the Octolapse printer
profile. Then make sure your print volume is correct (can be set either in
Octolapse or in the octoprint printer profile). If you send me the start
gcode I will review and let you know if anything else needs to be changed.

Your first print should be small and quick so that it's easy to spot issues
and debug.  I like to use a small thinwall calibration print for this.  You
can also use the 'test mode' debug profile, which prevents heating and
extrusion for a quicker test.

I should have some time tomorrow if you need more help, so feel free to ask
questions!

OctopusProteins, 2018-04-27:
Thanks for the quick response. 
If I am not using Z Hop, should I set the value to 0?
Which movement speed should I use (travel or print?)
Cura shows 0 for the maximum z speed, indicating that it is using firmware defaults. What should I put into Octolapse for Z speed?

Here is the starting and ending GCODE. 
`G21       ;metric values
G90       ;absolute positioning
M82       ;set extruder to absolute mode
M106      ;start with the fan on
G28 X0 Y0 ;home X/Y
G28 Z0    ;home Z
G92 E0    ;zero the extruded length
G29       ;initiate auto bed leveling sequence
;start cleaning sequance
G1 X250 Y150 Z15 F4000
G1 X250 Y150 Z0.30 F1000
G1 X1 Y150 Z0.25 E15.0 F2000
G1 X1 Y150 Z0.25 E14.0 F4000
G1 X1 Y1 Z0.25 F16000
G1 X1 Y1 Z0.25 E15.0 F4000
G92 E0
M107      ;start with the fan off
;end cleaning sequance
;G92 X132.4 Y20 ;correct bed origin (G29 changes it)`

`M104 S0     ;extruder heater off
M140 S0     ;heated bed heater off (if you have it)
M106 S0     ;fan off
G91         ;relative positioning
G1 E-1 F300 ;retract the filament a bit
G1 Z+1 E-5 F9000 ;move Z up a bit and retract even more
G28 X0 Y0   ;home X/Y, so the head is out of the way
M84         ;steppers off
G90         ;absolute positioning`

FormerLurker, 2018-04-27:
O for zhop, movement speed = travel speed.  Set z speed to movement speed,
but it won't matter without zhop.  I think that will do it!

aluhrs13, 2018-05-21:
@OctopusProteins did this end up working for you?

OctopusProteins, 2018-05-22:
Sorry; yes, everything ended up working great. Thanks!

closed 2018-05-22 by OctopusProteins

schrill, 2019-12-07:
Can you share the settings.json file with the printrbot profile? Tnx

FormerLurker, 2019-12-07:
@schrill, can you send me some info or links about the printrbot?  With that I could create a profile in the new profiles repository, which will be available for download/import in v0.4.