FormerLurker/Octolapse #154: Themeify compatibility

foodbandlt, 2018-05-01:
#### If this is a feature request describe it here

Just thought this would be a cool touch.  Right now a lot of background colors and text colors are explicitly defined.  Was wondering if you'd be willing to define them as "inherit" to play nice with themes that define text and background colors at the top level.  Not sure how this plays with the regular Octoprint theme so it would require testing, but if you have time this would be great!

https://i.imgur.com/z5XZqTy.png

#### Version of Octolapse
 
Octolapse Version: 0.3.1

#### Version of OctoPrint

OctoPrint Version: 1.3.8

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES_OR_NO

#### What were you doing when the problem occurred

1. STEP_ONE_GOES_HERE
2. STEP_TWO_GOES_HERE
3. STEP_...

#### What should have happened?
PUT_YOUR_DESCRIPTION_HERE

#### What happened instead?
PUT_YOUR_DESCRIPTION_HERE

#### Operating System running OctoPrint and Octolapse

OS Name: OS_NAME_GOES_HERE
Os Version: OS_VERSION_GOES_HERE

#### Printer model & used firmware incl. version

Printer Model: PRINTER_MODEL_GOES_HERE
Printer Firmware Version: PRINTER_FIRMWARE_VERSION_GOES_HERE

#### Browser and version of browser, operating system running browser

Browser: BROWSER_VERSION_GOES_HERE
Browser OS: BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.  I'm currently saving for some additional test cameras in preparation for develiping multi-camera support.

FormerLurker, 2018-05-01:
I've seen some screen caps with that plugin enabled and have noticed that
it doesn't play nicely.  As soon as I have some time I'll install the
plugin and see if I can't find some simple changes to at least improve
things.  I'll let you know how that goes.

Thanks for the suggestion!

FormerLurker, 2018-08-11:
@foodbandlt, I just pushed an initial commit that should fix things somewhat for Octolapse and themeify, though there is definitely more work to be done.  It looks pretty good with the default theme :)

The code is in the devel branch.  If you want to try it out feel free.  Lots has changed, and there is some new (less tested) functionality in there.

FormerLurker, 2018-09-10:
This is working now in v0.3.4!  Closed.

closed 2018-09-10 by FormerLurker