FormerLurker/Octolapse #156: Remove settings.Rendering class

Shadowen, 2018-05-02:
See if we can remove the settings.Rendering class. It seems to just be basically implementing a dictionary.

Just a reminder that I'll get back to soon.

Shadowen, 2018-05-10:
Use http://docs.octoprint.org/en/master/plugins/mixins.html#settingsplugin and http://docs.octoprint.org/en/master/modules/plugin.html#octoprint.plugin.plugin_settings?

Shadowen, 2018-05-10:
@FormerLurker is there a reason why OctoLapse settings are all implemented outside of the OctoPrint Settings framework? I feel like everything could be implemented without settings.* classes, etc if we just used OctoPrint's SettingsPlugin/SettingsManager utilities.

FormerLurker, 2018-05-10:
There are a few reasons, but they mostly boil down to me being unable
and/or unwilling to learn/figure out how to get things working as I wanted
them to within the exiting framework.  Remember this was my first ever
attempt at anything like this, and I really found the existing structure to
be counter-intuitive.  Combine that with the fact that I implemented the
settings structure first, and that I got REALLY REALLY frustrated doing it
(learning knockout and using a very outdated version of bootstrap) and you
get the current setup.

However, I have to say that I am not really happy with the 'store
everything in a dict' method, either.  They take way less work to
implement, but then are, IMO, unwieldy to use, and are not self-documenting
like a class structure.  This is probably just a bias from my previous
experience (database world).

That is probably not very satisfying.  What I'd REALLY like is a database,
and an ORM of some type.

Shadowen, 2018-05-10:
I'm not happy with having to add a new "setting" in like 3-5 different places. I agree you should have to specify the name, but preferably only once (and in the same place as the default is specified). It seems like we have multiple defaults scattered throughout the code base as well...

I don't think you want a database; you just want a JSON store with a fixed schema. This schema can be defined either in Python source of a standalone JSON file.

Perhaps something like [this](http://json-schema.org/example1.html)? Seems like overkill though.

FormerLurker, 2018-05-10:
You're right, there is too much work to do ATM to add new settings, and I'm sure it can be simplified in many ways, including what you've suggested.  I'll look into it, but I can't promise how soon it will be.

I'll look into the 'multiple defaults' issue, though you may be confusing default profiles with default values for the settings (there's a difference in this case).  It's equally possible that there are indeed multiple default values, and those should be eliminated.

Also, I'm pretty sure the existing settings classes can be simplified on its own, and I welcome more suggestions.

FormerLurker, 2018-05-10:
Hey, I just took a look at the settings class (I avoid this whenever possible) and a very large part of it are copy functions.  I'm not familiar with an easy way to clone an object in python, and I just did it with brute force.  Surely there's an easier way to do this in python, right?

Shadowen, 2018-05-10:
Does [this](https://docs.python.org/2/library/copy.html) do the job? :)

Shadowen, 2018-07-26:
I've been thinking about this problem again, and I still think the best way is use pure JSON for the settings. This means that they will be easily convertible to dictionaries in Python, plain text JSON for storage, and JS objects in JavaScript.

So for the following setting object:
**setting.json**
```
{
  "settingA": false,
  "settingB": {
    "width": 5,
    "height": 10
  },
  "listOfThings": [
    "One",
    "Two"
  ]
}
```

We'd get the following (almost identical) dict in Python when we run json.loads() on it:
**Python dict**
```
{
  "settingA": False,
  "settingB": {
    "width": 5,
    "height": 10
  },
  "listOfThings": [
    "One",
    "Two"
  ]
}
```

The schema would be stored in a separate file (settings-schema.json) containing something like this:
**settings-schema.json**
```
{
  "type": "object",
  "properties": {
    "settingA": {
      "type": "boolean",
    },
    "settingB": {
      "type": "object",
      "properties": {
        "width": {
          "type": "integer",
        },
        "height": {
          "type": "integer",
        }
      }
    },
    "listOfThings": {
      "type": "array",
      "items": {
        "type": "string",
      }
    }
  }
}
```

I'm thinking a nice editor like [this one](https://jsonschema.net/) would help out a lot in writing the schema.

--

This documents the fields, and with the appropriate libraries, even does all the validation and stuff too. Does this meet the requirements?

FormerLurker, 2018-07-27:
Pretty neat!  It looks like it meets most of the requirements.  I've got a few questions though.  How do we store the default values in such a scheme?  One of the biggest pains for me in python is not having good intellisense, which is why I like to use a class for the settings (my plan was to use properties).  Is there some way to have our cake and eat it too?

FormerLurker, 2018-07-27:
I found this article that has some neat ideas:  https://code.tutsplus.com/tutorials/serialization-and-deserialization-of-python-objects-part-1--cms-26183

Shadowen, 2018-07-28:
> Hey, I'm going to undo a commit to Octolapse/devel that I think was a
> mistake: b0dccbe
>
> This looks partially finished and broke settings load. We can work these
> changes in again from your settings branch. It looks promising by the way!

I thought that was in my branch? I don't remember submitting PR for that.

Shadowen, 2018-07-28:
Wait, do I have push permissions for `FormerLurker/devel`? Maybe I pushed directly into there by accident because I didn't set up my remotes properly D:

Oops. I might just unbind my remote push to FormerLurker lol...

FormerLurker, 2018-07-28:
I added you as a collaborator way back in the beginning, so you have push
permissions it seems, lol!  I'll look into that and see if I can somehow
remove that.  I know if I were you I wouldn't want push permissions!

FormerLurker, 2018-08-01:
I reviewed the documentation, and except for the drawback you mentioned it looks pretty good.  However, as you alluded, this would make things difficult for new contributors.  I won't count it out 100% yet, though, but I'd like to try my hand at creating a base class that does what we want.  Here are the requirements as I see them:
For the Python Classes
1.  Each class must generate a dict automatically (to_dict)
2.  Each class must be able to instantiate with no parameters and return an object initialized with all the default settings.
3.  Each class must be able to take an object of the same and return a copy.
4.  Each class must be able to take a dict and return an object set to the values within the dict.

If you want to start on this, think about implementing a base class that can perform the above four operations and send me whatever you come up with.

As for the javascript, I don't think we'll be able to get away without creating our own viewmodel (short of implementing the protobuf scheme).  However, this would still be a rather huge simplification.

Thanks for all of your thoughts and evaluations.  I find it really helpful and interesting to think about!

Shadowen, 2018-08-01:
How does [this](https://github.com/Shadowen/Octolapse/blob/settings_refactor/octoprint_octolapse/settings.py) look?

Basically, creating a *Settings class is like this:

```python2
class PrinterSettings(Settings):
    def __init__(self, name="New Printer", **kwargs): # Default arguments are implemented using keyword args.
        # We can define arbitrary properties with defaults here.
        self.description = ""
        self.retract_length = 2.0
        self.retract_speed = 6000
        ...
        
        super(PrinterSettings, self).__init__(name, **kwargs)
        # self.update(kwargs) is called in the __init__ by the superclass.
```

Check out [this test](https://github.com/Shadowen/Octolapse/blob/settings_refactor/octoprint_octolapse/test/test_settings.py) for usage. It doesn't generate a dict on its own, but I don't think that's actually required. It generates the JSON string directly and can parse back from that string though.

Parsing nested classes was fun ;) I'm pretty sure it works, but only if all the classes subclass from my superclass (obviously, I need some sort of control over how they serialize and deserialize). All the standard Python<->JSON conversions apply (dict<->object, ints, strings, bools, etc.).

Comments, questions?

FormerLurker, 2018-08-01:
Ohhhh.. that is nice!  I'm going to play around with that soon, probably this weekend.

There are some things within the settings that need to return dicts, but they don't actually NEED to be in the settings file.  I guess that means they shouldn't be there!

I'm super glad you have done all this research.  I'm actually excited to remove and replace those bloated classes.  I'm as surprised as you!

Shadowen, 2018-08-02:
Heh, I just realized I have no idea how I intended to do the JS side of
this... Maybe just handle everything as a JSON and tack on the _class tag
as it leaves.

FormerLurker, 2019-10-12:
So, the settings have been completely rewritten, removing most of the concerns expressed in this issue, but the viewmodels remain.  Because of this i'm going to go ahead and close this issue, but at some point I'd like to revisit and clean up the viewmodels, so maybe that needs a new issue.

Thanks!

closed 2019-10-12 by FormerLurker