FormerLurker/Octolapse #157: [Request] Add an option to change the temp folder

szafran81, 2018-05-02:
#### If this is a feature request describe it here

Add an option to customize the temp folder for saving snapshots.

#### Version of Octolapse
 
Octolapse Version: 0.3.1

#### Version of OctoPrint

OctoPrint Version: 1.3.8

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES_OR_NO

#### What were you doing when the problem occurred

1. STEP_ONE_GOES_HERE
2. STEP_TWO_GOES_HERE
3. STEP_...

#### What should have happened?
PUT_YOUR_DESCRIPTION_HERE

#### What happened instead?
PUT_YOUR_DESCRIPTION_HERE

#### Operating System running OctoPrint and Octolapse

OS Name: OS_NAME_GOES_HERE
Os Version: OS_VERSION_GOES_HERE

#### Printer model & used firmware incl. version

Printer Model: PRINTER_MODEL_GOES_HERE
Printer Firmware Version: PRINTER_FIRMWARE_VERSION_GOES_HERE

#### Browser and version of browser, operating system running browser

Browser: BROWSER_VERSION_GOES_HERE
Browser OS: BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.  I'm currently saving for some additional test cameras in preparation for develiping multi-camera support.

FormerLurker, 2018-05-03:
I'd have to double check, but I'm pretty sure I'm using a subfolder of the Octolapse data folder to store the temp images.  A long time ago I had a user setting for the temp folder path/filename, but removed it to simplify some of the controls which had been exploding out of control :)

@ejjenkins , I'll take a look at the Octoprint settings.  Using this path may be a good compromise.

szafran81, 2018-05-04:
@ejjenkins Can you please be more specific ? Note that I didn't ask about how to change any of the OctoPrint dirs (I'm perfectly satisfied with having them on my emmc module), but just the dir where Octolapse saves the snapshots (I want them on my HDD and not the emmc).

ejjenkins, 2018-05-09:
@szafran81 - Wow, you are correct, I assumed OctoLapse used the same directories.  I have now added a link to my HDD as well.  It was saving the video there, but not the temp directory.

@FormerLurker - It would certainly be easier for most to use the existing OctoPrint directories or add a spot in OctoLapse settings for changing directories.

Again, thank you for all the work you are doing on this.

FormerLurker, 2018-05-09:
No problem!  It is an easy enough change.  I'm planning to also add a naming template for the temp images.

szafran81, 2018-05-10:
Also a bit of OT but I doubt it deserves a separate topic. A preview for the filename template would be nice to have (to see what the filename would look like after timelapse is ready).

szafran81, 2018-05-13:
Haven't looked at the code but most of the code for that should be there since it's used to generate the filename at timelapse generation.

Simple...
My filename template looks like this:
{PRINTENDTIME}__{FAILEDFLAG}{FAILEDSEPARATOR}{FAILEDSEPARATOR}{GCODEFILENAME}
under it display something like this:
"Filename preview: 20180513090213__FAILED__<GCODE-FILENAME>.ext"
or even simpler, just present each placeholder with a static example:
{FAILEDFLAG} = FAILED
{PRINTENDTIME} = 20180513090213
etc.

If you like I can still make a separate topic out of this.

FormerLurker, 2018-05-13:
Interesting idea @szafran81.  I will ponder this, but please open a separate issue so things don't get lost.

FormerLurker, 2020-02-26:
This has been added to the latest octolapse devel/rc release:

[image]

Let me know if you want to try it out,  I'm going to go ahead and close this issue, but if you want to comment on this request, please do so in [this thread](https://github.com/FormerLurker/Octolapse/issues/455).

Thanks for the suggestion, and sorry it took so long to implement!

closed 2020-02-26 by FormerLurker