FormerLurker/Octolapse #159: Octolapse Startup Failed Canceling print. Error: The ffmpeg C:\ffmpeg\bin does not exist.

Crunch69, 2018-05-03:
Octolapse Startup Failed
Unable to start the timelapse. Canceling print. Error: The ffmpeg C:\ffmpeg\bin does not exist. Please configure this setting within the Octoprint settings pages located at Features->Webcam & Timelapse under Timelapse Recordings->Path to FFMPEG.

Octoprint 1.3.8 on prusa MK3 FW 3.2.0 on Win 10 pc
Octoprint work correctly and streams video when I activate Octolapse i get the message above FFMPEG works correctly and was tested playing videos it just for some reason does not work with Octolapse

Crunch69, 2018-05-03:
Yes the file does exist on the drive c:\ffmpeg\bin\ffmpeg.exeI can open a video point to that location and the video plays or type commands in cmd.  Only when I put the path  in Octoprint with octolapse timelapse and try test path or try to just run octolapse it says path is incorrectCheers Art

Shadowen, 2018-05-09:
Please open a new issue to keep things organized :) First thing I'd do is to figure out why the coordinates are negative (See Y: -3?).

This error also fixes itself (AFAIK) if your printer homes after it happens.

closed 2018-05-09 by Shadowen