FormerLurker/Octolapse #160: [Info] Octolapse and MultiCam

szafran81, 2018-05-09:
#### If this is a feature request describe it here

FEATURE_REQUEST_DESCRIPTION_GOES_HERE

This is just for information purposes. Since I'm in the middle of a 4 day print I can't check it. Would be very nice if someone could check and write here if Oclopse works with https://plugins.octoprint.org/plugins/multicam/
Also I think it would be nice if Octolapse could work with 2 cams - maybe something like PIP on TVs.

If this post is offending someone then sorry, and please remove it.

#### Version of Octolapse
 
Octolapse Version: OCTOLAPSE_VERSION_GOES_HERE

#### Version of OctoPrint

OctoPrint Version: OCTOPRINT_VERSION_GOES_HERE

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES_OR_NO

#### What were you doing when the problem occurred

1. STEP_ONE_GOES_HERE
2. STEP_TWO_GOES_HERE
3. STEP_...

#### What should have happened?
PUT_YOUR_DESCRIPTION_HERE

#### What happened instead?
PUT_YOUR_DESCRIPTION_HERE

#### Operating System running OctoPrint and Octolapse

OS Name: OS_NAME_GOES_HERE
Os Version: OS_VERSION_GOES_HERE

#### Printer model & used firmware incl. version

Printer Model: PRINTER_MODEL_GOES_HERE
Printer Firmware Version: PRINTER_FIRMWARE_VERSION_GOES_HERE

#### Browser and version of browser, operating system running browser

Browser: BROWSER_VERSION_GOES_HERE
Browser OS: BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.  I'm currently saving for some additional test cameras in preparation for develiping multi-camera support.

FormerLurker, 2018-05-09:
Thanks for your request!  I'm actively working on multicam support, but it will take a while to work out exactly how.

FormerLurker, 2018-05-09:
Whoops, sent too early, sorry.  If you have any ideas about how this should work, let me know, thanks!

Wolfsblvt, 2018-05-16:
Would be also cool to have the option to just create two timelapses.

FormerLurker, 2018-05-16:
@Wolfsblvt, I'm working on this one.  I am thinking that maybe I can add a post-processing section to handle this request, but it will have to wait until multi-cam multi-timelapse has been added.

FormerLurker, 2018-08-11:
@Wolfsblvt, I have implemented multi-camera snapshots.  It is super beta, but is available on the latest devel branch.  Let me know if you want to try it out.  I'm going to close this ticket since it is solved, but feel free to message back if you want some details.

closed 2018-08-11 by FormerLurker