FormerLurker/Octolapse #161: Not counting layers

Growx, 2018-05-12:
#### Version of Octolapse
 
Octolapse Version: 0.3.1

#### Version of OctoPrint

OctoPrint Version: 1.3.8

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES

#### What were you doing when the problem occurred

1. Started the print
2. Finished the print

#### What should have happened?
Create the timelapse

#### What happened instead?
Octolapse Synchronization Failed
No frames were captured.

Octolapse - Position Error
Position - Coordinates (X:100.88000,Y:301.20000,Z:20.10000,E:1180.38030) are out of the printer area! Cannot resume position tracking until the axis is homed, or until absolute coordinates are received.

#### Operating System running OctoPrint and Octolapse

OS Name: OctoPi
Os Version: 1.3.8

#### Printer model & used firmware incl. version

Printer Model: Anet A6
Printer Firmware Version: Marlin 1.1.8

#### Browser and version of browser, operating system running browser

Browser: BROWSER_VERSION_GOES_HERE
Browser OS: BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: https://gist.github.com/Growx/31a24f14f99c671b184318324306bb50

#### Link to settings.json

Link to settings.json with all passwords removed: SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  https://gist.github.com/Growx/4aa5059f97a8360f8df4f8561a8297f8

#### Link to octoprint.log

Link to octoprint.log:  https://gist.github.com/Growx/ef439e7cb15d7370a6813b4eae9ba1d1

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.  I'm currently saving for some additional test cameras in preparation for develiping multi-camera support.

FormerLurker, 2018-05-15:
Hey, SO SORRY I didn't respond to this earlier.  Actually, I thought I DID respond, but obviously didn't.

Thanks for providing the logs!  Unfortunately the plugin_octolapse.log looks to be truncated, and I think the good bits would have been at the very bottom in this case.  However, I should be able to plug in your settings and test out your gcode file with the virtual printer.  If I have time I'll give it a shot after work.

Thanks for submitting your issue!

Growx, 2018-05-15:
Hi FormerLurker, no problem man, I solved the issue by setting the Priming Height to 0 and disabling it. I think that the octolapse isn't getting out of this loop and isn't counting the layers. I tested with 0.2, 0.5 and 0.75 mm, but no one was taking snaps or counting layer and height. When set to 0 to disable the priming, it worked normally.

FormerLurker, 2018-05-15:
Hmm, then something must be wrong with the priming detection.  I'll still take a look to see why this is the case.  If I find a problem and solution maybe you would be willing to test it?

Growx, 2018-05-16:
Of course yes

FormerLurker, 2018-05-16:
Hey, your gcode worked fine, so it must be an issue with your print volume.  Check your printer profile settings (I assume you are using the Anet A8 profile, right?), check the 'Override Octoprint Printer Profile' bpx and change your print volume to:

XMin: 0
XMax: 220
YMin: 0
YMax: 220
ZMin: 0
ZMax: 250

Please double check this with the MFG to make sure my numbers are correct.  After this, save your changes and retry the gcode.  Let me know how it works!

Growx, 2018-05-29:
Hi FormerLurker, I forgot to tell you the result. The issue was solved, I disabled the priming height on Octolapse and know it is working fine. Thank you.

closed 2018-05-29 by Growx

FormerLurker, 2018-05-29:
Oh, cool!  Do you happen to know what the priming height was set to, and what your first layer height was?  I'm just trying to make sure that the priming setting is working as intended.

V Glad you got it working!