FormerLurker/Octolapse #162: [Request] Option for snapshot timeout after head positioning

DyGyg, 2018-05-12:
I see some vibration of background and the table in my video (my camera is fixed on the table):
https://www.youtube.com/watch?v=8F0g_8Blo8Y

Is it possible to set some timeout between head stabilisation and taking snapshot?
This option in plugin settings would be great IMHO.

DyGyg, 2018-05-13:
the same vibrations here:
https://www.youtube.com/watch?v=ffneRGxPnAw
i dont understand how fix it ... :\

DyGyg, 2018-05-13:
hm... and here it's much less vibrations:
https://www.youtube.com/watch?v=7lLzyeyO3mA
only the different i see is that stabilisation point is alittle closer to the camera
may be the problem is in camera set focus time.... :\ will try different points

DyGyg, 2018-05-13:
forget about another difference! may be it's important - in timelapses with "vibration" i've set stabilization in fixed point in milimeters. may be some calculation errors gives little different results for movements of table and head? as a result - little different position of head and table at screenshots?

DyGyg, 2018-05-13:
P.S. and i've found snapshot delay in camera config. so renaming the issue....

DyGyg, 2018-05-13:
https://www.youtube.com/watch?v=4aWe45sY4EU
Stabilization - Center (X and Y - relitive 50% ), snapshot delay - 125ms. Model is in center of the table - close to the camera. "Vibrations" is still here.... (((
And i remembered - another difference in setting i've made - choose "Layer change - Zhop & Rectract only", instead simple "Layer change"... 
Next i will try delay 250ms + "Layer change"

FormerLurker, 2018-05-13:
Sorry it took me so long to respond!

I've been seeing this issue more recently after I altered the threading code for the snapshots.  The problem is I never seem to have the issue on my hardware (125MS delay is all I need).  I'm going to try to create a branch that takes the snapshot snapshot synchronously but that does all post-processing on a thread.  If the problem is within Octolapse this should reveal it.

In your case you haven't had to use extremely high delays (I've seen as high as 3000MS), so let me know if 250MS works for you.  The min expected delay would be 1/Framerate*1000, though things like autofocus, white balance, etc add to this time.

DyGyg, 2018-05-14:
hi!
hm... you are right may be, i dont remember octolapse version with which "NONvibrating" video was made...
Will try 250ms delay later and will wait the branch - ready to help with tests.

DyGyg, 2018-05-24:
sorry for long silence, finaly i've tryed 250ms delay and "vibrations" are gone )
https://www.youtube.com/watch?v=bFUqb3PzP1s
i think it's great ) thanks again for yor great plugin

FormerLurker, 2018-05-24:
That's great!  I'm still working on reducing this delay, though 250ms seems reasonable.  There's always room for improvement!  

Thanks for letting me know that it's now working for you.  I'll close the issue now.

closed 2018-05-24 by FormerLurker