FormerLurker/Octolapse #164: Octolapse stops extruder

GeekDad63, 2018-05-14:
#### If this is a feature request describe it here

FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: 0.3.1

#### Version of OctoPrint

OctoPrint Version: 1.3.8

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES test mode

#### What were you doing when the problem occurred

1. turned on octolapse
2. started print

#### What should have happened?
Printer should have printed

#### What happened instead?
extruder stopped ectruding filament

#### Operating System running OctoPrint and Octolapse

OS Name: osx 
Os Version: 10.13.4 (High Sierra)

#### Printer model & used firmware incl. version

Printer Model: PRINTER_MODEL_GOES_HERE
Printer Firmware Version: PRINTER_FIRMWARE_VERSION_GOES_HERE

#### Browser and version of browser, operating system running browser

Browser: Chrome Version 66.0.3359.139 (Official Build) (64-bit)
Browser OS: BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  I don't know how to do this

#### Link to octoprint.log

Link to octoprint.log:  I don't know how to do this

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.  I'm currently saving for some additional test cameras in preparation for develiping multi-camera support.

GeekDad63, 2018-05-15:
> Test mode is supposed to stop the extruder. It is meant for trying out settings without printing or heating. If you want a log, try diagnostic mode live print, else just use the no logging mode.
>
> Let me know if you have any other questions!

Thank you. I didn’t know that I’ll try the other modes.

John

FormerLurker, 2018-08-11:
Closing this ticket out, I think it is solved.

closed 2018-08-11 by FormerLurker