FormerLurker/Octolapse #165: [Suggestion/Request] Move the filename template in a more generic place

gege2b, 2018-05-15:
Hi there

#### If this is a feature request describe it here
The filename template is a really good feature, but having each rendering profile giving the same setting is not very ergonomic

- What if I want to select another rendering profile which has a wrong filename template and forgot to change it ?
- What if I add a few new profile in a row ? have to copy/paste each time the template ?

So my suggestion is to add/move the filename template stuff in a more generic place (the main settings seems the better choice) to set the template here

A more granular way is to make this generic setting a default one, and keep the template setting in each rendering profile, so we could override them when needed

I don't know if I'm really clear, don't hesitate to ask for more information :)

#### Version of Octolapse
Octolapse Version: Octolapse 0.3.1

#### Version of OctoPrint
OctoPrint Version:      OctoPrint 1.4.0.dev1190+gd8eb5754

FormerLurker, 2018-05-15:
So, if I'm understanding you correctly you want something like the following:

Main Settings ->  Default Rendering Directory, Default Rendering Filename
Rendering Settings -> Add a token for the default rendering directory and default rendering filename, so that custom overrides can be created at will.

FYI, I am currently working on multi-camera timelapse/rendering, so i have to be very careful how this is implemented since we will have camera->rendering pairs, and the file name templates cannot be the same, else we will have conflicts.  I always have the most trouble with settings when designing new features, and this one is a doozy :)

Also, regarding adding new profiles, you can copy an existing profile, which would maintain all of your custom settings.  I realize this doesn't solve the original problem, but wanted to point that out because it's been very useful for me :)

FormerLurker, 2018-05-15:
I'm sure it's possible, but it needs careful thought.

Out of curiosity, what is your native language?  I enjoy learning more about the people who use Octolapse.  Feel free to email me at FormerLurker@protonmail.com any time!

I'll keep thinking about this and will update the issue as I learn more.

gege2b, 2018-05-15:
I'm french, and was sleeping or yawning most of the time during my english courses at school :rofl:  
Thanks to internet and US TV shows !

Thanks you too, for the interest you give to this request and this great plugin!

FormerLurker, 2019-10-12:
I know this is an old issue, sorry it's been sitting idle for so long.  Here is an update:

Multicam has been implemented, but I've still not added an override for the default folders.  Also, some people are really happy that it is possible to have multiple rendering filename templates, so I will probably leave that as it is.  Issue #157 will remain open, which is asking for a way to change the default folders, and I will eventually (hopefully) get around to that.

For those reasons I'm going to go ahead and close this issue down.  If you want me to reopen it, let me know, or feel free to create a new issue.

Thank you very much for posting!!!!

closed 2019-10-12 by FormerLurker

gege2b, 2019-10-21:
Hi there
no problem, It's pretty understandable :)
thank you for the time you give for this issue