FormerLurker/Octolapse #167: [Feature request] "Test mode" logging profile should strip firmware retract GCode

gege2b, 2018-05-16:
Hi there

#### If this is a feature request describe it here
"Test mode" logging profile is working fine (at least on my setup) except he doesn't strip the Firmware Retract/Unretract GCode (G10/G11 in all firmware supporting this feature, as long as I remember), so while printing in test mode, the extruder still makes the retract/recover during all the print if the firmware and the Gcode are FW_Retract-enabled

It's not really a problem if filament is unloaded (as recommanded) but still useless, avoiding them would slightly speed up the test-print time and will reduce unneeded  wear on geared extruder

#### Version of Octolapse
Octolapse Version: 0.3.1

#### Version of OctoPrint
OctoPrint Version: current dev ( 9a881012e200da3e071ca41307459cfbb987fb50 )

FormerLurker, 2018-05-16:
I'm still learning about gcode and have never seen these before, so thanks for posting the issue!  It's not too difficult to add this, so I will do so!  Hopefully I'll be able to work this in soon.  I'll update this ticket once this is finished.

FormerLurker, 2018-05-16:
Ug, G10 is also used for tool offsets on some firmware.  G11 won't be a problem, but I'll need to handle g10 as a special case.  It looks to me like G10 without a P parameter is a retract according to [this](http://reprap.org/wiki/G-code#G10:_Tool_Offset).

I really wish there was a gcode standard for 3d printers!

gege2b, 2018-05-16:
[it's tricky](https://www.xkcd.com/927/) :rofl: 

agree on the G10, looking if the Pxxx is present should be enough to decide if the gcode can be stripped

FormerLurker, 2018-08-11:
I'm closing this out since I think it works.  Let me know if I should reopen it!

closed 2018-08-11 by FormerLurker