FormerLurker/Octolapse #168: Issues with Octolapse on Prusa i3 Mk3

edasque, 2018-05-23:
#### Version of Octolapse
 
Octolapse Version: 0.3.1

#### Version of OctoPrint

OctoPrint Version: 1.3.8

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: Yes

#### What were you doing when the problem occurred

1. Configured Octolapse with Prusa i3 Mk3 , Fixed - Extruder at back left - Layer Change - Mk3 - RPI camera snapshot URL
2. Started the print

#### What should have happened?
Once the print started, Octolapse should have made a timelapse without errors

#### What happened instead?
A lot of error messages appeared on the screen, one after the other, continuously. I had to shut off Octolapse. The message was a variation of:
```
Octolapse - Position Error
Position - Coordinates (X:148.40400,Y:96.84300,Z:0.15000,E:115.76783) are out of the printer area! Cannot resume position tracking until the axis is homed, or until absolute coordinates are received.
```

#### Operating System running OctoPrint and Octolapse

OS Name: Raspian on RPI3B
Os Version: Jessie

#### Printer model & used firmware incl. version

Printer Model: Prusa Mk3
Printer Firmware Version: 3.2.0

#### Browser and version of browser, operating system running browser

Browser: Chrome Mac
Browser OS: 66.0.3359.181

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.  I'm currently saving for some additional test cameras in preparation for develiping multi-camera support.

Passaita, 2018-05-23:
Second this.
Exactly the same error and as a result an attempt to uninstall was unsuccessful will result in another issue.

ninjamojo, 2018-05-23:
The answer to this issue is to configure your print volume within Octoprint (printer profiles) OR you can configure it within Octolapse.  I will be adding these volumes to the default settings soon, so sorry about having to deal with it until then.

Let's fix this within Octolapse instead of Octoprint for now.  Take a look at the printer profile within your Octolapse settings and find the Printer Volume section and enable Override Octoprint Printer Profile.  Now I'm not 100% sure what the print volume for the MK3 is, but I think the minX and minY are slightly negative.  You can probably find these values in your start gcode in Slic3r (search for X- and Y-).  Please double check these values, but I think they are close:
25 x 21 x 21
MinX = -3 (??)
MaxX = 250
MinY = -3 (??)
MaxY = 210 
MinZ = 0
MaxZ = 210 (Might be a bit less than this)

These values are used for safety since not all printers restrict movements to in-bounds positions.  However I think that the MK3 DOES have protection against this.  Also, please let me know what your final values are and I'll update the default profile.

I hope that helps!

Naphier, 2018-08-10:
In my end gcode I tell the bed to move out of the way with:
`G1 Z+0.5 E-5 X-20 Y-20 F9000 ;move Z up a bit and retract filament even more`
This would cause a similar octolapse error. 
I'm using the Wanhao Duplicator i3 (I think v2.1 - it's the Monoprice Maker Select v2).
Bed size is X: 200, Y: 200, Z: 180
Cura has "Printhead settings"
```
X min 20
Y min 10
X max 10
Y max 10
Gantry 55
```

I tried inputting the following into Octolapse's "Override Octoprint Printer Profile":
```
X - Width   Min -20    Max 200
Y - Depth   Min -20    Max 200
Z - Height  Min 0      Max 180
```

But... I think this is not right because Octolapse wouldn't run at all with these settings. No failure error, didn't even look like it was trying to run.

So I turned off "Abort Snapshot When Out-Of-Bounds" and everything seems just fine. What might be nice is the capability to send a command to Octolapse via the gcode to tell Octolapse that the print is done. Something like "if octolapse sees `G1 Z+0.5 E-5 X-20 Y-20 F9000` then no need to take any more snapshots, print is done.

Naphier, 2018-08-31:
I just removed that from my end gcode, it wasn't that helpful and I can just as easily run a post-print process. I wasn't getting repeated errors, just one, All seems to be OK now, but any repositioning after print was causing this issue even with the bounds check disabled. I had a pre-print wipe method going on too and had to remove that since I force out of the print area just a little bit.

FormerLurker, 2018-09-10:
I'm going to close this issue since it sounds like you got things working.  If you want me to re-open it just reply and I will do so.

Thanks for posting!

closed 2018-09-10 by FormerLurker