FormerLurker/Octolapse #169: [Request] Monoprice Select Mini v2 Profile

TheKrush, 2018-05-23:
#### If this is a feature request describe it here

Please create a profile for the Monoprice Select Mini v2.

#### Version of Octolapse
 
Octolapse Version: 0.3.1

#### Version of OctoPrint

OctoPrint Version: 1.3.8

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES_OR_NO

#### What were you doing when the problem occurred

1. STEP_ONE_GOES_HERE
2. STEP_TWO_GOES_HERE
3. STEP_...

#### What should have happened?
PUT_YOUR_DESCRIPTION_HERE

#### What happened instead?
PUT_YOUR_DESCRIPTION_HERE

#### Operating System running OctoPrint and Octolapse

OS Name: OS_NAME_GOES_HERE
Os Version: OS_VERSION_GOES_HERE

#### Printer model & used firmware incl. version

Printer Model: PRINTER_MODEL_GOES_HERE
Printer Firmware Version: PRINTER_FIRMWARE_VERSION_GOES_HERE

#### Browser and version of browser, operating system running browser

Browser: BROWSER_VERSION_GOES_HERE
Browser OS: BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.  I'm currently saving for some additional test cameras in preparation for develiping multi-camera support.

FormerLurker, 2018-05-23:
Thanks for your request!  I may have a profile for this printer, and will
check as soon as I have a bit of time (busy season).  If not I can help you
figure out the seetings, but I'll need a few things.  Let me check on an
existing profile first.

D3rpi, 2018-05-26:
I just installed the plugin and would also appreciate a Monoprice Select Mini v2 profile.

TheKrush, 2018-05-29:
[Spider.stl.zip](https://github.com/FormerLurker/Octolapse/files/2051577/Spider.stl.zip)

Here's a short ~18 minute print. I was actually using this when I first installed the plugin (using the Monoprice maker profile) it worked mostly except there was a lot of excess filament ending up on the side of the model where it kept moving the nozzle out of the way for the picture.

D3rpi, 2018-05-30:
[50mm tower.zip](https://github.com/FormerLurker/Octolapse/files/2051900/50mm.tower.zip)
Gcode for a 23 mins print(as per CURA) of the 50mm tower. 
Actual Print time on my Mini Select V2: 34mins

rgelb, 2018-06-13:
Checking in as well.  Would love a profile!!!

ninjamojo, 2018-06-13:
Sorry for the delays getting this profile implemented!  @rgelb (and everyone on the thread), I should be back in business shortly after the 1st of July, and have a long todo list worked up.  This one shouldn't take too long since I have so many willing testers :)  If you want to keep me on my toes, check back sometime after the 1st.

rgelb, 2018-07-20:
Any progress?  Or a roadmap?

FormerLurker, 2018-07-22:
@TheKrush, I've looked through your gcode and have come up with the following:

Issues with spider.gcode - There is no command to home the printer (G28), so Octolapse can't work with this gcode right now.  You need to add the home command to your start gcode, but i'm assuming there is a reason you don't.

Retraction Length: 4

Z-Hop Height: 0.0

Axis Speed Display Units: Millimeters per Minute

Movement Speed (travel): 4500

Retraction Speed:2400

Detraction Speed:2400

ZMovement Speed: 2812

X/Y/Z Axis Mode - I'm not sure about this one.  There is no G90 in your start gcode (there should be, at the very top).  If you put the G90 in you can leave this to 'Require Explicit', otherwise set to 'Default to Absolute'

E Axis Mode - Require Explicit - However, I see that you have an M82 and then an M83 immediately following.  Is this really the standard start gcode for your slicer profile?  The M82 is unnecessary and leads me to believe some changes may have been made.

G90/G91 Influences Extruder - No idea, but leave it as 'Use Octoprint Settings' for the time being.  You could look this up or do some tests by sending some manual gcode to the printer to figure this out.  Let me know.

Auto-Detect Position - Checked
Additional Location Detection Commands - Leave this empty

Override Octoprint Printer Profile: Checked

xmin 0
xmax 120
ymin 0
ymax 120
zmin 0
zmax 120

Abort Snapshot when Out-Of-Bounds - checked
Printer Position Tolerance 0.001
Snapshot Command: snap
Priming Height: 0.75

If these are available (depends if you are using the latest release candidate)
Use Retraction/Detraction settings for Firmware Retract/Detract: unchecked

Use Z-Hop (Z-Lift) settings for Firmware Retract/Detract: unchecked

After you add a home command and hopefully G90 (I'll look over it if you want), try these settings in your profile and let me know.

@D3rpi, I'll try to look through your code tonight if I can scrape together some time :)

@rgelb, I'm back in business and working on two printer profiles, but have exhausted most of my time this weekend by fixing some other issues and getting external DSLR cameras working in Octolapse.  If it's ready, this profile should be in the next release.  I need some feedback on the latest release candidate ([see and read this issue for details](https://github.com/FormerLurker/Octolapse/issues/198)) before I can drop the new final version.  Tell your friends :)

TheKrush, 2018-07-23:
I just use Cura to create my gcode. I haven't ever directly edited the gcode.

TheKrush, 2018-12-06:
Here are all the settings I used when setting up  Ultimaker Cura. Most of these came from the official profiles with magic numbers specifically for MPSM printers.

[Cura.zip](https://github.com/FormerLurker/Octolapse/files/2651834/Cura.zip)

For that spider I was using the `MPSM-D&DFigurine.curaprofile` profile. But they are all mostly the same.

Other than those here's the machine settings I have in Cura.

[image]

I also have this start GCODE setup in Octoprint
```
G21 ; metric values
M140 S50 ; set bed temperature to 50
M117 Returning to home position... ; Put message on LCD screen
G90 ; absolute positioning
M82 ; set extruder to absolute mode
G28 X0 Y0 ; move X/Y to min endstops
G28 Z0 ; move Z to min endstops
G1 Y60.0 Z15.0 F9000 ; move the platform down 15mm
M117 Heating Bed... ; Put message on LCD screen
M190 S50 ; wait for bed temperature to 50
M104 S175 ; set hotend temperature to 175
M117 Heating Hotend... ; Put message on LCD screen
M109 S175 ; wait for hotend temperature to 175
M106 P0 S255 ; turn on fan
G92 E0 ; zero the extruded length
G1 F200 E5 ; extrude 5mm of feed stock
G92 E0 ; zero the extruded length again
G1 F9000 ; Set feedrate to 9000mm/m
M117 Printing... ; Put printing message on LCD screen
```

FormerLurker, 2018-12-06:
@TheKrush,

Have you upgraded Octolapse to v0.3.4?  You will be pleased to know that you can now select your slicer type within the Octolapse printer profiles, which makes entering your settings easier.  You will also have to configure your build volume to match your printer by selecting the 'Override Octoprint Printer Profile', entering 0 for all of the min values, and entering the volume above for your max values.  This info appears to be contained within your machine settings (the .jpg you posted).

I might have some time this weekend to create a profile myself from your cura settings, but if possible try to create one using the new plugin yourself first.  Otherwise I'll send you a settings.json file.  Please let me know either way, and thanks for posting all of that data!  I'll definitely include the profile within the defaults once it is perfected in any case!

D3rpi, 2019-03-24:
I'm playing around with the plugin this weekend. Edited the Maker Select V2/Wanhao i3 beta profile to match my mini v2's bed size(under the print volume), X/Y/Z mode to absolute and it worked great. 
Hope that helps.

FormerLurker, 2019-03-24:
@3drpi, once you have used your profile for a while, would you mind exporting your settings.json and posting it somewhere for me?  That way i can copy the profile and add it to the defaults.  Thanks!

D3rpi, 2019-03-24:
@FormerLurker Sure, I will post them here.

D3rpi, 2019-03-24:
@FormerLurker Here you go
[mpsmv2.zip](https://github.com/FormerLurker/Octolapse/files/3000863/mpsmv2.zip)

Also uploaded a test I did on google photos with my PS3 camera. It's probably not the best lapse you've seen!
https://photos.app.goo.gl/ugjZm82awxPNWT6e7

FormerLurker, 2019-03-25:
For a PS3 camera that's pretty good!  Increase your light, that should help.

Thank you so much for the profile!

FormerLurker, 2019-10-12:
So, this profile has been added to the new Octolapse profile repository that will be available starting with V0.4, so I'm considering this closed!  Thanks for posting and participating!

closed 2019-10-12 by FormerLurker