FormerLurker/Octolapse #170: Unable to uninstall

Passaita, 2018-05-23:
An attempt to uninstall Octolapse plugin populated progress window and then prompted to system reboot that has been performed. Once reconnected to the system Octolapse header is still there and plugin is in the list of installed modules but now an attempt to uninstall it again results in:
`Uninstalling plugin "Octolapse"
/home/pi/oprint/bin/python2 -m pip uninstall --yes octoprint_octolapse
Skipping octoprint-octolapse as it is not installed.
Done!`

FormerLurker, 2018-05-23:
Hmm.  Maybe it is cached in your browser?  Can you try another browser or clear your cache?

If that doesn't work we will have to dig a bit deeper.

FormerLurker, 2018-05-23:
Hmm.  Maybe it is cached in your browser (probably not...)?  Can you try
another browser or clear your cache?

If that doesn't work we will have to dig a bit deeper.

oderwat, 2018-05-24:
I couldn't install Octolapse on my RasPi because pillow failed to compile / install. This was caused by a missing packaged: libjpeg8-dev .. sudo apt-get install libjpeg8-dev

FormerLurker, 2018-05-24:
@oderwat , thanks for mentioning that.  I think I missed a prerequisite there.  Adding it to the list.

Nuttavoot, 2018-06-10:
I cannot install Octolapse as well, it also get the error "Failed building wheel for Pillow".

FormerLurker, 2018-06-13:
@Nuttavoot, your issue is generally caused by low resources.  If you are running octopi, you can install pillow separately from the command line like so:

pi@octopi:~/oprint/local/bin $ ./pip --no-cache-dir install pillow

Check out [this closed issue](https://github.com/FormerLurker/Octolapse/issues/107) for more details.  Let me know if you have any problems, preferably on the issue I linked to above.  I will reopen the linked issue if necessary.

FYI, a pull request was made [here](https://github.com/foosel/OctoPrint/pull/2535) to address this issue inside of OctoPrint itself, but it's a somewhat rare thing.  I will put this back on my radar once I get a chance to do more Octolapse.

dancorrigan1, 2018-06-26:
Not sure if this relates completely, but mine just errored about pip having an upgrade.. so I upgraded it, re-attempted the install via Octoprint plugin manager, and bingo, installed.

I ran:   ~/oprint/bin/pip install --upgrade pip

Nuttavoot, 2018-07-13:
@FormerLurker I am not using Octopi, I install Octoprint from scratch into OrangePi and this command "~/oprint/local/bin $ ./pip --no-cache-dir install pillow" does not run on mine.

FormerLurker, 2018-07-13:
@Nuttavoot, can you send me the console output for the command:

`~/oprint/local/bin $ ./pip --no-cache-dir install pillow`

Are you sure that pip is in `~/oprint/local/bin`?  On my instance it's at `~/oprint/bin`, but since you installed manually it might not be there I suppose.

Also, you might want to try upgrading pip if you haven't already via 

`~/oprint/bin/pip install --upgrade pip`

or maybe here

`~/oprint/local/bin/pip install --upgrade pip`

depending on where octoprint is installed.

Nuttavoot, 2018-07-15:
@FormerLurker Here is the console output:

login as: pi
pi@192.168.0.110's password:
  ___                               ____  _   _     _ _
 / _ \ _ __ __ _ _ __   __ _  ___  |  _ \(_) | |   (_) |_ ___
| | | | '__/ _` | '_ \ / _` |/ _ \ | |_) | | | |   | | __/ _ \
| |_| | | | (_| | | | | (_| |  __/ |  __/| | | |___| | ||  __/
 \___/|_|  \__,_|_| |_|\__, |\___| |_|   |_| |_____|_|\__\___|
                       |___/

Welcome to ARMBIAN 5.38 stable Debian GNU/Linux 9 (stretch) 4.14.18-sunxi
System load:   0.97 0.82 0.80   Up time:       2:42 hours
Memory usage:  20 % of 493MB    IP:            192.168.0.110
CPU temp:      49°C
Usage of /:    12% of 15G

[ 0 security updates available, 1 updates total: apt upgrade ]
Last check: 2018-07-14 23:48

Last login: Fri Jul 13 17:58:17 2018 from 192.168.0.120

pi@orangepilite:~$ ~/oprint/local/bin $ ./pip --no-cache-dir install pillow
-bash: /home/pi/oprint/local/bin: No such file or directory
pi@orangepilite:~$

Nuttavoot, 2018-07-15:
@FormerLurker This is the command I installed OctoPrint:

-	Install OctoPrint (Pyton 2.7 allredy installed by default )
cd ~
sudo apt-get install python-pip python-dev python-setuptools python-virtualenv git libyaml-dev build-essential
git clone https://github.com/foosel/OctoPrint.git
cd OctoPrint
python -m virtualenv venv
./venv/bin/pip install pip --upgrade
./venv/bin/python setup.py install
mkdir ~/.octoprint

Nuttavoot, 2018-07-15:
@FormerLurker I have managed to run Pillow installation command but still getting these error:

pi@orangepilite:~/OctoPrint/venv/local/bin$ ./pip --no-cache-dir install pillow
Collecting pillow
  Downloading https://files.pythonhosted.org/packages/d3/c4/b45b9c0d549f482dd072055e2d3ced88f3b977f7b87c7a990228b20e7da1/Pillow-5.2.0.tar.gz (14.5MB)
    100% |████████████████████████████████| 14.5MB 1.2MB/s
Installing collected packages: pillow
  Running setup.py install for pillow ... error
    Complete output from command /home/pi/OctoPrint/venv/bin/python2 -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-5fHgcd/pillow/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-record-8NM0Kr/install-record.txt --single-version-externally-managed --compile --install-headers /home/pi/OctoPrint/venv/include/site/python2.7/pillow:
    running install
    running build
    running build_py
    creating build
    creating build/lib.linux-armv7l-2.7
    creating build/lib.linux-armv7l-2.7/PIL
    copying src/PIL/FtexImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
    copying src/PIL/PpmImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
    copying src/PIL/ImageFont.py -> build/lib.linux-armv7l-2.7/PIL
    copying src/PIL/JpegImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
    copying src/PIL/PsdImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
    copying src/PIL/ImImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
    copying src/PIL/SgiImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
    copying src/PIL/XVThumbImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
    copying src/PIL/Image.py -> build/lib.linux-armv7l-2.7/PIL
    copying src/PIL/_binary.py -> build/lib.linux-armv7l-2.7/PIL
    copying src/PIL/PyAccess.py -> build/lib.linux-armv7l-2.7/PIL
    copying src/PIL/ImageSequence.py -> build/lib.linux-armv7l-2.7/PIL
    copying src/PIL/DdsImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
    copying src/PIL/OleFileIO.py -> build/lib.linux-armv7l-2.7/PIL
    copying src/PIL/XpmImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
    copying src/PIL/McIdasImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
    copying src/PIL/IcnsImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
    copying src/PIL/ImageTransform.py -> build/lib.linux-armv7l-2.7/PIL
    copying src/PIL/BmpImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
    copying src/PIL/ImagePalette.py -> build/lib.linux-armv7l-2.7/PIL
    copying src/PIL/IptcImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
    copying src/PIL/TiffImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
    copying src/PIL/GdImageFile.py -> build/lib.linux-armv7l-2.7/PIL
    copying src/PIL/_util.py -> build/lib.linux-armv7l-2.7/PIL
    copying src/PIL/CurImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
    copying src/PIL/TiffTags.py -> build/lib.linux-armv7l-2.7/PIL
    copying src/PIL/FliImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
    copying src/PIL/WebPImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
    copying src/PIL/ImageGrab.py -> build/lib.linux-armv7l-2.7/PIL
    copying src/PIL/WmfImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
    copying src/PIL/ImageEnhance.py -> build/lib.linux-armv7l-2.7/PIL
    copying src/PIL/ImageShow.py -> build/lib.linux-armv7l-2.7/PIL
    copying src/PIL/PdfImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
    copying src/PIL/ImageOps.py -> build/lib.linux-armv7l-2.7/PIL
    copying src/PIL/GbrImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
    copying src/PIL/ImageTk.py -> build/lib.linux-armv7l-2.7/PIL
    copying src/PIL/BlpImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
    copying src/PIL/PcfFontFile.py -> build/lib.linux-armv7l-2.7/PIL
    copying src/PIL/WalImageFile.py -> build/lib.linux-armv7l-2.7/PIL
    copying src/PIL/Jpeg2KImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
    copying src/PIL/ImageMorph.py -> build/lib.linux-armv7l-2.7/PIL
    copying src/PIL/TgaImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
    copying src/PIL/EpsImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
    copying src/PIL/_tkinter_finder.py -> build/lib.linux-armv7l-2.7/PIL
    copying src/PIL/IcoImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
    copying src/PIL/ImageWin.py -> build/lib.linux-armv7l-2.7/PIL
    copying src/PIL/GimpGradientFile.py -> build/lib.linux-armv7l-2.7/PIL
    copying src/PIL/ImageDraw.py -> build/lib.linux-armv7l-2.7/PIL
    copying src/PIL/features.py -> build/lib.linux-armv7l-2.7/PIL
    copying src/PIL/Hdf5StubImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
    copying src/PIL/JpegPresets.py -> build/lib.linux-armv7l-2.7/PIL
    copying src/PIL/__init__.py -> build/lib.linux-armv7l-2.7/PIL
    copying src/PIL/DcxImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
    copying src/PIL/ImagePath.py -> build/lib.linux-armv7l-2.7/PIL
    copying src/PIL/ImageDraw2.py -> build/lib.linux-armv7l-2.7/PIL
[... 55 lines cut ...]
    warning: no previously-included files matching '.git*' found anywhere in distribution
    warning: no previously-included files matching '*.pyc' found anywhere in distribution
    warning: no previously-included files matching '*.so' found anywhere in distribution
    writing manifest file 'src/Pillow.egg-info/SOURCES.txt'
    running build_ext

    The headers or library files could not be found for zlib,
    a required dependency when compiling Pillow from source.

[... cut ...]
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-install-5fHgcd/pillow/setup.py", line 807, in <module>
        raise RequiredDependencyException(msg)
    __main__.RequiredDependencyException:

    The headers or library files could not be found for zlib,
    a required dependency when compiling Pillow from source.

    Please see the install instructions at:
       https://pillow.readthedocs.io/en/latest/installation.html

    ----------------------------------------

FormerLurker, 2018-07-15:
It looks like either I'm missing some requirement in my setup.py, or pillow
is.  Maybe this relates to orange pi?  I'm not a Linux expert at all, but
I'll research and see what I can come up with.

FormerLurker, 2018-07-15:
It looks like there are two required libraries that may be causing you problems.  Try this:

sudo apt-get install libjpeg-dev

then try installing pillow?  I fear this may be related to zlib, but haven't figured out how to install that one yet.

Let me know if the above helped.

FormerLurker, 2018-07-15:
Actually, this command might install everything, try it first:

sudo apt-get install libjpeg-dev libfreetype6 libfreetype6-dev zlib1g-dev

Nuttavoot, 2018-07-16:
I used these command to install my webcam, would above commands would cause issues ?

-	Webcam
cd ~
sudo apt-get install subversion libjpeg62-turbo-dev imagemagick ffmpeg libv4l-dev cmake
git clone https://github.com/jacksonliam/mjpg-streamer.git
cd mjpg-streamer/mjpg-streamer-experimental
export LD_LIBRARY_PATH=.
make

FormerLurker, 2018-07-16:
@Nuttavoot, it's possible.  Pillow requires libjpeg, and perhaps there is something incompatible about the version you installed?  However, the error you posted indicates a problem with zlib.

I found this install command [here](http://pillow.readthedocs.io/en/4.0.x/installation.html#linux-installation) under the Raspian section:

`sudo apt-get install libtiff4-dev libjpeg8-dev zlib1g-dev libfreetype6-dev liblcms2-dev libwebp-dev tcl8.5-dev tk8.5-dev python-tk`

Can you try that and post the console log if it fails?

Nuttavoot, 2018-07-17:
Please be reminded that I am using Armbian not Rasbian, this is the console log:

Welcome to ARMBIAN 5.38 stable Debian GNU/Linux 9 (stretch) 4.14.18-sunxi

pi@orangepilite:~$ sudo apt-get install libtiff4-dev libjpeg8-dev zlib1g-dev libfreetype6-dev liblcms2-dev libwebp-dev tcl8.5-dev tk8.5-dev python-tk
Reading package lists... Done
Building dependency tree
Reading state information... Done
Package libjpeg8-dev is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source
However the following packages replace it:
  libjpeg62-turbo-dev

Package libtiff4-dev is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source
However the following packages replace it:
  libtiff5-dev

E: Package 'libtiff4-dev' has no installation candidate
E: Package 'libjpeg8-dev' has no installation candidate
pi@orangepilite:~$

Nuttavoot, 2018-07-17:
This commands installed with no error,

sudo apt-get install libjpeg-dev libfreetype6 libfreetype6-dev zlib1g-dev

FormerLurker, 2019-10-12:
OK, this is an old issue now, but I've figured out that most pillow installation/uninstallation issues are actually related to PSU!  I know it's hard to believe, but there are now many examples of this.  I'm going to go ahead and close this issue, but if you have any more issues/questions, feel free to open another thread, or I can reopen this one.  Just let me know, and thanks for posting!!!!

closed 2019-10-12 by FormerLurker