FormerLurker/Octolapse #171: Temperature not being set by Octoprint when Octolapse enabled.

bluemeane, 2018-05-26:
#### Version of Octolapse
 
Octolapse Version: 0.3.1

#### Version of OctoPrint

OctoPrint Version: 1.3.8

Diagnostic Logging was Enabled: YES_OR_NO

#### What were you doing when the problem occurred
1. I loaded up  print, and started it.
2. The printer immediately goes into bed leveling and then starts to print with no command being sent to heat up the bed or extruder first even though it is in the gcode.

#### What should have happened?

When I hit print it should send the normal commands at the beginning of the gcode file and preheat the bed, and extruder before the bed leveling, or the printing begins.

#### What happened instead?

Whatever the plugin is injecting to the code is superseding or overwriting/replacing the first lines of the gcode.

#### Operating System running OctoPrint and Octolapse

-->
OS Name: OctoPi
Os Version: 0.15.0PE

#### Printer model & used firmware incl. version

Printer Model: Prusa MK3
Printer Firmware Version: 3.1.3

#### Link to the gcode file you were printing when the problem occurred

Happens with all my gcode files.

FormerLurker, 2018-05-29:
Before you try anything else, can you make sure you are not using test mode (debug settings on the main tab).  If you are, just change the debug profile to 'Live Print' and try again, then let me know if that worked.

Thanks!

FormerLurker, 2018-09-10:
I'm going to close this issue out since it's been open for a while with no response.  If you want me to re-open it, just reply and I will do so.

Thanks for posting!

closed 2018-09-10 by FormerLurker

Zylvian, 2019-01-02:
@FormerLurker I too experienced this problem today, multiple times.

Happened on my Ender 3, first time using Octolapse. I got an error message about having to do G91 or G92, as well as a Octolapse message. The printer then started printing at 18 degrees. Problem stopped when I disabled Octolapse.

It's possible that Octolapse is overriding a few lines of code, as @bluemeane suggested.

FormerLurker, 2019-01-02:
@Zylvian, do you have 'test mode' enabled within the Octolapse settings (debug profile)?