FormerLurker/Octolapse #172: [Request] Supply custom ffmpeg options

DiddiZ, 2018-05-27:
#### If this is a feature request describe it here

Being able to directly supply command line options to ffmpeg would allow for a finer control of advanced ffmpeg settings (e.g. codec, preset, profile). This could be done through an addition textbox under `Advanced Rendering Options `.

Especially I want to be able to use h264 codec instead of mpeg4.

#### Version of Octolapse
 
Octolapse Version: 0.3.1

#### Version of OctoPrint

OctoPrint Version: 1.3.8

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES_OR_NO

#### What were you doing when the problem occurred

1. STEP_ONE_GOES_HERE
2. STEP_TWO_GOES_HERE
3. STEP_...

#### What should have happened?
PUT_YOUR_DESCRIPTION_HERE

#### What happened instead?
PUT_YOUR_DESCRIPTION_HERE

#### Operating System running OctoPrint and Octolapse

OS Name: OS_NAME_GOES_HERE
Os Version: OS_VERSION_GOES_HERE

#### Printer model & used firmware incl. version

Printer Model: PRINTER_MODEL_GOES_HERE
Printer Firmware Version: PRINTER_FIRMWARE_VERSION_GOES_HERE

#### Browser and version of browser, operating system running browser

Browser: BROWSER_VERSION_GOES_HERE
Browser OS: BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.  I'm currently saving for some additional test cameras in preparation for develiping multi-camera support.

FormerLurker, 2018-07-09:
Thanks for the great recommendation, and sorry that it's taken me so long to respond (been away).  I'll look into this the next time I work on the rendering module.

Shadowen, 2018-07-22:
I'll see if I can get this one done relatively quickly. I've been in the render code before.

Shadowen, 2018-07-24:
@DiddiZ Please confirm this issue can be closed now :)

Shadowen, 2018-07-24:
I'm not seeing a high demand for custom input to ffmpeg. I guess if you really wanted to, you could turn off rendering and tell Octolapse to leave the snapshots behind, then just run ffmpeg manually?

It seems like if you need that much power, you don't really need Octolapse's automation :/ I'm going to need a more specific use case to add such a feature.

DiddiZ, 2018-07-24:
I could render manually, but I'm a huge fan of Octolapse's automation. It's really handy to download the rendered video from Octoprint without any additional click.

To tell you about my use-case: I use the timelapses to review in case of failed prints, and to share them with friends via WhatsApp. WhatsApp allows only .mp4 containers with a h264 codec. Before now, I reencoded the timelapses on my PC, accepting the quality loss, which wasn't great.

FormerLurker, 2018-07-24:
Hey, @DiddiZ, will you look at this issue and maybe you can share your images with me somehow?  

https://github.com/FormerLurker/Octolapse/issues/196

I'm sharing these with another group, and will be using them to train an algorithm for detecting failed prints/bad layers and it's really hard to gather this data.  It would help a lot.  You can forget about the file naming conventions, I'd be willing to inspect and rename manually.

Larswad, 2023-05-22:
I can give one reason though that I think might be valid for this request.
Not running on a pi, I have my own regular PC server nearby so it was a much more powerful choice, besides, no need to run more servers than neccessary.
However, my ffmpeg under kubuntu is kinda flawed since the LD_LIBRARY_PATH points out libdrm.so (needed by ffmpeg) to a path in the amdgpu drivers in /opt (for some strange unknown reason). Not daring to change that systemwide, I could add to to the ffmpeg line in front of the command:
LD_LIBRARY_PATH=/usr/lib/x86_64-linux-gnu 

I think otherwise I agree, octoprint gives all the flexibility that's needed.

PR #200 (2018-07-23): Added H.264 as a codec choice.
Not sure why it shows so many commits, but only the last three should be relevant.

Changed the way rendering codecs and extensions are selected based on output_format.
Added H.264 as a codec choice. Fixes #172.