FormerLurker/Octolapse #173: [BUG?] Cam image is flipped!

bolsoncerrado, 2018-05-30:
#### If this is a feature request describe it here

FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: 0.3.1

#### Version of OctoPrint

OctoPrint Version: 1.3.8

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: no

#### What were you doing when the problem occurred

1. Creating a timelapse by starting a print

#### What should have happened?
Create a correct timelapse

#### What happened instead?
The cam image (and hence the video) is flipped.

#### Operating System running OctoPrint and Octolapse

OS Name: Raspbian
Os Version: Jesse Lite

#### Printer model & used firmware incl. version

Printer Model: PRINTER_MODEL_GOES_HERE
Printer Firmware Version: PRINTER_FIRMWARE_VERSION_GOES_HERE

#### Browser and version of browser, operating system running browser

Browser: BROWSER_VERSION_GOES_HERE
Browser OS: BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  octoprint.log

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: https://i.imgur.com/aaJBHIG.png

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.  I'm currently saving for some additional test cameras in preparation for develiping multi-camera support.

bolsoncerrado, 2018-05-30:
Just for the record, YES, i have both X and Y camera FLIP checkboxes enabled in Octoprint Camera configuration.

FormerLurker, 2018-05-30:
Octolapse's image orientation is currently completely independent of Octoprint's settings.  I hope to add a 'Use Octoprint Settings' option soon, fyi.

Check out the camera profile settings and look for 'Snapshot Transposition Options'.  Unless I'm mistaken, I believe you want 'Rotate 180 degrees'.

Let me know if that works!

bolsoncerrado, 2018-05-30:
Awesome will do!

Shadowen, 2018-07-23:
Please reopen if issue is still present. Otherwise I will assume it has been fixed.

closed 2018-07-23 by Shadowen