FormerLurker/Octolapse #174: Snapshot: idle until all axes are homed

adrianmihalko, 2018-05-30:
#### Version of Octolapse
 
Octolapse Version: 0.3.1

#### Version of OctoPrint

OctoPrint Version:  1.3.8

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: Test mode

#### What were you doing when the problem occurred

I am trying to use Octolapse with my Creality Ender 3 printer, but it's not working: Octolapse just waiting here: Idle until all axes are homed. I choosed Creality CR-10 profile, because it's pretty same as Ender 3.

[image]
[image]

I tried with live print and test mode too, nothing happens.

#### Operating System running OctoPrint and Octolapse

OctoPrint 1.3.8 running on OctoPi 0.15.0

#### Printer model & used firmware incl. version

Printer Model: Creality Ender 3
Printer Firmware Version: TH3D

FormerLurker, 2018-05-30:
Ok, it's pretty much what I expected (Missing G90).  I'm trying to figure out how to simplify or eliminate this issue.  Here is what I would do:

Open your start gcode in Cura and look for this line:

`M82 ;absolute extrusion mode`

Right after it add this:

`G90 ;absolute xyz mode`

Alternatively you can set the default XYZ axis mode to 'Absolute' within the Octolapse printer profile ([see the troubleshooting wiki here](https://github.com/FormerLurker/Octolapse/wiki/Troubleshooting#for-xyz-choose-one-of-the-following-options)).  I recommend you look through the printer settings and make sure your retraction/detraction speed and retraction length match your slicer EXACTLY.  This is probably the 2nd most common issue I see (right after missing g90/g91).  Might as well take care of that while you're working in your slicer.

Let me know if that helps!

FormerLurker, 2018-05-30:
That's what I like to hear!  Would you do me a favor and post a timelapse to youtube when you have a chance?  I enjoy watching them, which is one reason I wrote the app :)

agibson2, 2018-06-16:
Thank you for this bug post.  I just added G90 to my start gcode in Cura and hope that will fix it for me too.  It was there by default and commands are expected to be in G90  mode in the start code but it was not sent by the start gcode.

DeclanCSL, 2018-12-18:
@FormerLurker ..pls advice

[image]
octolapse is not running...
[image]
[image]

[Love-Birdsgcode.txt](https://github.com/FormerLurker/Octolapse/files/2690576/Love-Birdsgcode.txt)
im using simplify and i have add G90 as what you told us...

DeclanCSL, 2018-12-18:
[image]

DeclanCSL, 2018-12-18:
yeh can edi ..
now i can play edi ..Hurayyy
hahah
Thank you  @FormerLurker for your advice ..

FormerLurker, 2018-12-18:
Hey @DeclanCSL!  Making progress I see!  Can you download and post your settings.json file for me?  You can export from Octoprint settings-> octolapse.  Thanks!

DeclanCSL, 2018-12-18:
@FormerLurker but here cant upload 
didn;t support the file..
what should i do ?

DeclanCSL, 2018-12-18:
@FormerLurker 
{
  "auto_reload_latest_snapshot": true, 
  "current_printer_profile_guid": "7c0eb400-5a3e-45b3-aae9-cc01918a9bb6", 
  "default_printer_profile": {
    "small_perimeter_speed": null, 
    "bridging_speed_multiplier": 100, 
    "perimeter_speed": null, 
    "outline_speed_multiplier": 100, 
    "first_layer_travel_speed": null, 
    "top_solid_infill_speed": null, 
    "top_solid_infill_speed_text": null, 
    "movement_speed": 6000, 
    "solid_infill_speed": null, 
    "ooze_shield_speed": null, 
    "guid": "5d39248f-5e11-4c42-b7f4-810c7acc287e", 
    "priming_height": 0.75, 
    "small_perimeter_speed_multiplier": 100, 
    "has_been_saved_by_user": false, 
    "origin_z": null, 
    "abort_out_of_bounds": true, 
    "min_y": 0.0, 
    "min_z": 0.0, 
    "detract_speed": 3000, 
    "suppress_snapshot_command_always": true, 
    "default_firmware_retractions": false, 
    "speed_tolerance": 0.6, 
    "z_hop": 0.5, 
    "maximum_z_speed": null, 
    "infill_speed": null, 
    "small_perimeter_speed_text": null, 
    "max_z": 0.0, 
    "e_axis_default_mode": "require-explicit", 
    "external_perimeter_speed_multiplier": 100, 
    "max_y": 0.0, 
    "gap_fill_speed": null, 
    "ooze_shield_speed_multiplier": 100, 
    "description": "", 
    "support_structure_speed_multiplier": 100, 
    "bridge_speed": null, 
    "first_layer_speed_multiplier": 100, 
    "solid_infill_speed_text": null, 
    "slicer_type": "other", 
    "top_solid_infill_speed_multiplier": 100, 
    "g90_influences_extruder": "use-octoprint-settings", 
    "external_perimeter_speed_text": null, 
    "solid_infill_speed_multiplier": 100, 
    "origin_y": null, 
    "origin_x": null, 
    "above_raft_speed_multiplier": 100, 
    "min_x": 0.0, 
    "print_speed": null, 
    "first_layer_speed": null, 
    "retract_speed": 4000, 
    "default_firmware_retractions_zhop": false, 
    "axis_speed_display_units": "mm-min", 
    "auto_position_detection_commands": "", 
    "prime_pillar_speed_multiplier": 100, 
    "name": "Default Printer", 
    "retract_length": 2.0, 
    "snapshot_command": "snap", 
    "override_octoprint_print_volume": false, 
    "support_speed": null, 
    "external_perimeter_speed": null, 
    "skirt_brim_speed": null, 
    "first_layer_speed_text": null, 
    "z_hop_speed": 6000, 
    "num_slow_layers": 0, 
    "max_x": 0.0, 
    "prime_pillar_speed": null, 
[... 2314 lines cut ...]
      "name": "Off", 
      "value": "off"
    }, 
    {
      "name": "Blink", 
      "value": "blink"
    }, 
    {
      "name": "Auto", 
      "value": "auto"
    }
  ], 
  "camera_powerline_frequency_options": [
    {
      "name": "50 HZ (Europe, China, India, etc)", 
      "value": "50"
    }, 
    {
      "name": "60 HZ (North/South America, Japan, etc)", 
      "value": "60"
    }
  ], 
  "show_position_changes": true, 
  "current_debug_profile_guid": "3811da8f-182d-44bf-8fcb-95614e4daf1b"
}

DeclanCSL, 2018-12-18:
@FormerLurker 
[新文字文件.txt](https://github.com/FormerLurker/Octolapse/files/2691090/default.txt)

DeclanCSL, 2018-12-18:
[json file.txt](https://github.com/FormerLurker/Octolapse/files/2691096/json.file.txt)

FormerLurker, 2018-12-18:
Thanks!  I'll run it through the debugger over lunch and will get back to you.

DeclanCSL, 2018-12-18:
ok ill try then to change it,,
thank you for your advice...

wiliovi, 2019-07-04:
hola he tenido un problema, al imprimir con octolapse mi impresora a entrado en bucle creo que se debe al G90. Dejo por aqui el gcode. Saludos.

; Ender 3 Custom Start G-code
G28 ; Home all axes
G92 E0 ; Reset Extruder
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X0.1 Y20 Z0.3 F5000.0 ; Move to start position
G1 X0.1 Y200.0 Z0.3 F1500.0 E15 ; Draw the first line
G1 X0.4 Y200.0 Z0.3 F5000.0 ; Move to side a little
G1 X0.4 Y20 Z0.3 F1500.0 E30 ; Draw the second line
G92 E0 ; Reset Extruder
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
; End of custom start GCode
G90 ; absolute xyz