FormerLurker/Octolapse #175: Warning:  A timeout occurred while requesting the current position!

mattyams, 2018-06-01:
#### Version of Octolapse
 
Octolapse Version: Octolapse v0.3.1

#### Version of OctoPrint

OctoPrint Version: OctoPrint 1.3.8 running on OctoPi 0.14.0

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES

#### What were you doing when the problem occurred

1. Loaded GCODE
2. Hit Print

#### What should have happened?
There should have been a Timelapse rendered.

#### What happened instead?
No timelapse was rendered and error "octolapse synchronization failed" was in the top right.

#### Operating System running OctoPrint and Octolapse

OS Name: OctoPi
Os Version: 0.14

#### Printer model & used firmware incl. version

Printer Model: Monoprice Maker Select V2.1
Printer Firmware Version: Marlin 1.1.8

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: https://pastebin.com/7mGMPNCA

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log: https://pastebin.com/A9kY3BLg

#### Link to octoprint.log

Link to octoprint.log:  https://pastebin.com/6M6YNhWD

FormerLurker, 2018-07-09:
Your issue may be solved in the newest development build of octolapse, which can be installed from this url:  https://github.com/FormerLurker/Octolapse/archive/devel.zip

Let me know if you have problems, or if this fixes the issue.

Also, sorry for the delay!  I've been away for a good while and am slowly working through the accumulated issues.

FormerLurker, 2018-07-09:
This issue may be related to #149.  Also, if you run a print with Octolapse, maybe you could use the 'Full Diagnostic - Live Print' debug profile and post your full log?  That will help with diagnosis if you run into problems.  Thanks!

FormerLurker, 2018-09-10:
This issue has been open for a while, so I'm going to go ahead and close it out.  If you want me to re-open it, just respond and I'll do so.

Thanks for posting!

closed 2018-09-10 by FormerLurker