FormerLurker/Octolapse #176: Error: invalid literal for int() with base 10

c-hri-s, 2018-06-07:
#### If this is a feature request describe it here

FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: 0.3.1

#### Version of OctoPrint

OctoPrint Version: 1.3.8

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES

#### What were you doing when the problem occurred

1. Start print from local gcode
2. Receive error: Octolapse Timelapse Stopped
An error was thrown by the gcode parser, stopping timelapse. Details: invalid literal for int() with base 10

#### What should have happened?
Octolapse shouldn't have crashed

#### What happened instead?
It did

#### Operating System running OctoPrint and Octolapse

OS Name: OctoPi
Os Version: 1.3.8

#### Printer model & used firmware incl. version

Printer Model: Original Prusa MK3
Printer Firmware Version: 3.2.1

#### Browser and version of browser, operating system running browser

Browser: BROWSER_VERSION_GOES_HERE
Browser OS: BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: https://gist.github.com/c-hri-s/d3e0ff2af4c11a7b9d3d02f97403c079

#### Link to settings.json

Link to settings.json with all passwords removed: SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  https://gist.github.com/c-hri-s/2af055427db951963ec2edddb0c823c6

#### Link to octoprint.log

Link to octoprint.log:  https://gist.github.com/c-hri-s/26503dd600ed45d71c2bff19b01b750b

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.  I'm currently saving for some additional test cameras in preparation for develiping multi-camera support.

FormerLurker, 2018-06-07:
Thanks for submitting your issue!

In looking at your gcode, there are some non-comments on blank lines, which is not allowed.  See these lines towards the beginning of your code:

```
M201 X1000 Y1000 Z200 E5000 ; sets maximum accelerations
 mm/sec^2
M203 X200 Y200 Z12 E120 ; sets maximum feedrates
 mm/sec
M204 S1250 T1250 ; sets acceleration (S) and retract acceleration (T)
M205 X10 Y10 Z0.4 E2.5 ; sets the jerk limits
 mm/sec
M205 S0 T0 ; sets the minimum extruding and travel feed rate
 mm/sec
```
Try changing them like so:

```
M201 X1000 Y1000 Z200 E5000 ; sets maximum accelerations mm/sec^2
M203 X200 Y200 Z12 E120 ; sets maximum feedrates mm/sec
M204 S1250 T1250 ; sets acceleration (S) and retract acceleration (T)
M205 X10 Y10 Z0.4 E2.5 ; sets the jerk limits mm/sec
M205 S0 T0 ; sets the minimum extruding and travel feed rate mm/sec
```

I'm guessing both Octoprint and your printer just ignore invalid gcode like this, but for better or worse Octolapse does not.  Please let me know if this fixes your issue or if you continue to have problems, and thanks for filling out the issue form fully!  That made the issue really easy to spot.

Also, there is a bug in the live version of Octolapse that might cause a problem with the following lines of your start gcode:

```
G28 W ; home all without mesh bed level
G80 ; mesh bed leveling
M104 S[first_layer_temperature] ; set extruder temp
M109 S[first_layer_temperature] ; wait for extruder temp
G1 Y-3.0 F1000.0 ; go outside print area
G92 E0.0
```
If you have any trouble where your extruder isn't heating up properly after the mesh bed leveling, change the above code to the following:

```
G28 W ; home all without mesh bed level
G80 ; mesh bed leveling
G92 E0.0 ; Add this line to deal with Octolapse bug
M104 S[first_layer_temperature] ; set extruder temp
M109 S[first_layer_temperature] ; wait for extruder temp
G1 Y-3.0 F1000.0 ; go outside print area
```

Fyi, this bug is fixed, but has not been released yet (busy season for me).  It happens when a command that Octolapse can't parse or doesn't care about comes after a command that requires position detection (both G28 and G80).  Let me know if you have any trouble with this and maybe I can give you a new install url that will install a sneak peak at the newest code, though it's not yet tested as much as I'd like.

c-hri-s, 2018-06-07:
Thanks for the info - I uninstalled Octolapse as I was hitting the issue with the extruder heating you mentioned.

I've regenerated the gcode, I'll reinstall the plugin and give it a go with the hints above and let you know. Thanks for getting back to me (and an awesome plugin).

FormerLurker, 2018-06-07:
If you're REALLY brave, you can try installing from this URL (within the plugin manager):  https://github.com/FormerLurker/Octolapse/archive/master.zip

The bug I mentioned has been fixed there, and I've run several successful tests using that code.  You can also output an animated GIF with this version and upload a watermark image (not very well tested).

c-hri-s, 2018-06-07:
I'm brave :)

That works perfectly for me, no issues with the revised gcode and the heating problem has gone.
Thanks again

closed 2018-06-07 by c-hri-s