FormerLurker/Octolapse #178: Octolapse Causes Disconnect when Enabled during Heat-Up

XBrav, 2018-06-20:
#### Version of Octolapse

Octolapse Version: 0.3.1

#### Version of OctoPrint

OctoPrint Version: 1.3.8

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES

#### What were you doing when the problem occurred
Begin a print job with Octolapse enabled.

#### What should have happened?
The M110 command should be accepted to reset the line counter to 0, allowing the print to move forward.

#### What happened instead?
The M110 command fails as shown below.
```
Recv:  T:28.75 /0.00 B:64.47 /65.00 @:0 B@:127
Recv:  T:28.87 /0.00 B:64.37 /65.00 @:0 B@:127 W:2
Recv:  T:28.56 /0.00 B:64.44 /65.00 @:0 B@:127 W:1
Recv: echo:busy: processing
Recv:  T:28.69 /0.00 B:64.56 /65.00 @:0 B@:127 W:0
Recv: ok
Send: N0 M110 N0*125
Recv: echo:Unknown command: ""
Recv: ok
Send: N1 M113 S2*96
Recv: Error:Line Number is not Last Line Number+1, Last Line: 2
Recv: Resend: 3
Printer requested line 3 but no sufficient history is available, can't resend
Changing monitoring state from "Printing" to "Error: Printer requested line 3 but no sufficient history is available, can't resend"
Changing monitoring state from "Error: Printer requested line 3 but no sufficient history is available, can't resend" to "Offline (Error: Printer requested line 3 but no sufficient history is available, can't resend)"
Connection closed, closing down monitor
Closing down send loop

```

If Octolapse is disabled prior to this command being issued, then re-enabled afterwards, the error does not occur. Something is being injected into the terminal buffer which causes the M110 command to be disregarded, and creates a print fault.

#### Operating System running OctoPrint and Octolapse

OS Name: OctoPi
Os Version: 0.14.0

#### Printer model & used firmware incl. version

Printer Model: Monoprice Select Mini V1 (RAMPs retrofit)
Printer Firmware Version: Marlin 2.0.x

#### Browser and version of browser, operating system running browser

Browser: Chrome
Browser OS: Windows 10 x64

FormerLurker, 2018-06-20:
Thanks for reporting the issue so thoroughly!  I'm out of town until next
week, but will take a look at this when I get back.

XBrav, 2018-06-20:
No worries, I've developed a plugin or two before. If I get a chance, I'll look into the code and see why it's issuing something. I'm trying to find the setting in the meantime to disable line counts and checksums to work around it, as Octolapse is not activating if enabled after this stage.

speedy80, 2018-06-28:
Hi XBrav, i have exactly the same issue and terminal output looks same as yours. 
I hope this can be fixed in a future release as due to the error I have never been able to use octolapse.

FormerLurker, 2018-07-12:
I found the line of code causing the 'Recv: echo:Unknown command: ""' terminal line, and it's from OctoPrint itself, not Octolapse.

Can you try deleting the entire M110 line (including the line break) from your gcode and manually typing it in?  Perhaps an invisible character got into your start gcode somehow?  Alternatively you could post the file here and I'll take a look.

FormerLurker, 2019-10-12:
Sorry about the super late response, but this problem should be fixed in v0.3.4, but definitely in V0.4 alpha (not yet released), since unicode characters are now (hopefully) properly handled.  Let me know if you'd like me to reopen this issue, and thank you all for posting!!!

closed 2019-10-12 by FormerLurker