FormerLurker/Octolapse #179: Cache entry deserialization failed, entry ignored

simpson1991, 2018-06-20:
#### If this is a feature request describe it here

Hi, i cant install Octolapse and im not realy sure why,

```
Installiere Plugin von https://github.com/FormerLurker/Octolapse/archive/rc/maintenance.zip...
/home/pi/oprint/bin/python2 -m pip install https://github.com/FormerLurker/Octolapse/archive/rc/maintenance.zip
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting https://github.com/FormerLurker/Octolapse/archive/rc/maintenance.zip
Downloading https://github.com/FormerLurker/Octolapse/archive/rc/maintenance.zip (2.5MB)
Requirement already satisfied: OctoPrint in /home/pi/oprint/lib/python2.7/site-packages (from Octolapse==0.3.0rc3) (1.3.8)
Collecting Pillow (from Octolapse==0.3.0rc3)
**Cache entry deserialization failed, entry ignored**
Downloading https://files.pythonhosted.org/packages/89/b8/2f49bf71cbd0e9485bb36f72d438421b69b7356180695ae10bd4fd3066f5/Pillow-5.1.0.tar.gz (14.3MB)
```

And after this an 10minutes later, i get a "cant get install message" no matter if master or maintenance.

But Im able to install every other Plugin :/

#### Version of Octolapse
 
Octolapse Version: master /maintenance

#### Version of OctoPrint

OctoPrint Version: 1.3.8

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES_OR_NO

#### What were you doing when the problem occurred

1. STEP_ONE_GOES_HERE
2. STEP_TWO_GOES_HERE
3. STEP_...

#### What should have happened?
PUT_YOUR_DESCRIPTION_HERE

#### What happened instead?
PUT_YOUR_DESCRIPTION_HERE

#### Operating System running OctoPrint and Octolapse

OS Name: OS_NAME_GOES_HERE
Os Version: OS_VERSION_GOES_HERE

#### Printer model & used firmware incl. version

Printer Model: PRINTER_MODEL_GOES_HERE
Printer Firmware Version: PRINTER_FIRMWARE_VERSION_GOES_HERE

#### Browser and version of browser, operating system running browser

Browser: BROWSER_VERSION_GOES_HERE
Browser OS: BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.  I'm currently saving for some additional test cameras in preparation for develiping multi-camera support.

simpson1991, 2018-06-20:
So i was able to install it, but i made a clean install of octoprint, made an upgrade via console and installed the maintenace version. Because the master version had trouble with compiling at the end.
But i still need to test it because i dont get any image.. maybe while printing?

FormerLurker, 2018-07-09:
If you are using OctoPrint 1.3.9 (rc1 or rc2) there is a bug that severely impacts Octolapse.  If you are using 1.3.8 you should be fine.

Are things running OK?

FormerLurker, 2019-10-12:
OK, this looks to be solved.  I wanted to mention that I've found over the last year that most problems installing Pillow are due to PSU issues.  I know that's hard to believe, but I've now seen this many times, and have been able to verify this myself.  I'm going to go ahead and close this, but if you continue to have problems, please feel free to post a new issue!  Thanks for your contribution!

closed 2019-10-12 by FormerLurker