FormerLurker/Octolapse #181: Octolaps does not work properly.

simpson1991, 2018-06-21:
#### If this is a feature request describe it here

Hi i got it almost to work, but sometimes octolapse does not take the picture in the right positions, i uploaded it to youtube. https://youtu.be/EnY-XwdZVlE.

My settings
Starting Script in S3D:
```
M82;
G90;
G28 ; home all axes
G29; autobed level
G1 Z5 F6000; Lift nozzle
G1 X0 Y-9;
G1 X50 E20 F425;
G92 E0;
```

```
Octolaps_Printer:

Axis Mode:
XYZ Axis Mode Default to absolut
E Axis Mode Require Explicit M82/m83
G90/91 Inf.... Use Octoprint Settings

Auto-Detect Psotion Enabled

Print Volume
X 0    - 250
Y -10 - 220
Z -0   - 200

(actual home position is x:50,y=-9,z=5,e=0)

Tolerance 0,001
Priming Height 0,5

```
```
Stabilization

Fixed Coordinate 100mm (both)
```

```
Snapshot

Layer Change
```
`Snapshot Delay 300ms`

#### Version of Octolapse
 
Octolapse Version: 0.3.1

#### Version of OctoPrint

OctoPrint Version: 1.3.7

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES_OR_NO

#### What were you doing when the problem occurred

1. STEP_ONE_GOES_HERE
2. STEP_TWO_GOES_HERE
3. STEP_...

#### What should have happened?
PUT_YOUR_DESCRIPTION_HERE

#### What happened instead?
PUT_YOUR_DESCRIPTION_HERE

#### Operating System running OctoPrint and Octolapse

OS Name: OS_NAME_GOES_HERE
Os Version: OS_VERSION_GOES_HERE

#### Printer model & used firmware incl. version

Printer Model: PRINTER_MODEL_GOES_HERE
Printer Firmware Version: PRINTER_FIRMWARE_VERSION_GOES_HERE

#### Browser and version of browser, operating system running browser

Browser: BROWSER_VERSION_GOES_HERE
Browser OS: BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.  I'm currently saving for some additional test cameras in preparation for develiping multi-camera support.

FormerLurker, 2018-07-09:
Hi, and thanks for your report!  I've been slowly working through issues since getting back from a long trip, so sorry about the delay!

[Check out this issue](https://github.com/FormerLurker/Octolapse/issues/149), I believe the workarounds mentioned towards the end of the thread should solve your problem.

Alternatively you can install from the development branch using this url within the plugin manager:  https://github.com/FormerLurker/Octolapse/archive/devel.zip

Just make sure you are using OctoPrint 1.3.8, since there are some serious issues with 1.3.9 (rc1 and rc2) that I'm working on.

Warning:   There are some untested features in the development branch, and the code may change.   If you have problems with the development version, reinstall the latest release and use the workaround (relative extrusion).

Let me know if you if this solves your issue.  If not, we'll figure it out!

FormerLurker, 2018-07-09:
I'm moving this to issue #183, so please comment there from now on.  Thanks!

closed 2018-07-09 by FormerLurker