FormerLurker/Octolapse #182: Octolapse will not start

smeg1, 2018-06-22:
#### If this is a feature request describe it here

FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: Octolapse 0.3.1

#### Version of OctoPrint

OctoPrint Version: 1.3.8

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: NO

#### What were you doing when the problem occurred

1. updated OctoPrint 
2. 1.3.9.rc1
3. Octolapse now not running

#### What should have happened?
Octolapse should have started but cannot get it to start installed and re installed
down graded OctoPrint to 1.3.8

#### What happened instead?
still unable to start Octolapse

#### Operating System running OctoPrint and Octolapse

OS Name: octopi
Os Version: OS_VERSION_GOES_HERE
[image]

#### Printer model & used firmware incl. version

Printer Model: PRINTER_MODEL_GOES_HERE
Printer Firmware Version: PRINTER_FIRMWARE_VERSION_GOES_HERE

#### Browser and version of browser, operating system running browser

Browser: BROWSER_VERSION_GOES_HERE
Browser OS: BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.  I'm currently saving for some additional test cameras in preparation for develiping multi-camera support.

smeg1, 2018-06-24:
Ok left it for a few days and thought I would just do a print without Octolapse, but its now started ok don't know why?

FormerLurker, 2018-06-24:
That's odd!  Sorry for the late response, I'm on still on vacation, but can
you tell me if a new version of octoprint just dropped?  I may need to do
some more testing for install and uninstall.  I am planning a coding
marathon the weekend after next, so I'll make sure to look into this then.
As always, thanks for reporting your issue!!!

smeg1, 2018-06-24:
Np never expect you to drop things for this,  I know you have a life outside of this but strange yes was a update OctoPrint OctoPrint: 1.3.9rc1 (release candidate)
Release Notes <https://github.com/foosel/OctoPrint/releases/tag/1.3.9rc1>, also Octolapse said to update pip which I did do
Cheers have a good holiday enjoy

gege2b, 2018-06-25:
Just for the record, I have the same problem with octoprint 1.40-dev since the release of the RC

FormerLurker, 2018-07-09:
Moving this to issue #183 since it has a better description.

closed 2018-07-09 by FormerLurker