FormerLurker/Octolapse #183: Randomly stall print with Octoprint 1.3.9rc1

lewnewby, 2018-07-02:
#### If this is a feature request describe it here

FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: v0.3.1

#### Version of OctoPrint

OctoPrint Version: Octoprint 1.3.9rc1

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES

#### What were you doing when the problem occurred

1. Enable Octolapse
2. Load GCODE to print from local storage
3. Start Print

#### What should have happened?
Print should have completed as desired

#### What happened instead?
Print stalls at random point (stall meaning print head stops moving but heat remains on both hotend and bed)

#### Operating System running OctoPrint and Octolapse

OS Name: OctoPi
Os Version: 14.0

#### Printer model & used firmware incl. version

Printer Model: Prusa i3 MK3
Printer Firmware Version: Prusa FW 3.2.1-576

#### Browser and version of browser, operating system running browser

Browser: Opera
Browser OS: Windows 10

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: https://gist.github.com/lewnewby/83c90da96fbfdf29a674fd05bc24d4c7

#### Link to settings.json

Link to settings.json with all passwords removed: https://gist.github.com/lewnewby/83c90da96fbfdf29a674fd05bc24d4c7

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  https://gist.github.com/lewnewby/83c90da96fbfdf29a674fd05bc24d4c7

#### Link to octoprint.log

Link to octoprint.log:  https://gist.github.com/lewnewby/83c90da96fbfdf29a674fd05bc24d4c7

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.  I'm currently saving for some additional test cameras in preparation for develiping multi-camera support.

FormerLurker, 2018-07-08:
@lewnewby, for some reason the gist link above for plugin_octolapse.log file contains your gcode file.  Can you try to repost the plugin_octolapse.log file please?

lewnewby, 2018-07-08:
I'll get it in a few minutes

Sent from my Verizon, Samsung Galaxy smartphone

"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/FormerLurker/Octolapse/issues/183#issuecomment-403301443",
"url": "https://github.com/FormerLurker/Octolapse/issues/183#issuecomment-403301443",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
},
{
"@type": "MessageCard",
"@context": "http://schema.org/extensions",
"hideOriginalBody": "false",
"originator": "AF6C5A86-E920-430C-9C59-A73278B5EFEB",
"title": "Re: [FormerLurker/Octolapse] Randomly stall print with Octoprint 1.3.9rc1 (#183)",
"sections": [
{
"text": "",
"activityTitle": "**FormerLurker**",
"activityImage": "https://assets-cdn.github.com/images/email/message_cards/avatar.png",
"activitySubtitle": "@FormerLurker",
"facts": [

]
}
],
{
"name": "Add a comment",
"@type": "ActionCard",
"inputs": [
{
"isMultiLine": true,
"@type": "TextInput",
"id": "IssueComment",
"isRequired": false
}
],
"actions": [
{
"name": "Comment",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"IssueComment\",\n\"repositoryFullName\": \"FormerLurker/Octolapse\",\n\"issueId\": 183,\n\"IssueComment\": \"{{IssueComment.value}}\"\n}"
}
]
},
{
"name": "Close issue",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"IssueClose\",\n\"repositoryFullName\": \"FormerLurker/Octolapse\",\n\"issueId\": 183\n}"
},
{
"targets": [
{
"os": "default",
"uri": "https://github.com/FormerLurker/Octolapse/issues/183#issuecomment-403301443"
}
],
"@type": "OpenUri",
"name": "View on GitHub"
},
{
"name": "Unsubscribe",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"MuteNotification\",\n\"threadId\": 352314263\n}"
}
],
"themeColor": "26292E"
}
]

FormerLurker, 2018-07-08:
Hey, can you adjust your printer profile for me?  Change the XYZ Axis mode to:  Default to Absolute

That should fix some of the errors at the beginning of the print.  Alternatively you could add a G90 to your start gcode right around the M83 command.  This is actually the preferred method.

Unfortunately the logs suddenly end without anything that is obviously useful.  I'm going to run your code through the debugger and see what happens.

Thanks for your prompt responses, btw!

lewnewby, 2018-07-08:
I'll do it this evening. Need to run some errands.

Hey, I love this plugin and the results from it. I have had a huge number of compliments from the videos. Even started posting them on my Youtube channel. So anything I can do to help I am more than happy to work on.

FormerLurker, 2018-07-08:
@lewnewby, I am narrowing down the issue and believe that it is related to OctoPrint 1.3.9 Rc1+.  Maybe try downgrading to the latest stable version of OctoPrint (1.3.8) and see if that fixes your problem?  I've already sent Gina a lot of info, and will continue to work with her to fix it, but it would be great to have additional confirmation that the issue started with 1.3.9 rc1!

lewnewby, 2018-07-09:
It absolutely did start when I updated to rc1. I had been on 1.3.8 previously and it was working quite well.

Would you like me to continue trying things or roll back octoprint and wait?

FormerLurker, 2018-07-09:
Yeah, that would probably be for the best.  I'll try to work with Gina to figure this out.  Thanks for your help thus far!

lewnewby, 2018-07-09:
Ok no worries, ping me if you need anything.

FormerLurker, 2018-07-12:
So, a solution has been developed for this issue.  It requires installing both the devel branch of OctoLapse and the latest `staging/maintenance` ([commit 0a634a7](https://github.com/foosel/OctoPrint/commit/0a634a76a0c9f492b7f6f747075aac517482e8f3)) branch.

Here is the URL for the current devel branch, which can be installed through the plugin manager:  https://github.com/FormerLurker/Octolapse/archive/devel.zip

The octoprint upgrade currently must be done manually, but I believe it will go into 1.3.9 rc3 soon.

devsfan1830, 2018-07-31:
@FormerLurker I believe I just had this happen on OctoPrint 1.3.9 final release and v0.3.1. Woke up this morning to my printer stalled with the print head in the center position where OctoLapse takes the snapshot. Heaters were all on for what was likely hours. The Pi seemingly entirely crashed. No web GUI, SSH would not respond. Is this the "deadlock" mentioned as solved in 0.3.3RC1? Is it advised that anyone updated to OctoPrint 1.3.9 should install the Octolapse RCs?

FormerLurker, 2018-07-31:
@devsfan1830, I believe the latest rc likely fixed this issue.  Octoprint received a fix in RC3, but it only works in conjunction with Octolapse 0.3.3RC1 or higher.

Try reinstalling Octolapse from this link via the plugin manager:  https://github.com/FormerLurker/Octolapse/archive/v0.3.3rc2.zip

This version now supports maintenance and development release update notifications as long as you change your settings in the 'Software Update' plugin (OctoPrint settings).

Please let me know if you have this problem again after upgrading.  I've been working on a new release for a good while now, and hope to have it ready soon.  Unfortunately I've not received any feedback from the [RC2 issue](https://github.com/FormerLurker/Octolapse/issues/198), so I'm not sure if all of the problems were solved.  Also, a lot has changed within the development branch since that release, so keep your eyes peeled for a new maintenance RC soon!

FormerLurker, 2018-08-11:
I'm closing this out since it has likely been fixed in the latest RC and devel RCs.  Let me know if you need me to reopen the ticket.

closed 2018-08-11 by FormerLurker