FormerLurker/Octolapse #184: Default to Absolute by default

MathijsG, 2018-07-04:
#### If this is a feature request describe it here

I discovered that Octolapse didn't work here with Cura, only with S3D. I thought it must be something about the not explicitly defining of printing relative or absolute in the slicer. Instead of modifying every slicer, I discovered that when I set up the settings to be like:

"Default to absolute X Y Z"
"Default to absolute E"

it did work stock out of the box with Cura and S3D.

Maybe we can default to this, instead of the "Require explicit" option that is default, which will often render Octolapse in a vague not working state. 

#### Version of Octolapse
 
Octolapse Version: 0.3.1

#### Version of OctoPrint

OctoPrint Version: 1.3.8

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES_OR_NO

#### What were you doing when the problem occurred

1. STEP_ONE_GOES_HERE
2. STEP_TWO_GOES_HERE
3. STEP_...

#### What should have happened?
PUT_YOUR_DESCRIPTION_HERE

#### What happened instead?
PUT_YOUR_DESCRIPTION_HERE

#### Operating System running OctoPrint and Octolapse

OS Name: OS_NAME_GOES_HERE
Os Version: OS_VERSION_GOES_HERE

#### Printer model & used firmware incl. version

Printer Model: PRINTER_MODEL_GOES_HERE
Printer Firmware Version: PRINTER_FIRMWARE_VERSION_GOES_HERE

#### Browser and version of browser, operating system running browser

Browser: BROWSER_VERSION_GOES_HERE
Browser OS: BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.  I'm currently saving for some additional test cameras in preparation for develiping multi-camera support.

FormerLurker, 2018-07-09:
I will consider this.  The problem is that if Octolapse has the wrong coordinate system it's quite possible that it will go completely amok, causing the print head to rampage around :)  Maybe I'm being too paranoid.

I'm wondering if some kind of interactive wizard could be useful to solve this problem, or some way to adjust the settings during a print?  I'm interested in any way to make setup more convenient while also being as safe as possible.

If anyone else has an opinion, I'm very interested in hearing it!

MathijsG, 2018-07-10:
Yeah a wizard should be nice. Now there is a whole a dimension between properly installed and properly setup, for many people sometimes looks like the software is broken, while in fact it's just some settings that are set up wrong. And when it's working it's very nice! If you'd ask me Octolapse should eventually replace the stock functionality and upgrade this stock functionality with your software. Maybe you should get this upstream so this will be in octoprint eventually? :)

FormerLurker, 2018-07-10:
I am well aware of the issue of being installed but not properly configured.  Let's face it, there are a ton of options, and many of them are simultaneously critical and confusing.  The XYZ and E coordinate system commands rank high among them, and I've been straining my brain to come up with a simple way of solving this problem.  Here are the options I've been thinking about:

1.  Add a wizard step that has lots of links and pictures explaining how to discover the current coordinate system based on the user's slicer settings.  This could work, but would depend a lot on having good documentation complete with pictures or maybe video.  Unfortunately it would still require some skill.
2.  Add some dynamic error checking to alert the user and suggest a fix.  Unfortunately this would mean a failed timelapse.
3.  Scan a gcode file and search for the settings.  If possible, import them.  This would be cool, and could solve retraction/detraction issues, but wouldn't work in all cases.
4.  Update the default profiles as you suggested.  This will work in most cases, but I feel this could cause some crazy behavior.  This may be the best option, and is certainly the simplest.  Maybe we can discuss the pros and cons of this option.  Perhaps I am just being paranoid?
5.  Some combination of 1 and 4 might also be a good idea.  

Also, I'd really like to tackle the retraction settings since 90% of the time poor print quality is the result of incorrect retraction length/speed or zhop.

FormerLurker, 2019-10-12:
So, I think this issue has now been more or less solved over the last year.  I've added some new profile options for the XYZ and E axis modes, including 'Force Absolute' option which just sends a G90 or M82 at the start of a print.  Additionally there is now a profile repository for Octolapse, and most of the profiles have 'Default to Absolute' set for both XYZ and E axis modes.  These features exist in Octolapse V0.4, which is currently unreleased.  If you'd like to try it out let me know, but for now I'm going to close this issue out.  Feel free to open a new issue if you have any additional issues, or to post here if you would like me to reopen this.

Thanks for your post!!!

closed 2019-10-12 by FormerLurker