FormerLurker/Octolapse #185: Print Won't Complete with Prusa i3 MK3

MakerBoz, 2018-07-07:
#### If this is a feature request describe it here

FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: 0.3.1

#### Version of OctoPrint

OctoPrint Version: 1.3.8

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: No

#### What were you doing when the problem occurred

1. Turn on Octolapse and print as usual
2. Octolapse takes pictures as it is supposed to
3. Print finishes and shows 100% completion, but shows "a few seconds" or "less than a minute" left on the print
4. Octolapse will not render the file because it is "Waiting for the print to complete before rendering"
5. Cancel button grayed out, so canceling not an option
6. Resetting printer still does not allow Octolapse to think the print is done.

#### What should have happened?
Print should have finished 100% with 0 time left, allowing Octolapse to render.  Alternatively, Octolapse could be allowed to render manually if the user specifies in case this is a problem specific to Octoprint and not Octolapse.

#### What happened instead?
Octoprint did not actually finish the print, so Octolapse could not render the video.

#### Operating System running OctoPrint and Octolapse

OS Name: OctoPi
Os Version:  0.15.0PE

#### Printer model & used firmware incl. version

Printer Model: Prusa i3 MK3
Printer Firmware Version: 3.2.1-576

#### Browser and version of browser, operating system running browser

Browser: Google Chrome V67.0.3396.99
Browser OS: Windows 7

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: https://pastebin.com/XLTiACtk

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  https://pastebin.com/YX4ef3i8

#### Link to octoprint.log

Link to octoprint.log:  https://pastebin.com/thMymRi8

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: https://imgur.com/XlSsNWD https://imgur.com/cmpxk4c

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.  I'm currently saving for some additional test cameras in preparation for develiping multi-camera support.

MakerBoz, 2018-07-08:
I've run a print without Octolapse enabled and it seems that this issue stems from Octoprint itself.  The print finished as it should with 100% completion, but the time still said "less than a minute left."  This means that the problem really came from Octoprint not realizing that the print is actually finished and setting the time accordingly.  Octolapse must rely on that timer reaching zero before it will allow the video to be rendered.  You could potentially get around this problem by relying on the % completion instead of the time, or just allowing the user to render the video without the print technically being completely done.

FormerLurker, 2018-07-09:
Thanks for the update!  I think there is a fix for this in OctoPrint 1.3.9 RC2, but there is a breaking change that locks the server when using Octolapse.  I'm working on [this issue] at the moment, and when it is fixed maybe you can upgrade both OctoPrint and OctoLapse (I'll let you know how) and see if your issue is solved too?

Thanks for your post!

MakerBoz, 2018-07-09:
Sounds great.  I would be happy to give it a shot.  Just let me know when you want me to update everything and we will see how it does.

Jonnie-Cache, 2018-07-19:
@FormerLurker I tried the release candidate last night. I'm not sure what went wrong, but here is what I got...

[https://d.pr/i/wSYdLH](url)

Jonnie-Cache, 2018-07-20:
@FormerLurker I'm sorry about that. I initially thought it was still broken (or there is another bug). Then, I thought it was working so I deleted my post. Now, I can confirm that there is a different problem with the rc. Twice now, I have received an error. I ran a test print with no problem, so I believe the settings are correct.

[Screenshot](https://d.pr/i/cMwJmZ)
[plugin_octolapse.log](https://d.pr/f/eJGfPN)
[octoprint.log](https://d.pr/f/9Xsckp)
[serial.log](https://d.pr/f/RWVteW)

FormerLurker, 2018-07-20:
Can you post your gcode file and your settings.json file please?  Hopefully I can figure out what is up.

Jonnie-Cache, 2018-07-20:
I realize that my previous comment was poorly worded. To be clear, the prints completed successfully. But, the OctoLapse failed.

[problem.gcode](https://d.pr/f/9EJh7l)
[settings.json](https://d.pr/f/9EJh7l)

FormerLurker, 2018-07-20:
I could tell from the picture you posted, no problem.  Thanks for the info, I'll dig in tomorrow after work.  I already have an idea where to look.

FormerLurker, 2018-07-20:
@Jonnie-Cache , it looks like you accidentally posted the gcode twice.  Can you try posting settings.json again?

Jonnie-Cache, 2018-07-20:
Sorry about that! I'll fix it as soon as I get home.

FormerLurker, 2018-07-20:
No problem.  FYI, your gcode ran through the default MK3 profile on the virtual printer no problem.  My bet is that there is a settings issue somewhere.

I did notice that your G90 comes a bit late in your start gcode.  Try moving this:

```
G21 ; set units to millimeters
G90 ; use absolute coordinates
M83 ; use relative distances for extrusion
```

to the top of your start gcode.  This is important becuause you have absolute G1 movements BEFORE you set your coordinate system currently.  This Octolapse probably won't care because it will pick up on the coordinates later in the print, but it's a good idea for safety's sake.

If you do what I suggest above you can also erase the second M83, though it's not hurting anything.  It is just redundant :)

Jonnie-Cache, 2018-07-28:
@FormerLurker I tried that and it still didn't work. :(

FormerLurker, 2018-07-28:
Ok, be sure to send me settings.json so I can help you figure this out!

Jonnie-Cache, 2018-07-29:
Oh, crud. I thought I did that already. Here it is:

[settings.json](https://d.pr/f/quC9Jq)

Jonnie-Cache, 2018-07-31:
OMG. I'm so sorry. I think the service doesn't like it when I upload json files. I zipped it this time and it seems to stay put. Hopefully this works...

[settings.json.zip](https://d.pr/TI1wjS)

FormerLurker, 2018-07-31:
Looking better :) I'll take a look tomorrow in the evening.  Shouldn't take too long!

FormerLurker, 2018-08-01:
OK, i took a look and noticed that your snapshot profile (layer trigger) is set to capture every 0.6MM.  Not sure if that's what you wanted.  However, it does take snapshots when I run it through Octolapse.

Can you do me a favor and change your snapshot profile (layer trigger) so that the 'Trigger Height' = 0?  Also, for this test, uncheck 'Trigger on Z-Hop Only'.  Set your 'Debug' profile to 'Test Mode - Full Diagnostic', unload your filament and run that file you sent (X-Cable_Holder_v1.1_MK3.gcode).

While the print is running enable the 'Position State' info panel by clicking on the slider on the Octolapse Tab (make sure to click on 'Current Settings' if you don't see this option).  If snapshots aren't taken send me a snapshot of this panel while the print is running, say after three layers, and download plugin_octolapse.log and send that to me (again).

Thanks, and sorry for the delay!

Jonnie-Cache, 2018-08-01:
Thanks. I'll try that now. I don't know why it was set to 0.6 MM either, but I changed that to 0. I made those other changes and will try that now.

Since you brought up the different "State" info panels, I'm not sure I understand that those settings do.

I'll test the new settings out tonight!

FormerLurker, 2018-08-01:
@Jonnie-Cache, No problem!  The info panels are just a way of looking under the hood, so to speak.  Here is an example of the panel I would like to see:

[image]

Additionally, you might want to try using the 'Full Diagnostic - Test Mode' debug profile.  This will prevent your printer from warming up or extruding to save time.  UNLOAD YOUR FILAMENT though, in case I goofed :)

Then upload the plugin_octolapse.log file (settings->Logs) like you did before and I'll be able to diagnose better.

Jonnie-Cache, 2018-08-02:
I don't know why it was set to that. I don't recall doing that. But, I've fixed those settings and I just tried it. It failed last night, but I'm not sure if I did something to cause it. It worked once last night and once again this morning. I didn't print the whole GCODE file, but let it go to layer 4 or 5.

I could swear that the panels weren't showing up for me before during prints. That's why I was asking about what they did. But, they are showing up now.

I don't know. Maybe it's not worth pursuing at this point. But, here are the log just in case...

[plugin_octolape.log](https://d.pr/lb1vcm)
[timelapse.mp4](https://d.pr/v/aeqb1c)

Let me know if you want me to do any other testing.

FormerLurker, 2018-08-02:
@Jonnie-Cache, there are some bugs that could 'lock out' octolapse and prevent it from working on subsequent prints without a restart of OctoPrint in the version you are using.  That's probably what happened.

This has been fixed, and I'm working to get a release put together this weekend.  I think the version you are using should notify you of updates, but make sure you switch your software update settings to 'Maintenance RCs' or 'Development RCs' since I'll release one of those before moving on to a stable release.

Jonnie-Cache, 2018-08-03:
Awesome! Thanks! I'll do that!

FormerLurker, 2018-08-11:
@Jonnie-Cache, did my previous suggestion help at all?  I'll leave this open for a bit longer while waiting for feedback.

Jonnie-Cache, 2018-08-13:
Sorry. I've actually been down. One of the wires broke off my heating element. Parts should be here tomorrow so that I can rebuild the hotend and get back up and running asap.

MakerBoz, 2018-08-13:
Mine has been resolved.  I'm not sure what exactly fixed it, but the newest version of both Octoprint and Octolapse work just fine together.  Thank you.

Jonnie-Cache, 2018-08-19:
Hi, @FormerLurker. Finally got back up and running yesterday. I made multiple attempts and different reboots. Still having problems. I was already on the Maintenance RC setting, but I tried Devs as well. I don't think it downloaded any new/different version of Octolapse.

FormerLurker, 2018-08-20:
@Jonnie-Cache, 

Try installing from this url:  

Then reboot and clear your cache (ctrl+f5).  Then you should be on V0.3.3rc3.dev0.  Then turn on 'full diagnostic - test mode' and print a few layers of a small test print.  Finally shoot me a copy of plugin_octolapse.log and settings.json and I'll dig in again and see what I can find.  Hopefully it just works finally :)

FormerLurker, 2018-08-20:
@Jonnie-Cache, I forgot the URL!  Here is is:  https://github.com/FormerLurker/Octolapse/archive/devel.zip

Jonnie-Cache, 2018-08-21:
I wish I could say that it worked, but it didn't. :( In this case, it gives me errors right off the bat and the print is aborted.

[plugin_octolapse.log](https://github.com/FormerLurker/Octolapse/files/2304619/plugin_octolapse.log)
[octoprint.log]
[octoprint.log](https://github.com/FormerLurker/Octolapse/files/2304621/octoprint.log)
[settings.json]
[settings.json.txt](https://github.com/FormerLurker/Octolapse/files/2304624/settings.json.txt)

FormerLurker, 2018-08-21:
If you only have one camera, disable the one you haven't configured (click the check box to the left of one or edit the settings).  I haven't implemented a good settings migration for that yet, but I just started that this evening :)

I'll take a look at your log, but I'm betting one of the issues will be solved when you disable one of those cameras.

Jonnie-Cache, 2018-08-22:
Weird. I'm sorry that I didn't notice that part of the wiki. I don't understand why it became an issue later. I don't think that I've used any different profiles for my camera. But, I'll give this a go and report back! Thanks!

FormerLurker, 2018-08-22:
@jonnie-cache, you'll be happy to learn that I added the proper migration for moving from the single to multiple camera version so others won't have issues, and have added webcam server detection and validation for the custom image preferences, including a link to the documentation in the wiki, so your experience won't be repeated.

Jonnie-Cache, 2018-08-22:
@FormerLurker You're awesome! Thank you!

FormerLurker, 2018-09-10:
This issue has been open for a while, and I think the problems have been resolved.  If you are still having issues just reply and I'll re-open it.

Thanks everyone for posting!

closed 2018-09-10 by FormerLurker