FormerLurker/Octolapse #187: [Request] DSLR/External Camera Triggering

SrgntBallistic, 2018-07-08:
#### If this is a feature request describe it here

Allow configuration for sending triggers to a DSLR via GPIO instead of or in addition to taking a snap with an onboard camera. And/Or launching an external script which can trigger an external camera by other means.

I would like to make some super high resolution and quality timelapses with Octolapse. In the past I've triggered my DSLR via arduino and think my method/circuit can work with the Raspberry Pi GPIO as well.

Alternatively allowing a python script to be run when a snap is taken could be beneficial as some DSLRs can be hooked up to a Raspberry Pi or PC via USB.

This would be really helpful as the only way I've seen to have a DSLR show up as a webcam on a raspberry pi is with a $100+ UVC USB to HDMI converter

Cheers,
Brian

#### Version of Octolapse
 
Octolapse Version: Octolapse 0.3.1

#### Version of OctoPrint

OctoPrint Version: OctoPrint 1.3.8 running on OctoPi 0.15.1

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES_OR_NO

#### What were you doing when the problem occurred

1. STEP_ONE_GOES_HERE
2. STEP_TWO_GOES_HERE
3. STEP_...

#### What should have happened?
PUT_YOUR_DESCRIPTION_HERE

#### What happened instead?
PUT_YOUR_DESCRIPTION_HERE

#### Operating System running OctoPrint and Octolapse

OS Name: OS_NAME_GOES_HERE
Os Version: OS_VERSION_GOES_HERE

#### Printer model & used firmware incl. version

Printer Model: PRINTER_MODEL_GOES_HERE
Printer Firmware Version: PRINTER_FIRMWARE_VERSION_GOES_HERE

#### Browser and version of browser, operating system running browser

Browser: BROWSER_VERSION_GOES_HERE
Browser OS: BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.  I'm currently saving for some additional test cameras in preparation for develiping multi-camera support.

FormerLurker, 2018-07-09:
Thanks for the suggestion!  There are actually a few requests like this now, and I will soon combine them into one, and will close this issue out when I do.  I plan to allow a few options:  GPIO signaling, Send Gcode, Call bat/bash script, acquire from stream (current option).

FormerLurker, 2018-07-23:
This feature is now working in the devel branch.  It took some doing to get the bash script to work.  I'm going to update the wiki soon with some tips on how to get it going.  If anyone wants to test it out I can provide a sample .bat or bash script and some installation instructions.

SrgntBallistic, 2018-07-23:
Awesome. I'm not sure I'll have time this week to test but definitely willing to help!

Thanks!

HayderJawad, 2018-07-24:
That's great @FormerLurker  I would be happy to test, if you agree.

FormerLurker, 2018-07-24:
Great!  I'll post install instructions here as soon as they are finished.

FormerLurker, 2018-07-24:
I created some beta instructions for getting a DSLR working on pi with gphoto2.  You can find it here: https://github.com/FormerLurker/Octolapse/wiki/Configuring-and-External-Camera

Please let me know what works and what does not.  I appreciate any input you have on the guide, especially where things are unclear, or if I need to add more links.  FYI, I did NOT run through the script from a fresh install of octopi, but that would be REALLY appreciated if you have the time.

Also, I just merged some pull requests and haven't had time to adequately test, so let me know if there are any problems with the plugin in general.

Thanks!

HayderJawad, 2018-07-24:
It works, just like what you instruct, well done @FormerLurker thanks.
I have some minor issue I will do some other tests and will list these issues later, one note you mentioned in the instructions this 
`sudo apt-get install gphoto2``

there is additional ` at the end of the line,,, this made some delay for me ;)

Thanks.

FormerLurker, 2018-07-24:
Awesome, I'm very glad it worked for you!  I fixed the issue you pointed out (extra `) and fixed the URL (changed and to an), but the old URL no longer works.  Here is the updated link:

https://github.com/FormerLurker/Octolapse/wiki/Configuring-an-External-Camera

Thanks for your feedback!

HayderJawad, 2018-07-24:
Now the video quality is scary, I can use my DSLR camera instead of the low quality webcam,
the script works just fine, I noticed that when I aborted printing for two times the Octoprint restarted!, but when the printing completed fine, nothing happens to the Octoprint.
here is a sample 
[AI3M_test_20180724200049.zip](https://github.com/FormerLurker/Octolapse/files/2225192/AI3M_test_20180724200049.zip)
I am glad to be the first one to test :+1:   :grinning:

FormerLurker, 2018-07-24:
Wow, that is insane!  It's above the resolution of my monitor (4k), so I can't even see how good it really is without zooming in.  You could probably crank up the bitrate in the rendering properties to get an even better image.  Also, there is now a new rendering codex available thanks to @shadowen - H.264/MPEG-4 AVC, in case you haven't seen that.  A fine combo if you ask me.

If you have the time, send me a video of an actual print.  I would like to see more of these scary quality videos :)

Oh, and to my knowledge you are indeed the first one to test!  Bravo and thanks!

SmartnPrint, 2018-07-28:
Hi!

Thanks for this awesome update and your amazing work!

I wanted to try this setup and can't get it to work.

I'm using the "v0.3.3rc1.dev"-Version. The newer versions won't work with my new installation of octoprint.

Sadly the dropdown-menu does not exist in my interface. I tried reinstalling, updating and so on.
The only dropdown is in the normal menue. I can choose between "Default Camera" and "Logitech Camera" there.

Maybe you got an idea.

Jonas

FormerLurker, 2018-07-28:
@SmartnPrint, yeah there is no present for this camera yet.  You have to create your own profile.  Go into the regular octoprint settings, then click the octolapse link (bottom left corner).  Now find the camera tab in the octolapse settings and click create new profile and follow these steps: https://github.com/FormerLurker/Octolapse/wiki/Configuring-an-External-Camera.

You might be better off installing from the devel branch too.  Several fixes in there.  Here is the link you can use to install from the plug-in manager: https://github.com/FormerLurker/Octolapse/archive/devel.zip

Do that before following the steps above.

SmartnPrint, 2018-07-28:
@FormerLurker Thanks for your answer!

Yeah, thats what i tried so far. I evaluated it one more time.

After pressing "Add Profile" I can type in a name and a description for the camera. Below that there is an area called "Webcam Setup". If i type in "External Camera - Script" as the name nothing changes. I also can not find the "Snapshot acquire Script"-field.

I'm currently running the 0.3.3rc2.dev0 version, as the newer one always fails to load: "No profile found - Internal Server Error".

I think that I am doing something wrong, but can't figure it out... Maybe you can give me one more hint.

Jonas

SrgntBallistic, 2018-07-29:
Version of Octolapse
Octolapse Version: Octolapse 0.3.3rc2

Version of OctoPrint
OctoPrint 1.3.9 running on OctoPi 0.15.1

So got this setup last night. No problems with detecting the camera or manually taking snapshops.

However every time I run a print a snapshot error occurs after the first snapshot. It seems like there's only ever 1 image in each folder created and thus rendering always fails.
[image]

Example of error:
[image]

Also the preview that's in that image is from the last complete print I did with the Pi camera

I've tried running it with both the sudo line in the py script uncommented as well as with the other line uncommented...

In the logs there's an error after trying to run the script about "Error output was returned from the snapshot script: [sudo] password for pi: ERROR: Could not get image."

I'm fairly certain sudo password is correct when I have it in there. I believe I set up everything as prescribed int the configuration page of the wiki.

Here's what my script file looks like
[image]

 Attached is my log. I've also tried both timed and at layer change and it doesn't seem to work either

[plugin_octolapse.log](https://github.com/FormerLurker/Octolapse/files/2238914/plugin_octolapse.log)

jetsloth, 2018-08-08:
@FormerLurker dude this is giving me a hardon. I've just spent last hour looking online how to do this with a DLSR. I'm happy to beta test - i'm at your service! 

I'll jump on the beta instructions page now and give it a go now. Let you know how I get on.

FormerLurker, 2018-08-08:
@SrgntBallistic and @jetsloth,

Some bugfixes/enhancements  have been pushed to devel, so please try installing from this most recent commit: https://github.com/FormerLurker/Octolapse/archive/fd2ab947061ea2d1e8c36fd978039b89eb04754c.zip

As always, ctrl + F5 (or shift + f5) to clear your browser cache after reboot, else you'll get old script, and nobody wants that, lol!

Please check your new DSLR camera profiles after they have been created and crank up the 'Snapshot Timeout' to 10000 (10 seconds).  That's outrageously huge, I understand, but the usb xfer of an ultra high res image is slow and sometimes unpredictable.  While we're testing it's better to avoid a timeout error if possible.  I need to improve the logging and error reporting too to make debugging somewhat bearable.  I'm working on a 'test script camera' button that should be very helpful for this purpose.

@SrgntBallistic, adding the late download option is quite tricky as it turns out.  It's much easier to defer until the end of the print and download all at once, though that too has its challenges.  They will be overcome, but I wanted to let you know I haven't forgotten about it.

Thanks!

SrgntBallistic, 2018-08-08:
@FormerLurker thanks for they update and work you're putting in. Hopefully finishing up a project tonight and can try the new version.

jetsloth, 2018-08-08:
@FormerLurker OK got it setup and 90% working.

If I have the snapshot delay at 0 nothing happens (no photo), but a value in there and it holds for ages (5 seconds like you mention) and takes the shot. It's kind of like the new user-defined delay option gets ignored.

SO i'm getting it to actually take multipple photos and stores them all in job folders in the temp dir. Just ran a full test print and finished with 67 photos and completed. However was getting snapshot errors the whole time so rendering did not happen.

Cannon 70D. Running the save time-lapse to Dropbox plugin also.
[image]
[image]
[image]
[image]
[image]
[image]

Hope that helps, the images are amazing qlty!!!
This is going to be mindblowing when it's stable!

FormerLurker, 2018-08-08:
@jetsloth, did you use the link I sent to install?  Your interface does NOT look right (the icons should be different, you should have multiselect camera options on the tab).  Either reinstall from the link I posted in my previous comment, or try pressing ctrl+F5 to refresh your cache.

Once this is done your screen should look like this:  
[image]

Also, the error popup (failed repeatedly to capture camera.....) I think is from the built in timelapse plugin.

Go back into your camera settings.  Now you should have 2 options:  Delay and Timeout.  Your delay should be at 0 and your timeout should be at 10000.

Try that and let me know how it goes.

jetsloth, 2018-08-08:
@FormerLurker Sorry yeh By the time I posted that last message I had then seen the updated Link. New interface is correct now. Will try it again. Set to 10000 timeout and 0 delay.

FormerLurker, 2018-08-08:
@jetsloth, that's cool!  These versions are changing so fast it's a wonder anyone can keep up :)

Also, I was looking at the first error message and you might want to double check your password in the snapshot script to make sure it's correct if you encounter more errors.

jetsloth, 2018-08-08:
@FormerLurker Could imagine! Looking good brother, the updates are working. I've got my C920 running along with the Cannon 70D (it's freezing in the garage so nicer to watch whats happening in the office :D)

3 Shots in, no errors.....snaps below. I'll let it go (short print) to see if it complies into a render.
[image]
[image]
[image]
[image]

FormerLurker, 2018-08-08:
@jetsloth, at least there were no errors!  I'm not sure what is going on with the jerky video.  I will look through the log to see if I can find out what is going on.  Even if there are missing frames, it should just skip over them.  It shouldn't appear to jerk like that, unless for some reason the same image is being copied down from the camera each time.  I'll definitely take a look after work.

Regarding the rendering time, remember there's going to be a lot more images than 65 if you use pre or post roll (first and last image gets copied to fill out the video length), but yes it takes an enormous amount of time.  I'm thinking of adding a setting for 'number or rendering threads'.  My pi has plenty to spare, and it should decrease rendering time in a roughly linear way.

The lighting is all controlled by your camera.  I recommend setting everything to manual.  On my Nikon the trick to simple and fast configuration seems to be to set the camera first to 'auto' mode, then half press and allow the camera to focus.  Then turn auto-focus off and switch the camera to 'manual'.  At least on my camera the current settings are retained when I switch to manual mode (you'll have to test yours out).  This give me a much more stable image, and each snapshot completes more quickly.  Also, having more lighting and lower exposure seems to help as well.

If you wouldn't mind, maybe you could try to capture a timelapse from both your webcam and your DSLR at the same time?  I'd like to know how that goes as well.

Thanks for your help thus far!

FormerLurker, 2018-08-08:
@jetsloth, I couldn't help myself and took a peek at the log.  It's still reporting an error when downloading images, (seems to be reporting a password prompt) though I'm not sure if it's legit or not.  Maybe you could paste in your script and just replace your password with random letters?  It's possible there is some small error.

There were two definite errors though:

*** Error ***              
Canon EOS Half-Press failed (0x2019: PTP Device Busy)
ERROR: Could not capture image.
ERROR: Could not capture.
*** Error (-110: 'I/O in progress') ***

Hmmm..  I wonder how this happened.  It occurred two times, fyi.

jetsloth, 2018-08-08:
@FormerLurker the script I used was from your example as is. Just updated my password. 
That error may have been from me half pressing. I know I did that a couple of times during the print. To get focus as I changed angels on the camera. SO that is more likely me.

Next print I'll do I'll leave it and not touch it so I dont become a variable hahahah.

Sorry I'm no good at markdown!

#!/bin/sh
# Put the arguments sent by Octolapse into variables for easy use
SNAPSHOT_NUMBER=$1
DELAY_SECONDS=$2
DATA_DIRECTORY=$3
SNAPSHOT_DIRECTORY=$4
SNAPSHOT_FILENAME=$5
SNAPSHOT_FULL_PATH=$6

# Check to see if the snapshot directory exists
if [ ! -d "${SNAPSHOT_DIRECTORY}" ];
then
  echo "Creating directory: ${SNAPSHOT_DIRECTORY}"
  mkdir -p "${SNAPSHOT_DIRECTORY}"
fi

# if sudo is not required (it shouldn't be, but I couldn't get it to work without it)
# uncomment this line and comment out the next
# gphoto2 --capture-image-and-download --filename "${SNAPSHOT_FULL_PATH}"
# Capture and download a snapshot with sudo
echo "I_love_sloths_lol_but_seriously" | sudo -S gphoto2 --capture-image-and-download --filename "${SNAPSHOT_FULL_PATH}"

if [ ! -f "${SNAPSHOT_FULL_PATH}" ];
then
  echo "The snapshot was not found in the expected directory: '${SNAPSHOT_FULL_PATH}'." >&2
  exit 1
fi

FormerLurker, 2018-08-08:
@SmartnPrint, It's been a while since we've talked, and I am wondering if you ever got things straightened out.  If not, please try upgrading Octolapse inside the plugin manager with this URL:  https://github.com/FormerLurker/Octolapse/archive/fd2ab947061ea2d1e8c36fd978039b89eb04754c.zip

My guess is that your browser was caching the old javascript, so be sure to press ctrl+f5 to reload your javascript after you restart.  You should now see the new options in the camera profile.

jetsloth, 2018-08-09:
@FormerLurker Dude this is looking sexy as $^*$^&!!!!! Ok so got late last night, was up till 4am working/playing with it hahahaha.

This morning setup another dry run.

### My notes
- Enabled the webcam and also Cannon70D
- Output was 3 renders in total = WebCam Octolapse + Default Octoprint timelapse + 70D Octolapse
- Print time was 3 hours
- Changed camera format to lower res of 2736 × 1824
- The final output was 22.5MB but felt like it rendered much quicker - didn't realise it had rendered in the background so can't be sure how long it took - deff quicker tho with the lower res on camera
- Camera Focus + ISO + white balance all set to manual
- Added a low depth of field (Niffty Fifty) lens hence the nice depth of field blur effect

### Results
**Default OctoPrint timelapse**
https://www.dropbox.com/s/gzm1sn8ixqc2wgc/Benchy-CR10_20180809103700.mpg?dl=0
**C920 Octolapse timelapse**
https://www.dropbox.com/s/55mhb779pu7lcxo/Benchy-CR10_20180809124801.mp4?dl=0
**Cannon70D Octolapse timelapse**
https://www.dropbox.com/s/d9zjf5jxfi1oed8/Benchy-CR10_20180809124801_1.mp4?dl=0

Going to try a real print now. Will post results.

FormerLurker, 2018-08-09:
@jetsloth, smooth as butter!  I appreciated the comparison with  a regular timelapse too, that does provide a nice contrast.  Is there any way to get your DSLR video in 4k?  720p was the best quality I could view on YouTube.  Can't wait to see a longer print!

jetsloth, 2018-08-09:
Na the only camera I have in 4k is the gopro...I wonder if that's on the cards next hahahah.
 
I'll try 1080p?

jetsloth, 2018-08-09:
@FormerLurker OK have increased image resolution to around 3500x2500 for each image. Says each image will be around 8MB. Over a 5-7 hour print that's some serious rendering. See how it goes. Works fine on lower res. 

Your intention to try support 4k?

FormerLurker, 2018-08-09:
Ah, yes.  I misread your resolution.  I don't think GoPros (at least not all) support direct capture, but it's worth a shot!

FormerLurker, 2018-08-09:
@jetsloth, I know it is possible.  I've seen some posted, but no idea how long it will take to process on a pi.  I'm thinking about adding a progress bar.  I'm definitely going to add a thread count option.

jetsloth, 2018-08-09:
@FormerLurker Yeh progress bar would be good, first print failed (bad first layer) and the video is still rendering out the 2 shots at that higher pixel count.

The quality of the 3 frame video is great tho! I might stick with it - prob take an hour to render tho.

jetsloth, 2018-08-09:
@FormerLurker If this live print test goes ok, might try 2 Cannon cameras connected at once.....

jetsloth, 2018-08-09:
@FormerLurker Great results mate. The original print I started would of been great but failed early on (extruder jam). Defaulted to a quick benchy for a test. 

The possibilities are endless now. It's more like a photography shoot, lighting, backdrop, zoom, lens choice, multipale angels etc. I'm excited to see how I can step up the quality of the timelapses from a creative point from here on in.

**Results below.**
Default
https://www.dropbox.com/s/wo7kxllo5q9oei0/Timelapse3DBenchy_20180809165317.mpg?dl=0
Octolapse webcam
https://www.dropbox.com/s/8y3irnwhz0d6zvh/Timelapse3DBenchy_20180809191133.mp4?dl=0
Octolapse Cannon70D (sorry moved the camera a couple times) 
https://www.dropbox.com/s/uwa82p9viym3rqc/Timelapse3DBenchy_20180809191133_1.mp4?dl=0

**Notes**
- I Kept the res at 2736x1824, For the early failed attempts I noticed higher res already made the render patchy but still going to persist with this, haven't had a print go long enough to try it
- I'll experiment with 2xCannon DLSR cameras and report back.

**Questions**
- Any idea how we can speed up the 10seconds timeout? Should I play with these values myself? I don't mind the extra time as it gives the camera time to do its thing. But maybe 1000-2000 may work better? I'll experiment.
- More of an idea really. In addition to the DLSR I wonder if you could send another gCode which triggers the rotation of another stepper motor E1 or E2 for example. On the sending of M240 to trigger camera, send the stepper a signal to rotate X revolutions (user defined). The idea behind this is that the camera is mounted to the stepper, and the stepper is pushing along a track. Thus, introducing 180,360 timelapse or paning timlapse - examples [here](https://www.youtube.com/watch?v=NnSAdTc8-ug) and [here](https://youtu.be/GPDhl03Q6ic?t=14s) Somthing to ponder nontheless!

Thanks, and I'll keep playing!

FormerLurker, 2018-08-09:
@jetsloth, you don't want to turn down the timeout.  This is just the maximum amount of time Octolapse will wait to acquire an image from your camera.  If the timeout is too low, the snapshot script will be cancelled, and your print will continue without taking a picture.

The key will be to reduce the time it takes to actually take the photo via GPhoto2.  I've been playing around with it and haven't yet figured out a way to do it, but I DO notice that it takes a lot longer to initiate the snapshot than I think it should.  There may be some initialization step that can be made manual.  I'll let you know what I find.

Regarding a timelapse track (as in your examples), there may be a way to do this with the snapshot script.  You're not the first person to suggest this, so I'll think about the easiest way to implement it.

I'm also thinking about adding libgphoto2 to my project, and using python to interact with a DSLR.  This would allow for much more flexibility (and no fumbling with scripts).  I'll play with it a bit and see if it can't be worked in.

Also, your timelapses are looking good!  You will have problems if you try to use two connected DSLRs with the sample script (take-snapshot.sh) you are using.  It autodetects the camera, and you'd need to specify which camera you are using.  You probably would be better off creating two scripts and using GPhoto2 to specify which DSLR to use in each script.  I can look up the option for you if you want.

Email me at formerlurker@pm.me and I'll hook you up with access to my discord server.  That's probably a better place to chat about various tweaks.  @Shadowen, feel free to email me also and I'll send you a link for discord as well.

FormerLurker, 2018-08-09:
@jetsloth, I added a setting to control the number of rendering threads.  It's in the 'advanced' section of the rendering profile.  I also redid a bunch of the validation, so let me know if you have any problems.

You can try it out by installing from this URL:  https://github.com/FormerLurker/Octolapse/archive/7280a1a28785ea18356cb5737bdec4c848cd442c.zip

jetsloth, 2018-08-10:
@FormerLurker Legend, I've just got a print going ATM so will give that a whirl when it's done.
The print itself is a bit meh in quality, but the lapse is looking very dramatic indeed. I'll post results again when done.

Tried the 2 camera setup last night, but lucked out, my second cannon camera has USB issue so wouldn't auto-detect. Will try again once I've got my camera repaired or access to another. So might email and hit you up for that second bash script if that's cool?

So re the threads, what does this control? Quality?

Xenomorphdelombre, 2018-08-10:
https://youtu.be/M8YzFjFdzS8

Small print captured with my Nikon D40 works very well

Shadowen, 2018-08-10:
Turning the number of threads up to the number of CPUs (cores really)
should improve your rendering speed :)

jetsloth, 2018-08-10:
@FormerLurker OK installing update, will try an updated print with higher quality on camera.

Latest print. Camera battery ran out at the end! Doh! 12 hour print.
**Webcam**
https://www.dropbox.com/s/1tcmvkdscltrjq8/NEW_Elephant_20180810121708.mp4?dl=0

I have no idea how I got the panning effect on the webcam ahahah, something must have been moving it. Looks cool tho, and plays on the other idea we spoke about yesterday.

**DLSR**
https://www.dropbox.com/s/jqb0zvun5sovsyk/NEW_Elephant_20180810121708_1.mp4.zip?dl=0

FormerLurker, 2018-08-10:
@Xenomorphdelombre, your fan is wiggly! Thanks for trying it out!

@jetsloth, that is some HQ video.  Did that print finish?  It looks like it was going to just fall off the bed, but in superb quality.  Looks like you're going to need an adapter for your camera.  I had to get one too.

Hypnopompia, 2018-08-10:
Worked great for me on a Canon EOS T3i!

https://www.youtube.com/watch?v=U_OEwfq-UtM

SrgntBallistic, 2018-08-10:
@FormerLurker 
I'm not sure what's up. It seems weird to me people are able to get it working with a Canon 70D, T3i but I can't with a Canon 60D.

I tried bot the newest posted builds with the same result. The camera snaps photos either every layer or on the set intervals. However, the same error occurs every snapshot - 
`Error output was returned from the snapshot script: [sudo] password for pi: ERROR: Could not get image.`

I did some more testing with manual command to gPhoto2 and found out captures triggered with either
`gphoto2 --capture-image`
or 
`gphoto2 --capture-image-and-download`
never actually saved to my memory card.

So could that be why the snapshot script is erroring out with "Could not get image"?

The only way I've been able to get it to actually save to the memory card is using the following command before capturing an image:

`gphoto2 --set-config capturetarget=1`

Not sure if that could have anything to do with my problem or if I've just somehow misconfigured something...

I have a 7D at work I could try with.

jetsloth, 2018-08-11:
@FormerLurker lol yeh the print finished but camera didn't last the distance. Yeh will need a power adapter. And yes very bad bed adhesion on that print! FML.

OK so I upped the threads to "10" and the camera res to 5472 × 3648 
https://www.youtube.com/watch?v=X0UPoYJGm4g&feature=youtu.be

It's such high res my MacbookPro actually skips frames on playback and kills the RAM hahahaha.

FormerLurker, 2018-08-11:
Wow, that looks great!  FYI, 10 threads probably won't give you optimal performance.  If you're using a Pi3B+ you have 4 available cores, so 4 would be the max here.  If you run with more threads than that you're going to probably lose performance due to thread switching overhead.  However, I wouldn't print anything if you're rendering with all 4 cores :)  I'm not sure if 3 or 4 would give you better performance, but that sounds like a fun test!

jetsloth, 2018-08-12:
@FormerLurker sweet awesome thanks.

> You can now send any arbitrary gcode via a custom camera gcode script (multiline should work too). If you can get it to work by sending gcode to the printer, I think this should cover it.

Is there a dev version I need to update to?

Lavarus, 2018-08-14:
I've got a question.  I have a Nikon 5300, I'm trying to setup via USB to pi 3b.

I have the latest Octoprint and lapse running; I've installed gphoto2 and the script.  However, my camera will not display in Control webcam streaming or in Octoprint, so I'm not sure what I'm doing wrong.  I do have the green gear in Octoprint, but no display of video.
[image]

In the script do we leave the parameters as $1, $2, etc, or change them?

FormerLurker, 2018-08-14:
@Lavarus, 

>However, my camera will not display in Control webcam streaming or in Octoprint

You won't get any streaming from your DSLR.  Also, the 'Welcome to Octolapse' image is a placeholder until a snapshot is taken.  Have you tried running a print with Octolapse enabled?  Once your camera takes a picture the 'Welcome to Octolapse' image should change to the latest DSLR capture thumbnail.  I'm planning on changing that placeholder to something a bit more reasonable so as not to confuse anyone.

Also, have you gotten your webcam to work with Octolapse at all?  If not I suggest starting there and running a few test prints.  Let me know if you need any help with that.

If you HAVE gotten your webcam working and know how to run a test print, please use the Test Mode Full Diagnostic profile, run a very small print (smaller the better else the log file will be huge), and if you don't get any snapshots send me your log file.

Thanks for posting your issue!

FormerLurker, 2018-08-14:
@Lavarus, oh, and you won't need to change the $1, $2 etc..  those are arguments passed into the script by octolapse (argument 1, argument 2, etc) copied into a variable for convenience and legibility.

Lavarus, 2018-08-14:
I don't have a webcam unfortunately.

I tried a test print, with no luck.  Octoprint says waiting to take a photo, and fails to render.
[image]

Does this look right?
[image]

FormerLurker, 2018-08-14:
@Lavarus , Can you post your plugin_octolapse.log file for me?  You can download it from the 'Log' settings page within the Octoprint settings.  Also, it looks like your nikon is setup to deliver raw (.nef) images.  Switch to jpeg within your camera, since that's the only source image format Octolapse currently supports.  Start out with a lower resolution and quality since you have to do some special things to get huge image files to work in a reasonable amount of time.

Also, you are running the script without any parameters, so it won't know where to put the resulting image or what to name it.  You don't really need to do that anyway.

Also, while a print is running (test mode is fine) and after a few layers, take a snapshot of the 'Position State' info panel (I see you have it enabled).  There could be other reasons (start gcode) that Octolapse is not triggering a snapshot.

Lavarus, 2018-08-14:
Newb mistake by not switching camera to Jpeg.  Done and lowered the resolution.   Still not working.

[image]

[octoprint.log](https://github.com/FormerLurker/Octolapse/files/2287696/octoprint.log)

FormerLurker, 2018-08-14:
Sorry, I need plugin_octolapse.log not octoprint.log.  It looks like your start gcode is correct though, so that's great!

Lavarus, 2018-08-14:
Okay
[plugin_octolapse.log](https://github.com/FormerLurker/Octolapse/files/2287741/plugin_octolapse.log)

FormerLurker, 2018-08-14:
That's the right file, but it doesn't look like you were using the 'Full Diagnostic - Test Mode' debug profile in Octolapse, so the log is mostly empty.  Go ahead and change your debug profile, print a few layers, then cancel and send me the result (it will be MUCH MUCH bigger).  Be sure to switch back to 'Test Mode' when you are finished.  No reason to create more logs than necessary!

I wish the debug process wasn't as tedious for everybody.  I'm investigating ways of delivering better error messages, but I suspect your problem is within the .sh file, and I'm still working on proper logging here.

Also, can you let me know when you installed the plugin and from what URL?  This could be helpful info.

Lavarus, 2018-08-14:
I just got the Pi, and Octoprint yesterday; Octoprint plugin from the repository.  I got the Octolapse plugin from the zip you linked far above.

Here is full diagnostic:
[plugin_octolapse (1).log](https://github.com/FormerLurker/Octolapse/files/2287780/plugin_octolapse.1.log)

FormerLurker, 2018-08-14:
Can you try installing from this URL and trying again:  https://github.com/FormerLurker/Octolapse/archive/eae5c49afe88ed2c3a4d490a92746b0f6b761e45.zip

It looks like the snapshot worked by looking at the logs, but there were some problems with error logging that (I think) have been fixed in the above build.  You should see camera errors display in a popup window if they occur.  Let me know the results of your next test, and post your log after it's complete.

I think we're getting close!

FormerLurker, 2018-08-14:
@Lavarus, I forgot to mention that you need to reboot and clear your browser cache (usually ctrl+f5) before the new version is completely installed and ready to use.  Otherwise the javascript may be stale, and we don't want that!

Lavarus, 2018-08-14:
[image]

Can't install without pip?

[27 more replies]

closed 2019-10-12 by FormerLurker