FormerLurker/Octolapse #188: [Feature] Image comparison for part failure detection

paukstelis, 2018-07-09:
Looked through the feature requests the other day and didn't see this, so here it goes. A very nice potential feature for Octolapse would be to do contiguous snapshot comparison to detect possible part failures. If some threshold of image difference is met between snapshots (in a user defined region?) the print is either paused or some type of alert is sent. Used in conjunction with CancelObject or ExcludeArea plugins this could save a lot of prints.

#### Version of Octolapse
 
Octolapse Version: Current

#### Version of OctoPrint

OctoPrint Version: 1.3.8/9

FormerLurker, 2018-07-09:
I'm actually starting to plan such a plugin, so this is a pretty good request.  It will probably be a separate plugin, however, and will involve the camera or multiple cameras.  It's a daunting project (makes Octolapse look like a walk in the park), but one with huge potential.

paukstelis, 2018-07-09:
Sounds good. I would think you have most of the infrastructure there now to do a basic implementation. Maybe just fire off a comparison thread after a successful snapshot is returned? Give the comparison class the ability to call a printer.pause which it looks like Octolapse knows how to handle.

paukstelis, 2018-07-13:
I've manage to hack up a version of Octolapse that sends push notifications (with a modified version of Gina's pushbullet plugin) when it detects significant differences (3 standard deviations from mean diference) from ssim comparisons of images. There are false positives when there are transitions between infill and when solid layers get put down. This depends a great deal on camera angle, but it does seem to work reasonably well in my tests where I throw some other object onto the print bed. It could perhaps be better with some more sophisticated image comparison methods like edge detection or whatnot.

FormerLurker, 2018-07-13:
Sweet, I'll check it out!  Make sure to pull changes from the devel branch, else it won't work with Octoprint 1.3.9.

If this ends up working well I'd love to get a pull request :)

Thanks for all your effort!  It's overwhelming, really.

paukstelis, 2018-07-13:
Its not anywhere near ready for a pull. I am a bit out of my league, as Octolapse is the most complicated thing I've tried to even tried to look at. I just slapped everything into the __init__ on_snapshot_end since I knew what was going on there. It needs its own helper class with threading, particularly if there ends up being more complicated comparisons taking place (I just found something simple I had used before). There will be a fair bit of settings options as well and I don't fully understand how those are laid out in octolapse just yet, so I might have to defer to those that understand it...

FormerLurker, 2018-07-13:
On snapshot end is already on its own thread I believe, but I need to check.  Just put it all in its own file and I'll add a hook for you.  I could even send some info, like the two images to compare, if it will help.

FormerLurker, 2018-07-13:
Oh, and the settings are ridiculous.  I'm working on simplifying them.  Let me know what settings you are thinking about adding.  It may be that this code belongs in its own plug-in if the settings are too crazy, but that wouldn't be too difficult.

paukstelis, 2018-07-13:
I don't think the settings will be too ridiculous at this point. Long term, I'd like to look into integrating with my CancelObject plugin. The idea being that the user can define some ROIs and assign those to a known object in the gcode. If the changes in the snapshot occur in one of those ROIs the object could be automatically cancelled. This would be taking it a bit away from being just a plugin for taking cool timelapses, but I think this is a feature users would like to have.

I'm on vacation next week so I may have time to clean things up enough to put into a test branch.

Shadowen, 2018-07-20:
I've been away for a while so I don't exactly know the state of things here, but could we calibrate the camera intrinsics + extrinsics, then render an estimation of the partial object from the gcode (or STL) to check against? This would rely on a stationary object relative to the camera (which we should have).

paukstelis, 2018-07-20:
@Shadowen that has been done [before](http://www.mdpi.com/2504-4494/1/1/2/pdf). They have a few other similar papers as well. Personally, I think that is a rather complicated solution for what should be a fairly simple question (did the print fail). Also, supports would complicate things since they won't be part of the STL. Maybe if you render against gcode, but that again sounds like a fairly complicated/resource intensive thing to do. The downside of the implementation I started on is that there are false positives, though it just means you get some extra pushes on your phone.

FormerLurker, 2018-07-20:
Yeah, if we had a way to turn a single layer into an image that could be useful, though maybe not as useful for the method that @paukstelis is currently toying with.  Both the classifier and the image comparison method have some inherent issues that will need to be worked out.   Also, I think a rendering generated from the gcode would preferable since we would only need one source file, and stl rendering would be very complicated to match up with the current layer.  

Honestly I think that multiple approaches will be necessary.  I've been working on creating a classifier, and I think a layer rendering AND traditional image comparison would be extremely helpful for this if we wish to have a high accuracy rate.  This way we could have three data points:  the current snapshot, a rendering of the gcode, and an input from @paukstelis image comparison (std deviation).  The output could be some 'layer is successful' confidence (0.0 - 1.0).  We could also take an input from @paukstelis's image comparison (std deviation) to get another valuable point of training data.

I'm sharing timelapse data with another group who is working on a similar project, and they've agreed to share data with me in return.  Training data is very hard to collect, but if we get it and add the appropriate meta-data, I think we could really be on to something.  The first step may be creating a program that can read in data files and output a training file.  I'm using [scikit learn](http://scikit-learn.org/stable/index.html) right now, and would expect our training file format to be supported.

What do you two think?

FormerLurker, 2018-07-20:
@paukstelis, wow that paper is crazy!  Very interesting.  I see what they are trying to do, but I feel it is way too complicated for the average user to handle setup and calibration.  It would def be possible if a mfg could embed the camera into known positions within the printer body itself.  It also looks like it takes a lot of processing time, but that could be solved.

I think what is needed is a one camera solution, just like yours.

FormerLurker, 2018-07-20:
OH, one last thing.  I don't think rendering from GCode needs to be very resource intensive.  We just need to render one layer at a time and build up an image as we go.  I know it's been done before since my slicer can do it pretty quickly for an entire gcode file.  Maybe we can just find an open-source algorithm?

paukstelis, 2018-07-23:
[Here](https://github.com/paukstelis/Octolapse/tree/devel-compare) is a basic implementation merged into the devel branch. I've disabled the tempsnapshot directory cleanup so I could keep individual frames to try other image comparison techniques. I've just started sprinkling in some things for thinking about doing user-defined ROIs. The push notifications will only do something if you use my fork of the Pushbullet plugin.

FormerLurker, 2018-07-23:
This is great!  I'll pull it down and try it out asap.  So cool :smiley:

paukstelis, 2018-07-23:
FYI, I have done a little experimentation and because SSIM comparison uses greyscale images, the color of the filament with respect to the background makes a difference. The other thing I have been doing to avoid false positives is to put the camera view parallel to the build plate. This minimizes large changes that can be detected when switching infill directions or when solid layers get put down.

FormerLurker, 2018-07-23:
Yeah, I can see how that might make a difference!  I like how nicely you've isolated the bulk of you're code into the FrameCompare class.  Have you thought about what settings you need and where they should go?

paukstelis, 2018-07-23:
I put the most basic ones in the init of framecompare. I'm sure there will end  up being a few more. It will be nice to make the two std. dev. thresholds dynamic, so the user could adjust them if they are getting too many false positives during a build. This will require some kind of callback on settings change. For debugging/testing it might be nice to also have a small overlay on the snapshot thumbnail that shows last SSIM score, mean, std. dev., etc. I've just been watching the logs.

FormerLurker, 2018-07-23:
If the settings are live you should be able to extract the latest settings without a callback by passing the settings object to your constructor.  Then you could use the settings object when you need them, and any changes will be returned after they are made.

paukstelis, 2018-07-24:
Cool. That will be useful. I made a few modifications after some testing last night. I've made the mean and standard deviation calculations into a sliding window and only flagged bad frames if the ssim score was below the mean. I found with long/big prints I was getting false positives because early scores were more different because of camera angle. When images started changing less and scores were higher it was above the std. dev. threshold, but because the images were more similar. Should have thought of that earlier....

FormerLurker, 2018-07-24:
That makes a lot of sense!  So it sounds like you also need a setting to control the size of your window.

I'll look at updating Octolapse to better accommodate your plugin.  I was thinking that I could send the 'beep' gcode periodically if your plugin detects a problem.  That way it will work even without the Pushbullet plugin. It's pretty easy to send a popup window to the clients too, though you might be doing that already (haven't run it yet).

paukstelis, 2018-07-24:
I would really like to allow the user to define one or more regions of interest that will be checked on each frame. I'm afraid my javascript is even worse than my python and I have no experience with html5 canvas, so that may take a while. 

Beep and pop-up sound like a good idea to include before asking some other people to try it out. At this point I've only used it on one printer with one camera. I'm sure there will be many other issues to consider.

FormerLurker, 2018-07-24:
I want to include region selection too (rects and circles) for my position restriction feature.  Maybe we can combine forces.

I'm thinking more and more that your code should be separated into a new plugin.  That way you could have your own tab and simplified settings.  I can create a plugin messaging hook to allow you to communicate with Octolapse, which would really simplify your code.  What do you think?  There is a guide that Gina wrote on the OctoPrint document site that gives really good instructions on how to start a new plugin package.  Let me know if you're interested, and I will help in any way I can.

paukstelis, 2018-07-24:
I'm fine with splitting it off. It will have to get snap completed, snap count, as well as all the path information to the snapshots. I've done a few plugins before so I'm familiar with the setup.

FormerLurker, 2018-07-24:
Of course you have written plugins, duh....  Sorry, been scatter brained lately.

paukstelis, 2018-07-24:
I wonder if it would be easier to expose a [helper(s)](http://docs.octoprint.org/en/master/plugins/helpers.html)? I haven't used them before. Not sure if there are advantages/disadvantages to using messages.

FormerLurker, 2018-07-24:
Oh, that's really cool!  Definitely better than messaging for our purposes, and pretty simple to add!

FormerLurker, 2019-10-12:
So this is a pretty old issue now that I've been neglecting.  I have started to add some helpers/hooks to Octolapse, and if you'd like to see one in particular, please create a new issue and i'll add it to the list.  Thanks for your post, and for your contributions!  Don't hesitate to send me any questions or ideas you might have!

For now I'm going to close this issue, but I can easily reopen it if you'd like.  Thanks again!

closed 2019-10-12 by FormerLurker