FormerLurker/Octolapse #189: [Help] Can not get any snapshots to take

RonnieBlaze, 2018-07-12:
#### Version of Octolapse
Octolapse Version: 0.3.1

#### Version of OctoPrint
OctoPrint Version: 1.3.8

#### When you ran into the problem, did you have diagnostic logging enabled?
Diagnostic Logging was Enabled: NO

#### What were you doing when the problem occurred
1. Printing

#### What should have happened?
Timelapse should have started taking pictures.

#### What happened instead?
It never starts taking pictures, Sits at "welcome to octolapse"

#### Operating System running OctoPrint and Octolapse
OS Name: OctoPi
Os Version: 0.15.1

#### Printer model & used firmware incl. version
Printer Model: CR-10 with auto bed leveler
Printer Firmware Version: TH3D_UFW_U1.R1.8j

#### Browser and version of browser, operating system running browser
Browser: Firefox 60.0.2
Browser OS: Windows 10

#### Link to the gcode file you were printing when the problem occurred
Link to Gcode File: https://pastebin.com/qA6pP0Mi

#### Link to settings.json
Link to settings.json with all passwords removed: https://pastebin.com/3gxqibbv

#### Link to plugin_octolapse.log
Link to plugin_octolapse.log:  https://pastebin.com/j1AYE5c7

#### Link to octoprint.log
Link to octoprint.log:  https://pastebin.com/FTUqQPFv

#### Screenshots and/or videos of the problem:
Screenshot/Video Links: 
https://screenshots.firefox.com/BgyxwMJdWPH4Tlpc/192.168.101.21
https://screenshots.firefox.com/juQcCsuAbWWdSwSh/192.168.101.21
https://screenshots.firefox.com/yAuZ3xJl58xO40Hq/192.168.101.21
https://screenshots.firefox.com/3SgKktmT1g9WqgIX/192.168.101.21
https://screenshots.firefox.com/KtL6WiJrPoKNDLJg/192.168.101.21
https://screenshots.firefox.com/2ATB16lCpbOab1AK/192.168.101.21

*** Starting G-Code in S3d
G29 ; auto level
G4 S10; wait for heaters to recover
G1 X0 Y20 Z0.2 F3000 ; get ready to prime
G92 E0 ; reset extrusion distance
G1 F200 E25 ; extrude filament
G92 E0 ; reset extrusion distance again
M420 Z1 ; fade ABL off after 1mm

M201 X3000 Y3000 Z3000
M566 X1200 Y1200 Z1200

G1 Z15.0 F6000 ; lift Z
M204 S800 ; Acceleration
M205 X10 Y10 ; X/Y-jerk

FormerLurker, 2018-07-12:
Hi!  What's going on is that Octolapse does not think your printer is homed, though the G29 probably does home the axis.  Unfortunately the position state info panel is not making this clear since X in this case means 'Not Homed'.  I've added a brand new issue (#190) to deal with this issue.

Unless there is a reason you don't want to home the axis in the traditional way (and there may be), replace:

`G29 ; auto level`

with

```
G28 ; home x y and z
G29 ; auto level
```

As always when adjusting the start gcode, be sure to watch your printer very closely during the startup process to make sure nothing unexpected happens.  It is possible that your printer will perform mesh bed leveling twice depending on how it is configured.  On a Prusa, one uses G28 W; to prevent mesh bed leveling when homing, but I'm not sure about the CR-10.

If I understood the G29 command and how it works with other printers I could add this to the list of commands that can home the X Y and Z axes.  I'll look into this so that others don't have this same issue.

Thanks, and let me know if it works or if you need any additional assistance.

FormerLurker, 2018-07-12:
Oh, and thank you for filling out the issue form so completely!  That's why I was able to diagnose and solve this so quickly :)

RonnieBlaze, 2018-07-13:
Adding the G28 before my G29 auto bed level command did fix the problem of getting it start taking snapshots.  The only problem now is the timelapse looks like a normal timelapse you can still see bed movement.  

https://youtu.be/FVG0eGZEhlU

My guess would be change the snapshot delay to a higher value.

FormerLurker, 2018-07-13:
Ahh, yes.  There is an issue with absolute extrusion and g92 commands if the printer returns a location after the g92.  The solution is to either slice with relative extrusion, or to upgrade to the devel branch.  [Check this thread out](https://github.com/FormerLurker/Octolapse/issues/149), you'll find more detailed info towards the bottom.

FormerLurker, 2018-07-16:
@RonnieBlaze, if you are still having problems with jitter, please try installing from this URL via the plugin manager:  https://github.com/FormerLurker/Octolapse/archive/7109901b92ec27d6c51b119680f038dd660a5b3d.zip

Note that this branch is development, so should be removed as soon as the next release drops.  Let me know if you still have any problems.

FormerLurker, 2018-07-23:
@RonnieBlaze, not sure if you've fixed this yet, but the issue should be taken care of in this release candidate:  https://github.com/FormerLurker/Octolapse/archive/v0.3.3rc2.zip

I'm going to close this issue.  If you have problems with the rc please comment [here.](https://github.com/FormerLurker/Octolapse/issues/198)

Thanks!

closed 2018-07-23 by FormerLurker