FormerLurker/Octolapse #191: Enhance PrintStart detection

FormerLurker, 2018-07-13:
Print start detection currently relies on some gcode tags from Octoprint that could be modified during further development.  Attempt to use PrinterInterfaceListener to start a timelapse as suggested by @foosel in [this OctoPrint issue](https://github.com/foosel/OctoPrint/issues/2718).

FormerLurker, 2018-07-13:
@foosel, I'm having trouble determining which hook to use for this, so maybe you could point me in the right direction when you have some time?  Thanks!

foosel, 2018-07-16:
I'm an idiot.

I was just about to look up the info for you (it would not have been a hook btw, but rather the `PrinterCallback` interface combined with `self._printer.register_callback(self)`) when I noticed that it won't work this way after all because the printer callback gets rate limited by the state monitor, so you might still miss stuff/not get it soon enough :/

I guess I'll just add a hook for the state change in the comm layer, that should take care of that then way more easily.

FormerLurker, 2019-10-12:
Somewhat closed :)  I may revisit this in the future since the print start stuff is pretty overly complicated, but it's working ATM, so..

closed 2019-10-12 by FormerLurker