FormerLurker/Octolapse #193: Please create a profile for delta Printers, in particular Anycubic Kossel Linear Plus.

mmgarcia30, 2018-07-16:
If this is a feature request describe it here
Please create a profile for delta Printers, in particular Anycubic Kossel Linear Plus.

Version of Octolapse
Octolapse Version: 0.3.1

Version of OctoPrint
OctoPrint Version: 1.3.8

When you ran into the problem, did you have diagnostic logging enabled?
Diagnostic Logging was Enabled: YES_OR_NO

What were you doing when the problem occurred
STEP_ONE_GOES_HERE
STEP_TWO_GOES_HERE
STEP_...
What should have happened?
PUT_YOUR_DESCRIPTION_HERE

What happened instead?
PUT_YOUR_DESCRIPTION_HERE

Operating System running OctoPrint and Octolapse
OS Name: OS_NAME_GOES_HERE
Os Version: OS_VERSION_GOES_HERE

Printer model & used firmware incl. version
Printer Model: PRINTER_MODEL_GOES_HERE
Printer Firmware Version: PRINTER_FIRMWARE_VERSION_GOES_HERE

Browser and version of browser, operating system running browser
Browser: BROWSER_VERSION_GOES_HERE
Browser OS: BROWSER_OS_GOES_HERE

Link to the gcode file you were printing when the problem occurred
Link to Gcode File: GCODE_FILE_LINK_GOES_HERE

Link to settings.json
Link to settings.json with all passwords removed: SETTINGS_JSON_LINK_GOES_HERE

Link to plugin_octolapse.log
Link to plugin_octolapse.log: LINK_GOES_HERE

Link to octoprint.log
Link to octoprint.log: LINK_GOES_HERE

Link to contents of Javascript console in the browser
Link to javascript console output: LINK_GOES_HERE

Screenshots and/or videos of the problem:
Screenshot/Video Links: LINKs_GO_HERE

FormerLurker, 2018-07-17:
Can you send me some sliced gcode for that printer?  Please use the default slicer settings so it has a higher chance of working with other Anycubic Kossel Linear Plus printers.

There are some commands that deltas use that are not yet understood by Octolapse, so if your code contains any of these it may take a bit longer to get working.  Also, I can't really test things out so you'll need to be my guinea pig :)

mmgarcia30, 2018-07-21:
Hi mate

No problem on be your Guinea Pig. Here you have a simple spiral vase gcode. remember that deltas has diferent cinematics and origin (0.0.0) it's the center of the round glass.

Regards

I attach both, gcode and STL.
[Spiral_Vase.zip](https://github.com/FormerLurker/Octolapse/files/2216334/Spiral_Vase.zip)

FormerLurker, 2018-07-22:
Can you slice me something that includes lift and retract?  Vase mode gcode doesn't include a lot of this stuff.  Also, if you can attach the default slicer profile?  FYI, I see no reason why your printer won't work in the current version of Octolapse.

Octolapse doesn't support circular beds, but you can trick it into working by using a square that encompasses the whole circle.  Unfortunately, that means some of the stabilization presets will try to move out of your print area.  You'll have to use the 'centered' preset for stabilizations until I can work in the circular out-of-bounds routines.

g137, 2018-07-26:
FormerLurker, did you get the file you were looking for? I too am excited to having a Anycubic Kossel Linear Plus printer file for Octolapse. It's the bee's knees!

PeggyFree, 2018-08-22:
I plus that demand.

FormerLurker, 2018-08-22:
@PeggyFree, @mmgarcia30, and @g137:

Sorry it's taken so long, but I'm currently putting all of my time and energy into getting the next version completed.  However, I did take the time to create a video explaining how to create your own custom printer profile.  If any of you has time, maybe you can run through the video and try creating a profile?  If you do, please then export your settings.json and send it to me.  You can export settings by going to Octoprint Settings (wrench at the top), selecting Octolapse on the left, then clicking 'Download Settings' button.  Once we have a profile that works I'll add it to the defaults.

[Here is a link to the video](https://www.youtube.com/watch?v=YsxjQhWKz5w&t=1s)

There may be some slight differences between what you see on the video and what you see within your printer profile depending on what version you are running exactly.  In fact, that video is already out of date when compared to the development version (lots of features were added since), but I plan to update it soon.

I hope this helps!

Canyonazo, 2019-03-14:
Hello, is there any unofficial version for delta printers?

mtroumelin, 2019-04-17:
Here is my settings that i use with my delta printer (220mm of diameter), of course the settings are set to my Simplify3D settings :)
Work like a charm :)
Thanks for this very nice plugin!

Basically, all i did was to create a new printer, change the outbox settings to have both negative and positive bounds, and change the "Fixed Extruder at center" settings to go to the fixed position 0,0 (not sure this last part was necessary).
Hope it'll helps

[Settings.txt](https://github.com/FormerLurker/Octolapse/files/3088308/Settings.txt)

FormerLurker, 2019-10-12:
So, you'll all be pleased to know that I've added full support to deltas/circular bed/center origin printers to Octolapse starting with V0.4!  This is still unreleased, and I do not have a profile yet for the Kossel linear plus.  However, if anyone on this thread wants to try the new version and export (another new feature) the printer profile,  I'd be more than happy to include it in the next release!  Message me if you would like to do this and i'll send instructions.  For now I'm going to close this issue and will create a new one if anyone here is willing to help.

Thanks!

closed 2019-10-12 by FormerLurker