FormerLurker/Octolapse #199: Excessive error messages blocking UI

Chaos99, 2018-07-22:
Diagnostic Logging was Enabled: NO

#### What were you doing when the problem occurred
Printing gcode that was not properly formatted to include correct triggers for octolapse.

#### What should have happened?
Warning/error, then no timelapse taken.

#### What happened instead?
Excessive warning messages cluttering the screen, blocking input to OctoPrint, then bringing the webserver down due to excessive load.

#### OctoPrint Version
OctoPrint 1.3.8

#### Operating System running OctoPrint and Octolapse

Raspian with OctoPrint compiled from Github.

OS Name: Raspian
Os Version: 8 (jessy) Kernel 4.9.35-v7

#### Printer model & used firmware incl. version

Printer Model: Makerbot-style custom build
Printer Firmware Version: Marlin 1.1.8

#### Browser and version of browser, operating system running browser

Browser: Firefox Quantum 61.0.1 (64-bit)
Browser OS: Linux, Manjaro (Arch-based)

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: https://pastebin.com/Eva5fs9K

#### Link to settings.json

Link to settings.json:

{"auto_reload_latest_snapshot": true, "current_printer_profile_guid": "2badba3c-f737-4391-a261-04dbc294a4af", "default_printer_profile"$

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  
approx. 85k lines of
2018-07-22 09:25:49,935 Position - Coordinates (X:None,Y:None,Z:None,E:0.00000) are out of the printer area!  Cannot resume position tr$
2018-07-22 09:25:49,956 Position - Coordinates (X:None,Y:None,Z:None,E:-5.00000) are out of the printer area!  Cannot resume position t$

#### Link to octoprint.log

Link to octoprint.log:  https://pastebin.com/yBtawxV5

#### Link to contents of Javascript console in the browser

Link to javascript console output: https://pastebin.com/vCxHJEwk

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: https://imagebin.ca/v/49YO8ej3Hedv

FormerLurker, 2018-07-22:
Sorry about that.  I'm aware of that popup issue, and am reworking the popups so they do not explode like that.

The problem is this line:

;G28 ; home all axes

Your home command is commented out, which I assume was done for good reason.  Unfortunately Octolapse won't work without this command.  You are one of 2 people I've seen who does not home before a print, so it's been a lowish priority to figure out a workaround.  I'll consider working it in sooner.  Uncomment this line and you won't get all those errors.

Second, I see that you are using gcode triggers  (snap command), but the command is commented out.  OctoPrint strips all comments, so Octolapse will never see those.  Change:

;snap

to

snap

and you'll be all set.

Please let me know the circumstances behind removing the G28, and I'll try to think up a workaround.  Octolapse uses this command to determine a known starting point for position tracking.  Look at the Position State info panel (octolapse tab, enable the slider under Current Settings).  That panel has been greatly enhanced in the latest release candidate of Octolapse, located here:  https://github.com/FormerLurker/Octolapse/archive/v0.3.3rc2.zip

Feel free to upgrade via the plugin manager, but first you MUST upgrade octoprint to the latest release candidate.  You can do this easily within the software update plugin (settings, software update) by clicking the wrench icon, and changing the OctoPrint version tracking to 'Release' and the OctoPrint Release Channel to 'Maintenance RCs'.  Update when prompted, then reboot when prompted.  Next install Octolapse from the URL above and reboot again.  Finally refresh your browser cache via ctrl+f5 after octoprint loads up.  You will now see the newest version number (0.3.3rc2) in the octolapse tab and will have access to the updated position state info panel (and lots of other features)!

Let me know if that helps!

FormerLurker, 2018-08-06:
FYI, I added rate limiting to the infamous 'out of bounds' popup error.  I will let you know when it's pushed out.

Was the last message I sent helpful to you at all?  I'll wait a bit for your reply before closing out the issue.

Chaos99, 2018-08-07:
Hi there, sorry for the missing and ultimately late answer. Especially as that was the most detailed, helpful, friendly and to all that fastest answer to a bug report I've ever received on Github (or any other platform for that matter). So although I'm happy with the workaround and you already pushed a fix, here are some of the details you asked for:

**Why is the home command commented out?**

My printer is very unreliable from time to time and I suffer from clogged nozzles or x/y shifts. But I usually don't throw away the half-finished prints. I just file or sand down the top failed layers with the model still attached to the bed and continue the print from there. I determine the start height by manually moving the nozzle right above the start layer and then get the current z height (M114).

Unfortunately, I know no printer software or plugin that let's me restart at a given layer (or height), so I just delete the finished layers from the gcode and restart. If I keep the G28 on startup in, it will push the nozzle into the model by trying to home z or after that when traveling x/y without raising z again. I therefore removed it even from my template, as I too often forgot to remove it during the edit. It's easier to remember to home manually when I turn on the printer.

Also, in case of x/y shifts, it might be easier to continue in a shifted coordinate system in case of shifts in one of the very first layers instead of loosing that precise shift when homing again.

**Why is there a commented out snap command.**

Well, there isn't. It's not a command at all. It IS a comment telling me were I assumed the snapshot would take place. I didn't know that I could trigger the snapshot with a command. 
I looked at the options again after you told me but I still had to look twice to find the command definition under printer settings. Also I was puzzled by the 'on zHop only'. (How does it combine with 'on command?').

**Why I ultimately disabled OctoLaps on my printer:**

The way OctoLaps needs to determine the absolute position is a mystery to me. It seems to have various forms of auto-detection, but without actually knowing what the system is looking for I don't dare to feed it my often hand-crafted, custom gcode in fear of crashes, error messages from hell or just unpredictable head movements. With auto-detection disabled, it still needs a G28 command (how is this not 'auto' too?) which I sometimes don't have.

What I'd need is a way of detecting layer changes with a custom regex. They are easy enough to spot.
And giving a fixed x/y coordinate to place the extruder during snapshots. 
If those options are in there, I couldn't find them within all those options.

closed 2018-08-07 by Chaos99

FormerLurker, 2018-08-07:
Thanks for your response!  I do try to be prompt.

> Unfortunately, I know no printer software or plugin that let's me restart at a given layer (or height), so I just delete the finished layers from the gcode and restart. If I keep the G28 on startup [...]

I don't know of any either.  You are a brave soul for trying this :)  I agree, a G28 would kill the print/printer.

>Well, there isn't. It's not a command at all. It IS a comment telling me were I assumed the snapshot would take place. I didn't know that I could trigger the snapshot with a command.

Wow, what a coincidence!  I have received more than one issue where the user assumed the snapshot command needed to be commented to work, since it's not a valid gcode.  Octolapse suppresses the snapshot command anyway, so it shouldn't cause any issues.  I agree that the command should probably not be in the printer profile.  It's a relic from a previous version and shifting priorities.  Not many people use the gcode trigger (which was the easiest one to write..  lol!).

>Also I was puzzled by the 'on zHop only'. (How does it combine with 'on command?').

It works the same way as with any other trigger, though it does seem totally unnecessary doesn't it?  It does prevent situations where someone puts the 'snap' command in a totally inappropriate place, but if you're using the gcode trigger you probably know what's going on!  Turns out removing this 'functionality' was harder than just making it match the other two triggers, so in it remains.  Maybe when Octolapse is finally out of beta I'll have this cleaned up.

> The way OctoLaps needs to determine the absolute position is a mystery to me. It seems to have various forms of auto-detection, but without actually knowing what the system is looking for I don't [...]

Well, the barrage of error messages is fixed in the latest devel branch commit.  The 'Autodetection' in this case is auto origin detection.  Your origin need not be at 0,0,0 (in fact, often it is not) after a G28.  With the autodetection enabled Octolapse sends an M400, then an M114 after certain (configurable) commands, and updates its position based on the result.  The other option is to tell Octolapse exactly where your origin is, eliminating the need for the extra M400+M114.  Believe it or not I added this feature as an afterthought not too long ago, and it seems to have made the manual origin entry obsolete.  I may remove manual origin entry at some point.

The rest of the position detection goes like this:
1.  Octolapse looks for G90/G91 M82/M83 to get the axis mode.  It won't do squat until it knows this, else the printhead will go rampaging about, and we don't want that!
2.  It must receive a home command for all 3 axes.  Initially Octolapse uses x=null, y=null, z=null for its coordinates, but will replace each coordinate value as it becomes known (G28 X; G28 Y; G28 Z would work just as well as G28).  I COULD just use the coordinates provided by the M114 for all axis, even if one hasn't been homed, BUT...  those coordinates may not be accurate, and Octolapse might crash into an endstop or worse.
3.  Once it knows both axis modes and an origin position (either with autodetect or by manually setting the origin), then it will update its internal position every time it sees movement on either X,Y,Z or E axes.  It is even cool with re-homing later in the print (though I did fix a bug with that just yesterday).

I'm trying to think of a way for Octolapse to intelligently query the printer (m400 + M114) when it needs to know a position, but i can't figure out how to get that to work without G90-91/M82-83/G28.  You CAN manually set the initial axis mode from within the printer profile, but this is dangerous.  Even the MK3 start gcode is missing a G90 AND contains absolute movements in the start gcode!  All a user has to do is send a G91 to the printer manually and run their start gcode and you could get an axis crash.

>What I'd need is a way of detecting layer changes with a custom regex.

As you read above, the problem is not really layer detection, but position/axis mode detection.  If you want manual layer detection it's probably best to use the gcode trigger and just put:
'''
snap
'''
in your layer change script.  A good chunk of Octolapse is just detecting layer/height changes on all the different printers using various methods.  Don't worry too much about Octolapse doing unpredictable things.  One of the reasons people have so much trouble getting it running in the first place is that I take the most conservative approach to detecting position and movement, as you saw with the missing 'home' command.  I could be much more permissive (assume a specific coordinate system, allow position detection without XYZ home, etc), but then there is the possibility that Octolapse is wrong, and does unpredictable things.  It's one of the only programs I know of that designs and sends dynamic gcode during a print, so extreme care went into making sure Octolapse knows what it is doing.  The 'exclude region' plugin is another, but as far as I know it only either splits or suppresses gcode, which is a lot safer than trying to move to the very edge of the build platform.

The problem I'm having trying to come up with a solution for your case is that it is such a special case that nothing in Octolapse is designed to work with it.  Perhaps I could add an option that would allow what you are doing to work.  I need to think about it.  For now I'd recommend only using Octolapse from a fresh print (with home).  Hopefully I will find out a way to make setup and position detection easier and more reliable.

Thanks!