FormerLurker/Octolapse #202: [Feature] Reduce flicker with image processing

Shadowen, 2018-07-23:
#### If this is a feature request describe it here
Use PIL to reduce flickering using histogram equalization or some time-series smoothing algorithm (?).

We might need OpenCV for this level of image processing though... is that allowed?

I noticed some problems on this reference video: https://youtu.be/uBeVbDJKHw0.

FormerLurker, 2018-07-23:
You're welcome to try OpenCV.  I have to warn you though, I have reports of installation trouble with Pillow, and hopefully OpenCV won't give us similar troubles.  If it can be done with Pillow, that would be great.  If OpenCV is required, it will probably be worth it in the end for such a useful feature!

Shadowen, 2018-07-27:
Hmm, histogram equalization (individual images) or smoothing the L-channel in LAB (across multiple timesteps) doesn't seem to do the trick for me. Not sure how it's usually done...

Also tried out the `deflicker` filter in `ffmpeg` to no avail. Maybe someone else can try it out if there's interest.

FormerLurker, 2019-10-12:
OK, this is much easier to fix using the new camera controls I've implemented than with post processing.  I did give this a go and didn't have very good results.  I'm sure it is possible, but I think it is now out of scope given the ease of setting manual exposure/focus starting in V0.4.

Closed :)

closed 2019-10-12 by FormerLurker