FormerLurker/Octolapse #203: Developer README

Shadowen, 2018-07-23:
Write a "getting started" guide for developers. It should include:

- Pulling code.
- Configuring and running unit tests.
- Installing OctoPrint
- Installing Octolapse into OctoPrint.
- Running the Octoprint server.
- Setting up the test printer/test profiles for Octolapse.
- Pushing/pull requests.
- A rundown of the release process (?) - I'm not familiar with this one.
- Major branches and their purposes.

FormerLurker, 2018-07-23:
That's a good idea!  Anything that might result in more useful pull requests would be fantastic.  Also, I assume you are working with picharm?  I also got OctoPrint/Octolapse running in visual studio a long while ago, so maybe I can dig up that process too.

My release process is in development.  Here's where I'm headed assuming development is done on the devel branch:

1.  Merge devel into rc/devel - update version number (Vx.x.xrcx-dev) - add a rc/devel pre-release to github
2.  Merge rc/devel into rc/maintenance when ready and remove -dev from version - add an rc/maintenance pre-release to github
3.  Add a github issue for feedback on the rc
4.  Fix any issues on devel branch
5.  Merge fixes into rc/devel and update the version number (vx.x.x.rc(X+1).dev
6.  Work with the users to test the rc/devel branch, adding an rc/devel release if necessary
7.  When issues are fixed, merge into rc/maintenance and remove -dev and add an rc/maintenance pre-release to github
8.  Create a new github issue for feedback, close the previous issue.
9.  Repeat from step 4 until major issues are solved.
10. merge into master and update the version number (vx.x.x) and add a new release to github

I'm not sure if you're aware, but the latest version of Octolapse will track master, rc/maintenance or rc/devel depending on the software update settings!  That should help with debug/testing a bunch.

Shadowen, 2018-07-23:
Yup, I'm using PyCharm. I largely develop using Twisted Trial (unittest) though. It works quite well for the rendering code, since I can generate images quickly rather than running a full timelapse.

I'm not sure how much other people keep up with the unit test code though, since it seems to be permanently broken :P

I don't think I'll make too much use of the actual version tracking since my branches aren't in the main repository, but that's really cool :) I'm on devel/rc right now.

FormerLurker, 2018-07-23:
I need to fix the unit tests :(  I'll work my way through them ASAP, and will simplify what I can.  I'm learning as I go :)

It's fine that you're on devel/rc, though it might be good to merge with and switch to the devel branch since I have some new code in there (external camera script support for DSLR), but I don't think it will conflict with what you are doing.  I may need to use the rc/devel branch to fix issues with rc2.

Shadowen, 2018-07-23:
Page started [here](https://github.com/FormerLurker/Octolapse/wiki/Developer-Guide).

closed 2018-07-23 by Shadowen

FormerLurker, 2018-07-23:
Excellent!  Next time I run through these steps and will likely add some images.  I will also add a link on the wiki menu on the right hand side.

Thanks!