FormerLurker/Octolapse #207: Octolapse creates artifacts

TalZiv, 2018-07-27:
Hi,

First of an amazing plugin idea, gave you a star.

#### Version of Octolapse
Octolapse Version: 0.31

#### When you ran into the problem, did you have diagnostic logging enabled?
Diagnostic Logging was Enabled: NO

#### What should have happened?
The object should be printed without artifacts

#### What happened instead?
The printed object suffers from artifcats, it looks like the nozzle start to extrude a bit prematurely when coming back from the snap shot.

I tried the "center left" and "center" profiles, same result.

when disabling octolapse and printing again the artifacts are gone

#### Operating System running OctoPrint and Octolapse
OctoPrint 1.3.9 running on OctoPi 0.15.1

#### Printer model & used firmware incl. version
Printer Model: Ender 3
Printer Firmware Version: marlin bugfix-1.1.x

#### Screenshots and/or videos of the problem:
https://i.redd.it/m68unf5pn5911.jpg
https://preview.ibb.co/hmQXCT/IMG_20180711_081727.jpg
https://preview.ibb.co/kyYKsT/IMG_20180711_081720.jpg
https://preview.ibb.co/gHBuRo/IMG_20180711_081700.jpg

Thank you for your time,
Tal

FormerLurker, 2018-07-27:
This looks like a settings mismatch between your printer profile and your slicer.  I have been working on a 'print quality' guide that could help.  Try looking over [this brand new guide](https://github.com/FormerLurker/Octolapse/wiki/Improving-Print-Quality), and let me know what settings you ended up changing (especially print profile settings and your snapshot settings), and how that affected your quality.  Also, if you would, please give me some general feedback on the guide.  It is obviously a work in progress, but it's better than nothing!

FormerLurker, 2018-07-28:
@TalZiv, I've been looking at the gcode generation with Octolapse, and I've found some possible issues.  I'm not sure right now if this has anything to do with your quality problems, but I wanted to let you know that I'm working on creating some tests on this code as we speak to verify that everything is working.

TalZiv, 2018-07-29:
Thanks for really digging into the problem, I matched the plugin settings with CURA, I will post a screenshot later and if there are discrepancies let me know and I will fix and test.

TalZiv, 2018-07-30:
Hi,
here is the screenshot: https://preview.ibb.co/k2fY9T/cura_octo.jpg

all the values other the retraction and Z-hop are set to 0.

FormerLurker, 2018-07-30:
@TalZiv, you need to fill in all of those axis speeds.  If they are set to 0 they will use the current speed of the printer, which is very very likely to be wrong.

Set Retraction and Detraction speed = 70 Millimeters per Second.

I can't see any other axis speed values from the screenshot you sent, but I think you'll find them if you look through the 'speed' settings in cura.

Also, 10mm of retraction seems pretty high, but maybe it's correct for your printer.  I'd look into that.

FormerLurker, 2018-08-11:
@TalZiv, let me know if my previous post solved your issues.  I'll wait a bit longer before closing this out.

OutlawAndy, 2018-08-13:
The “Improving Print Quality” guide is great.  The individual guides for each of the settings tabs were very, very helpful as well.  Helped me to dial everything in tight.  Getting great results now. 👍👍

It’s clear to me, based on the quality and depth of the documentation you provide, that you care a great deal about this project.  Your hard work and dedication is very much appreciated!!

Cheers!

Boymeetsmill, 2018-11-06:
**Octolapse Version:** 
    0.3.2

**OctoPrint Version:**
    OctoPrint 1.3.9

**Diagnostic Logging was Enabled:**
    NO

**What were you doing when the problem occurred?**
I was printing as normal.
    1. Download or create STL
    2. Slice STL in S3D (or Cura)
    3. Upload to OctoPrint
    4. Preheat Printer
    5. Print Model
    6. Rinse and Repeat

**What should have happened?**
    1. During the printing when a layer change occurs, the printhead should move to the (Left of right) of the X-axis. 
    2. Printhead movement is paused
    3. Image is captured by webcam
    4. Printing is resumed.

**What happened instead?**
    3a. During the movement of the printhead between paused position and resumed printing the print nozzle oozes material

I'm not sure if the nozzle is extra priming, or prematurely starting to extrude materal before getting the correct position. I have read the guide, and made sure to match all the settings. And gone through the steps of printing smaller files with Octolapse enabled, and with the standard OctoPi timelapse. The standard timelapse has a few print artifacts, but maybe 5% as many. 

**OS Name:**
    OctoPi

**Os Version: **
    0.15.0

**Printer Model:**
    Anet A8 

**Printer Firmware Version:**
    Marlin 0.1.7

**Link to Gcode File:**
https://www.dropbox.com/s/j9w0gw6l6n28125/Wolfhead.gcode?dl=0

**Screenshot/Video Links:**
    1. Output Timelapse Video: https://www.dropbox.com/s/mxt5a433ywv5vto/Wolfhead_20181104152319.mp4?dl=0
	2. Screenshot: Octolapse settings 1   https://www.dropbox.com/s/dkcsg2aapoxhnae/Screenshot%202018-11-05%2021.14.09.png?dl=0
	3. Screenshot: Octolapse settings 2   https://www.dropbox.com/s/9jt25q4nt1oa4eo/Screenshot%202018-11-05%2021.14.24.png?dl=0 
    4. Screenshot: Simplify3D Settings 1    https://www.dropbox.com/s/i5jr03ai9d8bwct/Screenshot%202018-11-05%2021.39.48.png?dl=0
    5. Screenshot: Simplify3D Settings 2    https://www.dropbox.com/s/lpapvahk76i3xm9/Screenshot%202018-11-05%2021.40.13.png?dl=0

FormerLurker, 2018-11-06:
Are you sure your retraction/detraction speeds are correct?  300mm/min is VERY slow.  The artifacts you are seeing are quite extreme (which is, I'm sure, why you posted an issue), and aren't expected.  Can you try printing a stringing test like this one WITHOUT Octolapse, but using the same simplify settings:  https://www.thingiverse.com/thing:607968

Send me a pic of the finished print.  If you get lots of stringing with the test print, you'll need to further tweak your retraction/detraction speed before moving on to Octolapse settings.

Thanks for posting!

Boymeetsmill, 2018-11-06:
I printed the 'Retraction Test' file linked to above with and without Octolapse enabled. Full diagnostic logging was enabled for the octolapse print, but check out the video I recorded with my cellphone. To answer your question above yeah the retraction speed is slow, but it works alright for what printing I've done. In the comparison photo I do have print defects, but whatever is causing this globbing and stringing is making things worse.

**Diagnostic Logging was Enabled:**
Yes

**Link to plugin_octolapse.log:**
https://www.dropbox.com/s/g97yjt2gc42mqle/plugin_octolapse.log?dl=0

**Link to Gcode file:**
https://www.dropbox.com/s/78qqofbv3nuqkjj/retraction_test_v1-0.gcode?dl=0

**Screenshot/Video Links:**

Side by Side of finished prints
https://www.dropbox.com/s/v0hc7tbu9edakfs/2018-11-06%2014.09.04.jpg?dl=0

Close up of finished print done with Octolapse
https://www.dropbox.com/s/iw4gb4pnflz2wlj/2018-11-06%2014.01.29.jpg?dl=0

Video recorded mid-print of oozing during stabilization capture (Octolapse enabled)
https://www.dropbox.com/s/4ntghlofo2w9059/2018-11-06%2012.13.09.mov?dl=0

Octolapse Rendered Timelapse
https://www.dropbox.com/s/tcvwagqdxshjrl2/retraction_test_v1-0_20181106204740.mp4?dl=0

FormerLurker, 2018-11-06:
Yikes, that is really awful.  I really want to get your issue licked, especially since you've provided so much info.

I can see the oozing in your video (thanks for including that, it's incredibly helpful), even before it gets to the stabilization point.  I'm not really all that familiar with the A8, but is that a bowden extruder, or does it just have pfte running into your extruder stepper?  I'm asking because my bowden needs a whole lot more retraction than my direct drive did (0.6mm to 4.0 for my bowden) due to a small amount of play within the pfte tube.  It could be that the retraction settings you are using work find for smaller travel distances.  Also, are you using ABS or PETG by chance?

I suggest trying this:

1.  Backup your simplify settings since we'll be messing with them.
2.  Update all of these values below BOTH in simplify and in Octolapse:
3.  Increase your retraction a bunch.  Try moving from 0.4mm to 4mm (both in simplify and in 
4.  Increase your Retraction/Detraction speed to 1500mm/min
5.  Can you turn your travel speed up safely?  Not sure what the max travel speed is for the A8, but the faster the better for Octolapse.
6.  Maybe try stabilizing in the center to avoid a longer move to the edge?
7.  Verify in your snapshot settings that 'Retract Before Move' is enabled.  It probably is, but it doesn't hurt to check.

Retry the test and make another video (mid print ooze) since that was extremely helpful!  If you get better results when using Octolapse you might want to try the test again without Octolapse and see if things improve there too.

If things do NOT get better (might not be perfect, but should at least improve), perhaps something else is going on.  I'm considering recommending that you upgrade to the newest pre-release version (0.3.4rc1), which has lots of bug fixes.  There is also a new feature that allows you to restrict snapshots to less important areas of your print (infill, interior perimeters, supports, etc).  That wouldn't stop the oozing, but it would reduce the visibility quite a bit.  However, if we can reduce the oozing to a more reasonable amount, the new 'high quality' profile in 0.3.4rc1 can achieve nearly the same quality as a timelapse without Octolapse.  It does come at some cost, namely that it's possible to miss frames in some cases (minimum layer times, or other things that mess with the feature), but IMHO it's a good trade.

Boymeetsmill, 2018-11-06:
Providing as much useful information as you can is usually the shortest way to get the answer you need...

My Anet A8 is pretty much stock, which is direct drive extruder, with a MK8 style hotend. I have the PTFE lined throat, and the prints I've been doing are ABS. 

I'll try your suggestions, but some of my initial concerns are...
1. Turning up my speeds will induce ringing on the outside walls. The frame on this printer is pretty crappy I've done work to make it more rigid, but the longer term solution is to move to a new frame. This includes the travel and jogging especially when the prints are tall and thin.
2. Stabilizing in the center can result in the carriage blocking the webcam, which is why I usually move it left of right.

Upgrading does sound like a decent option. I am super intrigued by the multiple camera, and external triggering additions to this plug-in.

FormerLurker, 2018-11-06:
Valid points.  Definitely try increasing retraction/detraction length and speed.  I too had issues running at max travel speed (stepper skips sometimes).  You definitely know your machine better than I do, so when in doubt, use your better judgement.

Boymeetsmill, 2018-11-07:
@FormerLurker I tried some of your suggestions...

1. Increased your retraction to 1.0mm from 0.4mm (150% increase)
2. Increased retraction speed to 1500mm/min from 300mm/min (500%)
3. Turned travel up to 4000mm/min from 3000mm/min (33% increase)
4. Stabilization remained unchanged, 'Left Center'
5. Retract Before Move is enabled.
6. Double checked all the settings were identical between slicer and plugin.

[Prints with faster settings](https://www.dropbox.com/s/3dp41s4zz6wzt3u/2018-11-07%2000.46.17.jpg?dl=0)

I did another with Octolapse enabled and disabled. I did not get a video, but the results aren't so different that I suspect the video would be either. The quality looks better on the Octolapse print, still globy but more so higher as the part gets smaller up higher. I may(should) print with stabilization in the center to see if the nozzle just oozes into the print or at all. The "control" sample, printed with octolapse disabled, looks a little better too, but maybe on the border of too much retraction. 

Still stumped though as to why the oozing during stabilization. Any other ideas?

As something to try I decided to upgrade to Octolapse 0.3.4rc1.dev3. I got it installed last night along with gphoto2. I noticed when looking through the log file I provided earlier that the webcam was not responding to changes of white balance, contrast, sharpness, or brightness. I edited the required /boot/octoprint files, but still not accepting commands. I haven't run a timelpase with the upgraded Octolapse because of messing with the webcam, will update later.

FormerLurker, 2018-11-07:
I'd love to see what happens with the new version, maybe there really is a bug of some kind?  I hope I didn't already send this link, but take a look at [this video](https://youtu.be/sDyg9lMqMG8) to see how setup works in the new version.  I don't recommend using the 'high quality' snapshot profile for this kind of test, since the layer time is likely below your minimum.  Once we figure this strange ooze issue you can move on to the high quality profile.

The only other thing I can think of is that you could try turning the temp down a bit while printing ABS?  Maybe 5 degree steps until the oozing is minimized but you still get good adhesion.  I'd stick with the higher temp for the first layer or two for sure, since you know that's working.  I've done lots of ABS printing and know how tricky it can be, especially on the first layer.

Regarding the webcam issue, can you manually browse to control.htm?  This would be the full url:

https://IP_OF_PI_HERE/webcam/control.htm

OR MAYBE 

https://IP_OF_PI_HERE:8080/control.htm

I'm trying to figure out a better way control the camera, but for now I'm hijacking that page and am just sending posts to it :)

Boymeetsmill, 2018-12-20:
@FormerLurker Any work on this oozing?

I did some maintenance and upgrades to my machine. CLA including replacing the nozzle. I thought my issues were related to a worn out nozzle, but then I ran a calibration print and saw odd "oozing". I've attached a link to an image of the print.

[Calibration Print Image 1](https://www.dropbox.com/s/co66sf3uiieounh/2018-12-19%2000.46.07.jpg?dl=0)
[Calibration Print Image 2](https://www.dropbox.com/s/zevyyizss927q63/2018-12-19%2000.46.13.jpg?dl=0)
[Timelapse](https://www.dropbox.com/s/x7tm9yuqttnlwnd/Two_Post_Retraction_Test_20181219075138.mp4?dl=0)

Yes, there is oozing. I moved the setup to a full enclosure and now everything stays at much higher temps. My garage is 12C and the box is 28C for PLA and 38C for ABS. Which is why I am trying to recalibrate and optimize. But to me it looks like there is greater buildup on the left. The distance between the two sets of pylons is greater than the distance from the left most pylon and the stabilization parking spot.

FormerLurker, 2018-12-20:
Hey, that's a pretty good test.  Can you send your gcode and settings (exported from Octolapse) for that test please?  I'm going to examine the gcode that octolapse generates to see what the retract and lift gcode generated by Octolapse looks like.  I recently found a bug in the gcode generation routine that can cause some speed issues if the preceding gcode is travel only (no extrusion), and maybe this is related.

Boymeetsmill, 2018-12-21:
@FormerLurker here are the requested files...

[Octolapse settings](https://www.dropbox.com/s/zp0ri5c2xdlq9sa/Oozing_OL_Settings.json?dl=0)
[File Gcode](https://www.dropbox.com/s/7n013lruyhq29qh/Two_Post_Retraction_Test.gcode?dl=0)

I have a suspicion there is something going on relating to the jog, and priming of the extruder (or something similar). My only evidence is the files in the comment above, and the last print I ran was a T-Rex skull for a Christmas present. The finish is awesome, there is almost no oozing, and I oriented that part along the X-axis. So the jogs to the "stabilizing point" were very short.

FormerLurker, 2018-12-21:
So, the first thing I notice regarding your settings is that your simplify 3d retraction distance is set to 2.2mm, and your printer profile within Octolapse is set to 1mm.  It looks like the rest of the settings are correct.  

Please try correcting your retraction distance and try another test.  I'm sure things will improve, but I'm not sure if it will fix the problem.

In the meanwhile, I'll be looking at the generated gcode.  So far I haven't found anything obvious.  I notice you aren't using z-lift, and always use it myself, so I'm going to triple check that process (or rather lack of this process).  My intuition says the code is correct.  In case you are interested, here is an example of the snapshot gcode from a log file I created:
Triggering command:  
```
G1 X163.499 Y100.693 E0.0080 ; from line 1610, this command is not sent until the snapshot gcode is finished.
```
Snapshot Gcode:
```
M83; set to relative extrusion
G1 E-1.00000 F1500.000 ; retract (notice that it's only retracting 1mm)
G1 X2.500 Y103.500 F4000.000 ; move to the snapshot position, a snapshot is taken after this command
G1 X163.287 Y100.630 ; return to the original position
G1 E1.00000 F1500.000 ; detract
M82; return to absolute extrusion
G1 F2250.000; set the print speed to the original print speed (it was changes above)
G1 X163.499 Y100.693 E0.0080 ; The triggering command.
```

My intuition is that the gcode from line 1609 (`G1 X163.287 Y100.630 E0.0021 F2250`)should have triggered the snapshot instead, but I'm not sure exactly why that happened.  Will look into that, but that shouldn't cause issues like you are seeing.

Also, I noticed that your gcode uses a method of retracting/detracting that I don't encounter too often (note that simplify is set to use absolute extrusion):

```
G92 E0 ; zero out E
G1 E-2.2000 F1500; Move to absolute position -2.2
;a little later in the code
G1 E0.0000 F1500
G92 E0; This is redundant, e was just set to 0 above, but it shouldn't mess with anything
```

It's possible that there is some subtlety here that I have not taken into account.

Anyway, let me know how adjusting the retraction affects your print.  Hopefully things get better.

FormerLurker, 2019-01-27:
@Boymeetsmill, I may have found the issue causing your extrusion problem!  I believe that G90/G91 on your printer may override the E axis mode in addition to the XYZ.  I was handling this properly (according to the printer profile settings) inside of the position tracking, but did not take it into account when generating the snapshot gcode!  I may have this fixed now, but I need a tester.  I don't have a machine to test this with, but it looks good in the debugger!

FormerLurker, 2019-01-28:
@Boymeetsmill Use this link in the plugin manager to install:  https://github.com/FormerLurker/Octolapse/archive/new_settings.zip

Warning - Download a copy of your settings.json file through the Octolapse settings page first.  I'm not 100% sure if the settings migration will work 100% (but have been testing it a lot), and you might want to roll back to v0.3.4 when you're done testing.  If you want to roll back, you can install the new version, copy the settings back to /home/pi/.octoprint/data/octolapse/settings.json, then reboot and everything will be back to the way it was!

Also, there are some new features you might also want to play with.  If you are using Slic3r or Simplify 3d you can edit your printer profile and select 'Automatic' as your slicer type, and your settings will automatically be updated.  If you are using Cura you can do this too, but you have to add some stuff to your start or end gcode to get it working.  Let me know!

Boymeetsmill, 2019-03-14:
I backed up my settings, and installed the RC (0.3.5rc1.dev0) per your instructions. But I haven't been able to print with the plugin running...

First couple prints (High Quality Mode) attempted I received an error, that I thought I screen shot but lost). The print would start, but nothing would happen.  I would attempt to cancel the print, and even disable the time lapse, but the only remedy was a reboot.

Last print was in compatibility mode, I didn't receive an error, but the program hung in a similar manner. I haven't been running logs, because I haven't even gotten a print to start. 

Should I run again with logs enabled?

FormerLurker, 2019-03-14:
Wait you installed 0.3.5?  I think you should re-install the master version from the plugin manager, because 0.3.5 isn't ready for public consumption.  Unfortunately you probably need to restore the default settings since I haven't included a downgrade migration for the settings :(  FYI the most recent development version is in the 'New Settings' branch right now, since I don't even feel it's ready for a development release.  A WHOLE LOT has changed in the new version and testing is quite the nightmare :)

If you really meant 0.3.4rc1.dev0, definitely upgrade to 0.3.4!  Lots of bugs were fixed and other improvements were added.

Let me know what you find regarding the version and perhaps run another test with logging enabled.  If you still have issues, you can send me your gcode, settings.json, and plugin_octolapse.log and I bet I can figure it out!

FormerLurker, 2019-03-14:
OH my gosh, I have been working on too many github issues, and royally messed up my response! 
 I was reading an earlier post that I'd sent, and not the most recent.  IGNORE EVERYTHING I SAID.  It's a good thing I decided to re-read the last few messages!

Ok, so a TON has changed in the new_settings branch since I sent you that last message.  I haven't created any guides or even updated the instructions on how to operate the new version.  I'm doing this all from memory, so we might need some back-and-forth to get everything working.  Also, thanks for helping me test this very very different version of Octolapse.  Here's what you should start with:

1.  If you didn't install the plugin yesterday, please reinstall from the same link above (https://github.com/FormerLurker/Octolapse/archive/new_settings.zip) because I've made some crucial updates in the last few days.
2.  It's possible your settings.json file is messed up, and restoring the defaults won't fix it because the existing 'defaults' have not been adjusted for the new version.  If this is the case, I'll fix it probably tomorrow and you can reinstall.  You can check the settings by opening and saving each profile that is selected on the Octolapse tab.  After you save, make sure to re-open and examine the settings to make sure they look correct.  Some of them might change if the settings file is corrupt.
2.  Make sure you printer profile slicer type is set to 'Automatic Configuration' since you are using simplify (hooray!)
3.  Make sure 'Disable Automatic Save' is UNCHECKED.  This will be useful for debugging since I haven't tested Simplify as well as Slic3r yet. (cura is also not thoroughly tested).
4.  Double check 'Auto-Detect Position' setting.  If you know your origin, disable and set your origin manually for now.  If not, it's probably OK, but I was having problems with this last week (fingers crossed that I fixed it).
5.  See if 'Snapshot Position Restrictions' is enabled.  If it is you can probably disable it.  It's for preventing snapshots that are outside of that area, and is super useful for some printers (like the Prusa) that prime outside of the normal print area, and for preventing snapshots below a certain position for printers that touch the nozzle to the bed when they home (I just learned that some do this).
6.  This one will require the terminal because I'm not sure if G90/G91 affect your extruder.  We can send a few commands from the OctoPrint terminal to see.  So go to the terminal in Octoprint and send the following commands in order:

```
M83
G90
```
MAKE SURE your filament is unloaded for this next command:
```
G1 E-100 F1000 ; I believe 1000mm/min is safe for your printer, no?  If not, set F lower here
```
Your printer should try to retract 100 mm.  After you send this next command, you'll need walk to your printer BEFORE it could retract 100 mm.  If you can't get to it in time, lower the speed (f parameter).  However, do NOT change the e parameter!
```
G1 E-100 F1000
```
If your extruder does NOT try to retract again, then G90/G91 do NOT affect your extruder.  In that case set the 'G90/G91 Influences Extruder' to 'False' in your printer profile.  If it DOES try to retract again, set this to True.  This is VERY important.

Next, go to snapshot settings.  Just use the regular 'Layer/Height' trigger.  Make sure 'Trigger on Zhop' is disabled, and that the extruder state requirements look like this:

[image]

Disable feature detection and position restrictions.

You might be interested in checking out the camera settings.  I added new controls to make it much easier to adjust.  Check out this patreon post for more info:  https://www.patreon.com/posts/tweak-your-with-25258242

Then turn on full diagnostic logging.  Some of the logging features are currently broken, but I think there will be enough info to help.

Now run a test print.  Something small and verify that Octolapse is working.  If not, send me your settings.json file, plugin_octolapse.log and your gcode.  I should be able to figure it out.

If it works you might be interested in trying the new stabilization, which is designed to absolutely minimize artifacts and maximize speed at the expense of a fully stabilized timelapse.  The timelapses still look pretty good, and may even look perfect depending on the model.  You'll have to create your own stabilization profile at OctoPrint Settings (spanner/wrench icon)->Octolapse->Stabilizations->New profile.  Set it up like this:

[image]

Then run a small test print.  You'll see a new panel on the Octolapse tab:  Snapshot Plans.  It shows you where the snapshots will be taken, and allows you to run an animation of the full timelapse.  Or you can manually move forwards or backwards, or go to the current plan.  All of the snapshots are pre-calculated, which requires a lot less CPU power too.  You will likely be the first person besides me to try this, fyi, and it's definitely not production ready, but I've done dozens of prints, including spiral vase mode prints.

I'm super interested in how this all turns out.  Please keep feeding me info/questions/bugs and I'll keep responding.  Also, thanks for being a guinea pig, and again I'm sorry I misread the last post completely.

FormerLurker, 2019-10-12:
So, the snap to print feature has been souped up, and V0.4 is nearing release.  I'm going to go ahead and close this, but if anyone wants to test it out, let me know!

Closed!

closed 2019-10-12 by FormerLurker

smplfy, 2020-06-09:
Maybe I'll pick this up where it was left off :) I'm experiencing similar issues, with a premature prime or extrude before returning to the print, resulting in really crusty prints. 

Below is a quick video to show what Im experiencing. I'm using the latest public release of V0.4 (AMAZING updates by the way, I'm a big fan). I'm getting this exact same result on both my Prusa MK3S MMU2S and a heavily modded CR-10S (bondtech direct drive e3dv6). For normal prints, I use the exact same profiles as my Prusa and I'm using Prusa Slicer for both. I've also attached my latest GCODE, but I get the same results with any I've tried. In this print I increased retraction to 2mm (which actually caused a clog on this print).

[Archive 2.zip](https://github.com/FormerLurker/Octolapse/files/4754325/Archive.2.zip)

Thanks again for all your hard work and help. We definitely appreciate such an amazing product.

FormerLurker, 2020-06-09:
@smplfy, so this is probably not the right place to discuss this since the issue here is for a much older version, but I will give it a shot.  Check out this setting:

[image]

Unless you put marlin 2 on your CR-10, that should be false.  Try selecting your make/model in the import drop down box at the top.

In fact, I would delete all of those printers in your printer profiles except your CR-10 and MK3, then make sure your make/model are selected (you probably won't need to customize anything)

Let me know if that helps.

smplfy, 2020-06-09:
Sorry about that! Thanks for the quick reply. Let me know if you want me to open a new issue.

I am running Marlin 2 with LA 1.5 running on this machine. With an SKR 1.3 mainboard and TMC2209 stepper drivers.

Thanks!

FormerLurker, 2020-06-09:
Then ignore what I said about the CR-10 setting.  They are known to be difficult to tune with Octolapse, unfortunately.  Now the Mk3 is a different story.  You really shouldn't have any issues.  Make sure you are NOT running a release candidate firmware (they actually switched that g90/g91 influences extruder behavior a while back, then reverted it in another RC), and try a new spool of filament (wet pla will string period).  

Now, I notice you must not have been using the 'Smart - Compatibility' trigger for this print, since the snapshot isn't starting in the right place (should be starting on the left side of the print.  Also, I noticed you don't have verbose gcode enabled (read the [getting started guide](https://github.com/FormerLurker/Octolapse/wiki/V0.4---Getting-Started) to learn about that).

Judging from your video you are using a DSLR, which adds a lot of time.  I can actually see your filament oozing.  If you try stabilizing in the center (or at least closer to your print) and not using a DSLR, you will have much better results on the CR-10.  Even with my Mk2.5S I have some issues when using a DSLR (adds 4-5 additional seconds).  You might also try the 'Snap to Print' trigger (not with a DSLR though for sure, it will really mess up your quality).  The extruder won't be out of the way, but your print quality should be really good.

Also, I'm seeing strings on your print that don't look like they are caused by Octolapse.  Get that worked out.  Could be wet filament.

Check the [troubleshooting print quality guide](https://github.com/FormerLurker/Octolapse/wiki/V0.4---Troubleshooting-Print-Quality-Issues) too in case I missed something.

smplfy, 2020-06-09:
Yeah, it’s strange. I’ll test what you mentioned. I was using the Smart Compatibility, but eventually switched because I got better results forbidding snapshots during extrusion using “real time”. I had used verbose previously but I guess I forgot to save that profile. From a hardware perspective, this machine is very close to the MK3S, and usually has very little stringing, but I do have some stringing in the support for this model.

I’ll go back and test Smart Quality with verbose code on. I’ve also switched to using the batch image download script, to have it spend as little time in the corner as possible.

I’ve actually been using Octolapse for quite a while (https://www.youtube.com/channel/UCMMQL0QRQQu0SRrVkKRI68A) and you helped me with some issues getting the DSLR working. After moving, and picking things back up months later though, I’ve run into these issues.

So, thanks, as always for your responsiveness!

-Daniel

FormerLurker, 2020-06-09:
@smplfy, any time!  FYI, for larger items with infill the smart triggers (I would stick with the compatibility one for now) will almost always outperform the real-time triggers.

Your Mk3 profile is using the 'use octoprint settings' for g90/g91 influences extruder, so if you import the default profile that should be fixed.  That is probably the most important setting to get right from a quality perspective, so double check that you are using a release version of the Mk3 firmware and I think those problems should go away.

Also, regarding the recent move, could your filament have gotten wet?  I've been collecting some print quality information for that quality guide I linked to, and have some before and after shots of using old wet filament vs new dry filament, and it is night and day.  I was shocked, actually.