FormerLurker/Octolapse #208: An idea, timelapse with 2 webcams.

sierdw84, 2018-07-29:
Is it an idea to be able to make a timelapse with two webcams at the same time? and then with ffmpeg put this in one movie.

`ffmpeg -i 1.mp4 -qscale 0 timelapse_1.mpg`
`ffmpeg -i 2.mp4 -qscale 0 timelapse_2.mpg`
`cat timelapse_1.mpg timelapse_2.mpg | ffmpeg -f mpeg -i - -qscale 0 -vcodec mpeg4 timelapse_output.mp4`
`

My knowledge of python is too little to be able to do this myself

how I had it in my head
[http://vdwielen.eu/voorbeeld_timelapse.jpeg](url)

FormerLurker, 2018-08-01:
I am working on a multi camera version as we speak, but currently it won't be able to combine them into a single timelapse.  I'm allowing any number of cameras, and it will quickly get hairy if we try to combine 3 or 4 timelapses into one.

That being said, it would be pretty easy to post-process two videos into one.  Perhaps I a guide could be written?  I'm not very handy with video editing myself.

Shadowen, 2018-08-02:
I suppose if you wanted to turn this into a feature request, we would need some post-processing script functionality. Maybe just run a provided Python script with predefined variables?

Maybe when multi-camera is fully implemented :)

FormerLurker, 2018-08-02:
Yes, exactly.  I'm going to be adding some additional script options anyway, so it would just be a matter of passing in the right arguments and creating the right bash/bat scripts.