FormerLurker/Octolapse #210: Fix FFMPEG check

FormerLurker, 2018-07-30:
The FFMpeg check breaks silently if there is no path entered.  Need to check for an exception and send an appropriate message to the client.

Shadowen, 2018-08-03:
On Octolapse v0.3.3rc2 (`rc/devel`?), when I try to start with no ffmpeg path set in OctoPrint, I get the following error on print start:

[image]

I also tested at `devel` `HEAD`. Can you give some replication instructions?

FormerLurker, 2018-08-03:
I think it is maybe a Windows only issue.

FormerLurker, 2018-08-03:
Unless I fixed it and forgot.  Saw it in a user log while debugging another issue.

FormerLurker, 2018-08-11:
I think this problem only exists when no value was ever entered in the OctoPrint ffmpeg location.  I was using the wrong version of get (should use global_get).  Closed.

closed 2018-08-11 by FormerLurker