FormerLurker/Octolapse #213: clean snapshots after rendering

rafaa1988, 2018-08-02:
i cannot uncheck the box near "clean snapshot after rendering" the box before the last in snapshot. It's a bug issue

FormerLurker, 2018-08-02:
I'll test it ouy, thanks for reporting this!

Shadowen, 2018-08-02:
I'm at `HEAD` on `devel` and cannot replicate this issue. Are you actually incapable of unchecking the box (the checkbox is disabled), or is it just not saving when you hit save?

Shadowen, 2018-08-02:
Also what release are you running? `stable`, `maintenance/rc`, `devel/rc`? Should be set according to your OctoPrint settings under Software Update.

FormerLurker, 2018-08-02:
I think this was already fixed actually.  I believe I had both of these options set to update the same settings variable, but let me confirm.

rafaa1988, 2018-08-02:
I was incapable of saving after unchecking the box. I'm using the stable master version of Octoprint and Octolapse.

FormerLurker, 2018-08-02:
I found the problem, and it has been fixed in devel:

```
job = TimelapseRenderJob(
            job_id,
            rendering,
            settings.current_debug_profile(),
            print_name,
            snapshot_directory,
            snapshot_file_name_template,
            output_tokens,
            octoprint_timelapse_folder,
            ffmpeg_path,
            thread_count,
            render_task_queue,
            time_added,
            on_render_start,
            on_complete,
            snapshot.cleanup_after_render_complete,
            snapshot.cleanup_after_render_complete
)
```

@rafaa1988, try installing from this url:  https://github.com/FormerLurker/Octolapse/archive/v0.3.3rc2.zip

That should fix your issue.  Let me know and I'll close this ticket.

Thanks!

rafaa1988, 2018-08-03:
solved. thanks

closed 2018-08-03 by rafaa1988