FormerLurker/Octolapse #215: Issues Logitech C920 Webcam Manual Focus

tixenre, 2018-08-03:
#### If this is a feature request describe it here

FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: 0.3.1

#### Version of OctoPrint

OctoPrint Version: 1.3.9

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: Yes and No

#### What were you doing when the problem occurred

1. Trying to configure manual focus on Logitech C920 Webcam

#### What should have happened?
Change focus

#### What happened instead?
Nothing

#### Operating System running OctoPrint and Octolapse

OS Name: OctoPi
Os Version: 0.15.1

#### Printer model & used firmware incl. version

Printer Model: Prusa MK3
Printer Firmware Version: Marlin 3.3.1

#### Browser and version of browser, operating system running browser

Browser: BROWSER_VERSION_GOES_HERE
Browser OS: BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: no gcode need

#### Link to settings.json

Link to settings.json with all passwords removed: SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.  I'm currently saving for some additional test cameras in preparation for develiping multi-camera support.

FormerLurker, 2018-08-03:
Hi there!  There have been some changes to mjpegstreamer in octopi that require some setup to get the camera controls working.  Check out this guide and let me know if it helps:https://github.com/FormerLurker/Octolapse/wiki/Camera-Profiles#custom-image-preferences

tixenre, 2018-08-04:
I had done everything from [https://github.com/FormerLurker/Octolapse/wiki/Camera-Profiles#custom-image-preferences](url) except for the 

Required octopi.txt changes to use custom image preferences

Why can't I change contrast, zoom, focus, etc?
In recent versions of Octopi, the default mjpegstreamer control.htm page is disabled. In order to automatically adjust camera settings, Octolapse needs access to control.htm. To enable access you must manually edit boot/octopi.txt and change the following lines:

camera_http_webroot="./www-octopi"

camera_http_options="-n"

to this:

camera_http_webroot="./www"

camera_http_options=""

Thanks!

FormerLurker, 2018-08-04:
Awesome!!!  I'm in the process of adding some testing to detect mjpegstreamer, and complain if control.htm isn't available.  Hopefully that will help others who are in your situation (which is anyone with the newest octopi build!)..

I'm going to mark this CLOSED!

closed 2018-08-04 by FormerLurker