FormerLurker/Octolapse #220: Possible issue with Gcode Trigger and Octolapse disabled

mmontminy, 2018-08-11:
#### If this is a feature request describe it here

FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: 0.3.1

#### Version of OctoPrint

OctoPrint Version: 1.3.9

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: NO

#### What were you doing when the problem occurred

Using the improve print quality wiki, I recently configured Octolapse to trigger snaps using an after layer change gcode command of Snap. This worked fine, until...

I started printing a number of uninteresting parts, and since I was still having some zit'ing issues caused by Octolapse, I turned it off using the disable slider on the Octolapse tab.

I started randomly having print failures, most often where the extruder stepper simply stopped working. All other aspects of the printer appeared normal, prints resumed, just no extrusion happening. Some prints worked, some failed.

Working with Prusa support, I went through all sorts of steps, factory resets, re-calibration, checking for faulty wiring, etc. Then on one nasty failure I was able to examine the Octoprint terminal, and scrolled back to see a number of errors from the printer. Scrolling back towards where they started occurring, I noticed they started in response to the "Snap" gcode.

It became obvious that with Octolapse disabled, it wasn't consuming the Snap command, and it was getting passed onto the printer, which didn't handle it well. 

This is more operator error (and the printers inability to handle this better), but there seems to be things that could be done to help avoid this issue.

If Gcode Trigger is enabled, and the user disables Octolapse, a warning/reminder that their gcode will be sending an unknown command to the printer.

Update the wiki/guides to use a valid, but harmless Gcode, maybe M117 Snap. I've seen suggestions of M118 Snap. I plan to try one of these.

If Gcode trigger is enabled, and Octolapse is disabled, go into a mode where it's not fully disabled, still monitoring gcode and consuming any snap triggers.

#### What should have happened?
Prints should have succeeded normally

#### What happened instead?
Printer behaved badly

#### Operating System running OctoPrint and Octolapse

OS Name: Octopi
Os Version: 0.15.1

#### Printer model & used firmware incl. version

Printer Model: Prusa MK2/S
Printer Firmware Version: 3.1.0

#### Browser and version of browser, operating system running browser

Browser: BROWSER_VERSION_GOES_HERE
Browser OS: BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.  I'm currently saving for some additional test cameras in preparation for develiping multi-camera support.

FormerLurker, 2018-08-11:
I have never tested that scenario, thanks for mentioning it.  I'll try to
add some code to suppress the snapshot gcode when octolapse is disabled,
and an option within the printer profile to enable this and will get back
with you.  Sorry you want through all of that debugging with prusa!

FormerLurker, 2018-08-11:
Ok, a fix has been added, as well as a few other small fixes and enhancements.  If you would, please reinstall from this URL (no need to uninstall first):  https://github.com/FormerLurker/Octolapse/archive/130609e0e55905176e284ac45a3251ea834c78fe.zip

You'll find the new setting within the printer profile already enabled by default.  Let me know if this helps or doesn't.  Thanks again!

mmontminy, 2018-08-11:
Wow, that was quick :) I'll install this version as soon as my current print is done and do some testing.

I still can't be 100% sure this caused my problems, but it certainly fit into the sequence of events. Things had been pretty stable until I turned off Octolapse, then various weird behaviors were happening with the printer. I believe Prusa support has also sent the thread to their firmware team to investigate. 

I wasn't going to report it, since really, this is user inflicted, but I figured anything that can be done to make printing smarter and safer is a good thing. One of my fails got stuck in a z-hop loop, leaving me with a pretty blackened part and a missing chunk of PEI. Seeing the potential disaster that could have become, I decided it was worth reporting.

FormerLurker, 2018-08-11:
When I decide which bugs to fix I usually run through a few checks:
1.  The bug could cause print quality issues (check)
2.  The bug is mysterious and confusing to debug (check)
3.  The bug causes third parties problems (check)
4.  The bug is easy to fix (check)

Therefore it was a good bug to quash, and it wasn't too difficult anyway.  It is very odd that you'd be getting these kinds of problems just by sending a garbage gcode.  My printer seems to just ignore them.  however, better safe than sorry, and let me know if you notice any difference.  If you STILL run into the same problem disable Octolapse from the plugin manager, reboot Octoprint and try again.  That way you can completely rule out Octolapse.  Let me know either way please.

Thanks!

mmontminy, 2018-08-12:
It seems like the fix does what it's intended to do. 
I currently have Octolapse set to take snapshots when it sees "M117 Snap".
I've been printing, and capturing timelapses as expected, and I've yet to see an Snap show up on the printer display, or Octoprint's terminal window.
I then disabled Octolapse for a few prints. I still don't see the Snap show up on the printer, or the terminal window. It appears it's consuming the snapshot command per the fix.

I think I'll stick with M117 Snap to be on the safe side, since it'll be harmless if it gets out to the printer, say I drop the gcode on an SDcard at some point. But this fix should help anyone who has a printer that doesn't handle a non-gcode snapshot command.

Thanks for the quick fix.

closed 2018-08-12 by mmontminy