FormerLurker/Octolapse #221: Autodetect vase mode when triggering on layer changes

yoyo42, 2018-08-12:
#### If this is a feature request describe it here

I typically have Octolapse set to trigger on layer changes. If I print something in vase mode (a rising spiral) then a layer change is detected very frequently and 95% of the print time is spent pointlessly moving to the snapshot position.

It would be nice if Octolapse could detect vase mode and (optionally?) switch to a more suitable mode such as a height based trigger. The layer height in use for vase mode could be detected by searching for the next instance of the same X-Y position and calculating the Z difference, but a simpler default of say 0.2mm would work fine since this is really just a safety mechanism.

It is annoying to print all of the brim/raft/base layers then find this happening and either letting it take forever to print or disabling Octolapse being the only choices available.

I don't know how much G-Code visibility Octolapse has, i.e. whether this could be detected at thye start of the print but if not a sensible fallback as soon as the issue is spotted (maybe using a 'minimum time between triggers' threshold lasting over a few triggers?) would be a useful feature.

Thanks   

#### Version of Octolapse
 
Octolapse Version: 0.3.1

#### Version of OctoPrint

OctoPrint Version: 1.3.9

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES_OR_NO

#### What were you doing when the problem occurred

1. STEP_ONE_GOES_HERE
2. STEP_TWO_GOES_HERE
3. STEP_...

#### What should have happened?
PUT_YOUR_DESCRIPTION_HERE

#### What happened instead?
PUT_YOUR_DESCRIPTION_HERE

#### Operating System running OctoPrint and Octolapse

OS Name: OS_NAME_GOES_HERE
Os Version: OS_VERSION_GOES_HERE

#### Printer model & used firmware incl. version

Printer Model: PRINTER_MODEL_GOES_HERE
Printer Firmware Version: PRINTER_FIRMWARE_VERSION_GOES_HERE

#### Browser and version of browser, operating system running browser

Browser: BROWSER_VERSION_GOES_HERE
Browser OS: BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.  I'm currently saving for some additional test cameras in preparation for develiping multi-camera support.

FormerLurker, 2018-08-13:
Maybe I could add an option to rate limit snapshots as you suggest.  I think this would be a straightforward process, and one that wouldn't cause too much confusion.  This would probably need to be disabled by default since even a short rate limit (say 30 seconds) would cause some timelapse issues even on a benchy (very low layer change time on the stack). 

It's still best to use a height trigger when printing in vase mode though.  I know how irritating it is to cancel a print, wait for the bed to cool before scraping the hard to remove first layer, and then start again, though, so I'll definitely look into it.

Thanks for your suggestion!

collse, 2018-11-22:
and I thought I was the only one thinking not being able to getting this to work properly - it did extend my print time to at least 300% - would be nice to see if you could specify a Z position and snapshot every other time whenever it is being surpassed no matter the X / Y position preferable on the deeper Y position ...

FormerLurker, 2018-11-22:
You can!  Set the height in your snapshot profile and octolapse will take a pic at that interval.  Op wanted me to automatically detect this situation, and I will implement that soon.

FormerLurker, 2019-10-12:
This has been implemented!  Vase mode detection works for Simplify3d, Slic3r and Cura (though cura requires a small script to be added to the start/end gcode).  There is also a failsafe 'minimum layer height' that prevents Octolapse from going totally haywire.  This will be available starting with V0.4, which is in pre-release testing right now.  I'm going to close this issue, but if anyone wants to test, please let me know!

Also, FYI, there is a new stabilization mode, snap to print, which improves print quality when using vase mode by preventing the extruder from leaving the printed part while still trying to stabilize the timelapse as much as possible.

Closed!

closed 2019-10-12 by FormerLurker