FormerLurker/Octolapse #223: [Request] Custom GCODE to temporarily disable Octolapse

mmontminy, 2018-08-29:
#### If this is a feature request describe it here

There are a fair number of prints I do where I don't want Octolapse enabled, it's simply unnecessary.

I think a nice feature would be a setting similar to the gcode snapshot trigger where Octolapse would either enable or disable for the current print based on seeing a specific gcode.

Use-case: I'm currently printing a fleet of scaled down Benchies to have samples of the various filaments. I'm always forgetting to turn off Octolapse before I print these. Since I have a pre-sliced gcode saved for this specific print, I could simply embed an M117 Octolapse Off code, and with this feature, configure Octolapse to look for that specific gcode and disable itself for the current print.

I'm effectively achieving this same result now, by using the snap trigger gcode, and omitting it from slices I don't want to timelapse. That works, but it creates an error popup after every print telling me no snapshots exist. It also seems unnecessary for Octolapse to be active if I know it won't do anything. 

#### Version of Octolapse
 
Octolapse Version: 0.3.3rc3.dev0

#### Version of OctoPrint

OctoPrint Version: 1.3.9

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES_OR_NO

#### What were you doing when the problem occurred

1. STEP_ONE_GOES_HERE
2. STEP_TWO_GOES_HERE
3. STEP_...

#### What should have happened?
PUT_YOUR_DESCRIPTION_HERE

#### What happened instead?
PUT_YOUR_DESCRIPTION_HERE

#### Operating System running OctoPrint and Octolapse

OS Name: OS_NAME_GOES_HERE
Os Version: OS_VERSION_GOES_HERE

#### Printer model & used firmware incl. version

Printer Model: PRINTER_MODEL_GOES_HERE
Printer Firmware Version: PRINTER_FIRMWARE_VERSION_GOES_HERE

#### Browser and version of browser, operating system running browser

Browser: BROWSER_VERSION_GOES_HERE
Browser OS: BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.  I'm currently saving for some additional test cameras in preparation for develiping multi-camera support.

FormerLurker, 2018-08-29:
That sounds like an interesting feature.  I've been thinking about adding gcodes that can control some of Octolapse's settings as well, but haven't come up with a format for the gcodes yet.  I'll think about this while I prepare for the next release.

Thanks for posting your ideas!

mmontminy, 2018-08-29:
For what it's worth, I like the notion of using M117 with some syntax to the message vs artificial codes.
It's likely to be universal to all printers, and unlikely to cause any harm if it makes it to the printer. A little annoying maybe if it overwrites a prior message if Octolapse isn't installed/available at worst.

I find it's also a visual indicator that Octolapse is/isn't processing my commands. When I see an M117 Snap on my display, it's a reminder that I forgot to enable it.

I think expanding this to incorporate the ability to pick profiles, etc would be pretty sweet. Some prints I might want to park the head, some I might want an orbit, etc.

FormerLurker, 2019-12-18:
OK, so I've added this feature finally (been a LONG while, I know).  I did not use M117, though I may add some alternative commands at some point, like parsing M117 messages into commands.  Here are the two commands (available only in the devel branch at the moment):
```
@Octolapse start-snapshots
@Octolapse stop-snapshots
```

Each print starts with snapshots enabled, and they can be disabled with the stop-snapshot command.  Then they can be re-enabled with the start-snapshots command.  This is extremely useful in start-end gcode, especially since I added a new trigger type 'snap-to-print' that can be tricked by start/end gcode and mess up the first/last frame of the timelapse.

I've also added the following command:
```
@Octolapse take-snapshot
```
which can be used to take a picture in addition to the custom command in the trigger profile (like G4 P0, which is common).

Let me know if you're interested in testing!

FormerLurker, 2020-02-26:
It's been a long time since this issue was updated, so I'm going to close it down.  Feel free to post another issue if you have any other suggestions or issues.  Thanks!

closed 2020-02-26 by FormerLurker