FormerLurker/Octolapse #224: doesn't work with octoprint 1.3.9

getharder, 2018-08-30:
after upgrading to octoprint 1.3.9, octolaps stops working. plugin is enabled, can be configured but does not creating snapshots during prints (z-layer change) as before.

#### Version of Octolapse
Octolapse Version: 0.3.1

#### Version of OctoPrint
OctoPrint Version: 1.3.9

#### When you ran into the problem, did you have diagnostic logging enabled?
Diagnostic Logging was Enabled: NO

#### What were you doing when the problem occurred

tried to reinstall, reconfigured the pluging

#### What should have happened?
doing snapshots :D

#### What happened instead?
doing no snapshots as before

#### Operating System running OctoPrint and Octolapse

OS Name: OctoPi 
Os Version: 0.14.0

#### Printer model & used firmware incl. version
Printer Model: Prusa MK3 i3
Printer Firmware Version: 3.3.1-845

#### Link to settings.json
Link to settings.json with all passwords removed: cannot download, url not found on octoprint

FormerLurker, 2018-08-30:
HI @getharder!  This bug has been fixed but I've not had time to get a release out because I added too many features at one time.  However, I'm in the final stages of documenting and testing, so maybe I'll finish soon.  

There are some maintenance release candidates available, but even those are now somewhat outdated.  Here is a url you can use to install the most recent release candidate via the plugin manager:  https://github.com/FormerLurker/Octolapse/archive/v0.3.3rc3.dev.zip

It should work with 1.3.9.  If you would rather use a newer (alpha) build and maybe help me with testing, let me know and I'll give you a link to install the most recent commit.

Thanks!

AbuMaia, 2018-09-01:
I'm also using OctoPrint 1.3.9, and I installed Octolapse v0.3.3rc3.dev, but whenever it is enabled, the printer just sits there doing nothing, until I cancel the print and disable Octolapse, then restarting the print it prints normally.

getharder, 2018-09-01:
Hi FormerLurker,

thx for your quick reply. Installed v0.3.3rc3, but printing doesn't start. OctoPrint shows the follwing error:

```
Octolapse Startup Failed
Unable to start the timelapse. Cancelling print. Error: The current camera profile did not pass testing. You can adjust and test your camera in the profile settings page.
```
checked camera settings in Octolapse settings, test snapshot works. 

what do you need to debug this problem?

best regards!

FormerLurker, 2018-09-03:
@getharder and @AbuMaia,

I just released a new version and figured you both might be interested.  @AbuMaia, since yours is working now, maybe it would be best to wait for the release version (this is an rc/devel release).

Here is a link to the release:  https://github.com/FormerLurker/Octolapse/releases/tag/v0.3.4rc1.dev

@getharder, your problem most likely has something to do with the custom image preferences within your camera profile being enabled while control.htm is disabled (newer versions of octopi have disabled control.htm by default).  The new version has better error detection and should provide you with [this link](https://github.com/FormerLurker/Octolapse/wiki/Camera-Profiles#required-octopitxt-changes-to-use-custom-image-preferences) to fix the error.  Before trying the new version, check your camera profile and try disabling custom image preferences (check at the bottom).

If any of you try the new release please leave your [feedback here](https://github.com/FormerLurker/Octolapse/issues/226).

Thanks!

getharder, 2018-09-05:
Hi @FormerLurker 

disabling the custom image preferences helped, had a small print yesterday but with the 0.3.1 release. I'll give the 0.3.4rc1 a try at the weekend.

best regards!

FormerLurker, 2018-09-05:
@getharder, that's great to hear!  You'll be glad to know that the new version checks for this problem and includes an error popup with a link to the solution.  It's users like you reporting these problems that allows me to fix and improve the program so others don't have the same issues.  Thank you for your contribution, and I look forward to getting some feedback on the new rc/devel release!

getharder, 2018-09-09:
@FormerLurker 0.3.4r1 works as expected. The only thing, after installing i had to reset all settings to default and reconfigure the plugin. i'll use it this week, if any problems occurs i'll report them.

thanks for your quick support and your great plugin!

best regards!

getharder, 2018-09-11:
@FormerLurker updated yesterday to to v0.3.4rc1.dev2, all settings remained, did a long test print (10hours) and it worked like a charm.

thx. for fixing it!

if any problems will arise, i'll respond to #232 

keep up the good work on your great plugin.

cheers!

closed 2018-09-11 by getharder

FormerLurker, 2018-09-11:
That's good to hear, thanks for reporting back!  If you get any good timelapses, shoot me a link please.  I love watching user generated content :)